coolsnowwolf/lede #4186: 编译失败：lede/include/toplevel.mk:225: recipe for target 'world' failed

awakezz, 2020-04-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败：
`package/Makefile:107: recipe for target '/home/zz/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/zz/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zz/lede'
Build failed - please re-run with -j1 to see the real error message
/home/zz/lede/include/toplevel.mk:225: recipe for target 'world' failed
`

### (2) 路由器型号和固件版本

A：友华 WR1200JS

### (3) 详细日志

A：
`*** Installing runtime libraries
install libcrypto.so.1.1 -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/lib/libcrypto.so.1.1
install libssl.so.1.1 -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/lib/libssl.so.1.1
*** Installing development files
created directory `/home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include'
created directory `/home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl'
install ./include/openssl/aes.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/aes.h
install ./include/openssl/asn1.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/asn1.h
install ./include/openssl/asn1_mac.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/asn1_mac.h
install ./include/openssl/asn1err.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/asn1err.h
install ./include/openssl/asn1t.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/asn1t.h
install ./include/openssl/async.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/async.h
install ./include/openssl/asyncerr.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/asyncerr.h
install ./include/openssl/bio.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/bio.h
install ./include/openssl/bioerr.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/bioerr.h
install ./include/openssl/blowfish.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/blowfish.h
install ./include/openssl/bn.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/bn.h
install ./include/openssl/bnerr.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/bnerr.h
install ./include/openssl/buffer.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/buffer.h
install ./include/openssl/buffererr.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/buffererr.h
install ./include/openssl/camellia.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/camellia.h
install ./include/openssl/cast.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/cast.h
install ./include/openssl/cmac.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/cmac.h
install ./include/openssl/cms.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/cms.h
install ./include/openssl/cmserr.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/cmserr.h
install ./include/openssl/comp.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/comp.h
install ./include/openssl/comperr.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/comperr.h
install ./include/openssl/conf.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/conf.h
install ./include/openssl/conf_api.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/conf_api.h
install ./include/openssl/conferr.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/conferr.h
install ./include/openssl/crypto.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/crypto.h
install ./include/openssl/cryptoerr.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/cryptoerr.h
install ./include/openssl/ct.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/ct.h
install ./include/openssl/cterr.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/cterr.h
install ./include/openssl/des.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/des.h
install ./include/openssl/dh.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/dh.h
install ./include/openssl/dherr.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/dherr.h
install ./include/openssl/dsa.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/dsa.h
install ./include/openssl/dsaerr.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/dsaerr.h
install ./include/openssl/dtls1.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/dtls1.h
install ./include/openssl/e_os2.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/e_os2.h
install ./include/openssl/ebcdic.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/ebcdic.h
install ./include/openssl/ec.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/ec.h
install ./include/openssl/ecdh.h -> /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/include/openssl/ecdh.h
[... 280 lines cut ...]
cp -fpR /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/etc/ssl/openssl.cnf /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/.pkgdir/libopenssl-conf/etc/ssl/
touch /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/.pkgdir/libopenssl-conf.installed
rm -rf /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/.pkgdir/openssl-util.installed /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/.pkgdir/openssl-util
mkdir -p /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/.pkgdir/openssl-util
install -d -m0755 /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/.pkgdir/openssl-util/usr/bin
install -m0755 /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/ipkg-install/usr/bin/openssl /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/.pkgdir/openssl-util/usr/bin/
touch /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/.pkgdir/openssl-util.installed
mkdir -p /home/zz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/zz/lede/tmp/.root-copy.flock -c 'cp -fpR /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/.pkgdir/libopenssl-conf/. /home/zz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/zz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.libopenssl-conf_installed
echo "libopenssl-conf" >> /home/zz/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/openssl.default.install
mkdir -p /home/zz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/zz/lede/tmp/.root-copy.flock -c 'cp -fpR /home/zz/lede/build_dir/target-mipsel_24kc_musl/openssl-1.1.1f/.pkgdir/openssl-util/. /home/zz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/zz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.openssl-util_installed
echo "openssl-util" >> /home/zz/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/openssl.default.install
[... cut ...]
time: package/libs/openssl/compile#571.02#169.18#339.69
make[2]: Leaving directory '/home/zz/lede'
package/Makefile:107: recipe for target '/home/zz/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/zz/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zz/lede'
Build failed - please re-run with -j1 to see the real error message
/home/zz/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 1
`

closed 2020-04-05 by coolsnowwolf

jiatongwu, 2020-04-14:
我也遇到 这个问题 了
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0/gcc -I/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0/gcc/. -I/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0/gcc/../include -I/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0/gcc/../libcpp/include -I/home/ubuntu/lede/staging_dir/host/include -I/home/ubuntu/lede/staging_dir/host/include -I/home/ubuntu/lede/staging_dir/host/include  -I/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0/gcc/../libdecnumber -I/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0/gcc/../libbacktrace   -o insn-emit.o -MT insn-emit.o -MMD -MP -MF ./.deps/insn-emit.TPo insn-emit.c

g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1099: recipe for target 'insn-emit.o' failed
make[5]: *** [insn-emit.o] Error 4
make[5]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-initial/gcc'
Makefile:4209: recipe for target 'all-gcc' failed
make[4]: *** [all-gcc] Error 2
make[4]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-initial'
Makefile:36: recipe for target '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-initial/.built' failed
make[3]: *** [/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-initial/.built] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#618.66#854.19#8053.39
toolchain/Makefile:98: recipe for target 'toolchain/gcc/initial/compile' failed
make[2]: *** [toolchain/gcc/initial/compile] Error 2
make[2]: Leaving directory '/home/ubuntu/lede'
toolchain/Makefile:96: recipe for target '/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
/home/ubuntu/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

idoall, 2020-06-13:
遇到同样的问题

checking whether mbrtowc handles incomplete characters... (cached) guessing yes
checking whether mbrtowc works as well as mbtowc... (cached) guessing yes
checking whether mbsrtowcs works... guessing yes
checking for mempcpy... (cached) yes
checking for memrchr... yes
checking whether mkdir handles trailing slash... yes
checking whether mkdir handles trailing dot... yes
checking for mkdtemp... yes
checking whether mkfifo rejects trailing slashes... yes
checking whether mknod can create fifo without root privileges... configure: error: in `/home/work/lede/build_dir/host/tar-1.30':
configure: error: you should not run configure as root (set FORCE_UNSAFE_CONFIGURE=1 in environment to bypass this check)
See `config.log' for more details
Makefile:31: recipe for target '/home/work/lede/build_dir/host/tar-1.30/.configured' failed
make[3]: *** [/home/work/lede/build_dir/host/tar-1.30/.configured] Error 1
make[3]: Leaving directory '/home/work/lede/tools/tar'
time: tools/tar/compile#12.79#10.82#28.37
tools/Makefile:161: recipe for target 'tools/tar/compile' failed
make[2]: *** [tools/tar/compile] Error 2
make[2]: Leaving directory '/home/work/lede'
tools/Makefile:159: recipe for target '/home/work/lede/staging_dir/target-x86_64_musl/stamp/.tools_compile_yynyynnyyynyyyyynyyynnyyyynyyyyyyyyyyyyyyyyyynynnyyyynynyyy' failed
make[1]: *** [/home/work/lede/staging_dir/target-x86_64_musl/stamp/.tools_compile_yynyynnyyynyyyyynyyynnyyyynyyyyyyyyyyyyyyyyyynynnyyyynynyyy] Error 2
make[1]: Leaving directory '/home/work/lede'
/home/work/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

txdywy, 2020-07-20:
何解..

e703, 2020-08-31:
same error

Configuring luci-i18n-upnp-zh-cn.
Configuring urngd.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring dnsmasq-full.
Collected errors:
 * check_data_file_clashes: Package ath10k-firmware-qca4019-ct-full-htt wants to install file /home/alan/lean-lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/lib/firmware/ath10k/QCA4019/hw1.0/board-2.bin
        But that file is already provided by package  * ath10k-firmware-qca4019-ct
 * opkg_install_cmd: Cannot install package ath10k-firmware-qca4019-ct-full-htt.
 * check_data_file_clashes: Package ath10k-firmware-qca4019-ct-htt wants to install file /home/alan/lean-lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/lib/firmware/ath10k/QCA4019/hw1.0/board-2.bin
        But that file is already provided by package  * ath10k-firmware-qca4019-ct
 * opkg_install_cmd: Cannot install package ath10k-firmware-qca4019-ct-htt.
 * check_data_file_clashes: Package ath10k-firmware-qca4019 wants to install file /home/alan/lean-lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/lib/firmware/ath10k/QCA4019/hw1.0/board-2.bin
        But that file is already provided by package  * ath10k-firmware-qca4019-ct
 * check_data_file_clashes: Package ath10k-firmware-qca4019 wants to install file /home/alan/lean-lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/lib/firmware/ath10k/QCA4019/hw1.0/firmware-5.bin
        But that file is already provided by package  * ath10k-firmware-qca4019-ct
 * opkg_install_cmd: Cannot install package ath10k-firmware-qca4019.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/alan/lean-lede'
package/Makefile:108: recipe for target '/home/alan/lean-lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install' failed
make[1]: *** [/home/alan/lean-lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/alan/lean-lede'
/home/alan/lean-lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

alanfans, 2021-01-25:
@idoall  我跟你差不多的错，我用非root用户就好了，你也试试

idoall, 2021-01-25:
> @idoall 我跟你差不多的错，我用非root用户就好了，你也试试

感谢回复，后来也是用非root用户编译通过。看来还是权限的问题。

alanfans, 2021-01-25:
@idoall 我正在编译中，哇，这么晚了还回复。

idoall, 2021-01-25:
> @idoall 我正在编译中，哇，这么晚了还回复。

正巧在线看到了。你可以先睡了，这个编译时间有些长

alanfans, 2021-01-25:
@idoall 几个月前用mac编译过，今天用的国外的服务器编译的，很快20来分钟已经80%了