coolsnowwolf/lede #4190: 关于Adbyby规则更新建议

AwesomeSky, 2020-04-06:
关于Adbyby的规则更新 有一点拙见
1.是否可以在每日六点更新主规则的同时也更新自定义订阅规则
2.是否可以自定自动更新和重启的时间
3.是否可以WEB端更新规则时后台自动更新 无需WEB端反馈更新成功之后才能更新条目

感谢大雕的辛勤更新

hplee0120, 2020-04-06:
自从添加第三方规则之后的版本，那个Adblock Plus 规则自动更新一直都是失效的。

hxlls, 2020-04-06:
[image]
特意更新了下，还真没失效，失效是你自己的问题了

AwesomeSky, 2020-04-06:
> 自从添加第三方规则之后的版本，那个Adblock Plus 规则自动更新一直都是失效的。然后这个可以自定义时间也挺好的，之前也有人建议过。然后第三方规则自动更新据大雕所说是防止有人添加过后然后自动更新的时候炸了。

确实第三方更新的时候可能会炸 所以感觉要是能有个复选框 自动更新的时候也更新第三方规则的复选框就好了 炸了的话就取消掉 不炸的话就用这种

liyuhaolol, 2020-04-06:
> 自从添加第三方规则之后的版本，那个Adblock Plus 规则自动更新一直都是失效的。然后这个可以自定义时间也挺好的，之前也有人建议过。然后第三方规则自动更新据大雕所说是防止有人添加过后然后自动更新的时候炸了。

adbplus block更新不了，是因为ipv6的问题，要么按照L大的建议，lede编译的时候不编译ipv6，或者你会改文件的话，改2个文件，让adbyby强制使用ipv4，就百分之百可以正常更新了，我已经改了，问题解决。

AwesomeSky, 2020-04-06:
> > 自从添加第三方规则之后的版本，那个Adblock Plus 规则自动更新一直都是失效的。然后这个可以自定义时间也挺好的，之前也有人建议过。然后第三方规则自动更新据大雕所说是防止有人添加过后然后自动更新的时候炸了。
> 
> adbplus block更新不了，是因为ipv6的问题，要么按照L大的建议，lede编译的时候不编译ipv6，或者你会改文件的话，改2个文件，让adbyby强制使用ipv4，就百分之百可以正常更新了，我已经改了，问题解决。

求改文件的方法 谢谢

snakwu, 2020-04-06:
@liyuhaolol  请问是改那两个文件，怎么改，谢谢！

liyuhaolol, 2020-04-06:
> > > 自从添加第三方规则之后的版本，那个Adblock Plus 规则自动更新一直都是失效的。然后这个可以自定义时间也挺好的，之前也有人建议过。然后第三方规则自动更新据大雕所说是防止有人添加过后然后自动更新的时候炸了。
> > 
> > 
> > adbplus block更新不了，是因为ipv6的问题，要么按照L大的建议，lede编译的时候不编译ipv6，或者你会改文件的话，改2个文件，让adbyby强制使用ipv4，就百分之百可以正常更新了，我已经改了，问题解决。
> 
> 求改文件的方法 谢谢

> > > 自从添加第三方规则之后的版本，那个Adblock Plus 规则自动更新一直都是失效的。然后这个可以自定义时间也挺好的，之前也有人建议过。然后第三方规则自动更新据大雕所说是防止有人添加过后然后自动更新的时候炸了。
> > 
> > 
> > adbplus block更新不了，是因为ipv6的问题，要么按照L大的建议，lede编译的时候不编译ipv6，或者你会改文件的话，改2个文件，让adbyby强制使用ipv4，就百分之百可以正常更新了，我已经改了，问题解决。
> 
> 求改文件的方法 谢谢

/usr/share/adbyby/adblock.sh 第3行 加入"-4" wget-ssl -4 --no-check-....................
/usr/lib/lua/luci/controller/adbyby.lua第46行 加入"-4"wget-ssl -4 -q --no-check-....................
保存重启路由器，更新，大概2.6M的文件，需要更新个30s左右，当然还要看你的网速。

liyuhaolol, 2020-04-06:
> @liyuhaolol 请问是改那两个文件，怎么改，谢谢！

同上，你试试吧，好使给我说一下，嘿嘿

AwesomeSky, 2020-04-06:
> /usr/share/adbyby/adblock.sh 第3行 加入"-4" wget-ssl -4 --no-check-....................
> /usr/lib/lua/luci/controller/adbyby.lua第46行 加入"-4"wget-ssl -4 -q --no-check-.................... [...]

看着有点奇怪啊 为啥两个"-4"后面一个有空格一个没有 --no-check后面都是省略号 我回去试下 非常感谢

snakwu, 2020-04-06:
> > @liyuhaolol 请问是改那两个文件，怎么改，谢谢！
> 
> 同上，你试试吧，好使给我说一下，嘿嘿

仍然无法更新，谢谢！

AwesomeSky, 2020-04-06:
> > > @liyuhaolol 请问是改那两个文件，怎么改，谢谢！
> > 
> > 
> > 同上，你试试吧，好使给我说一下，嘿嘿
> 
> 仍然无法更新，谢谢！

我这边改完之后 第三方的更新比原来快了 ABP一直是更新正常的 
你可以浏览器访问下
https://easylist-downloads.adblockplus.org/easylistchina+easylist.txt
看看是否正常打开 
或者将easylist-downloads.adblockplus.org加入SSR代理名单

snakwu, 2020-04-06:
> > > > @liyuhaolol 请问是改那两个文件，怎么改，谢谢！
> > > 
> > > 
> > > 同上，你试试吧，好使给我说一下，嘿嘿
> > 
> > 仍然无法更新，谢谢！
> 
> 我这边改完之后 第三方的更新比原来快了 ABP一直是更新正常的 
> 你可以浏览器访问下
> https://easylist-downloads.adblockplus.org/easylistchina+easylist.txt
> 看看是否正常打开 
> 或者将easylist-downloads.adblockplus.org加入SSR代理名单

浏览器可正常打开，就是高级设置里面点更新，一直没反应，条目数为0

liyuhaolol, 2020-04-06:
> > > > > @liyuhaolol 请问是改那两个文件，怎么改，谢谢！
> > > > 
> > > > 
> > > > 同上，你试试吧，好使给我说一下，嘿嘿
> > > 
> > > 
> > > 仍然无法更新，谢谢！
> > 
> > 
> > 我这边改完之后 第三方的更新比原来快了 ABP一直是更新正常的
> > 你可以浏览器访问下
> > https://easylist-downloads.adblockplus.org/easylistchina+easylist.txt
> > 看看是否正常打开
> > 或者将easylist-downloads.adblockplus.org加入SSR代理名单
> 
> 浏览器可正常打开，就是高级设置里面点更新，一直没反应，条目数为0

你把这个命令在控制台打一下，看看控制台会不会报显示ipv6，然后出现timeout
wget-ssl --no-check-certificate -O - 'https://easylist-downloads.adblockplus.org/easylistchina+easylist.txt' > /tmp/adnew.conf
这是默认的命令，如果出现了ipv6，timeout那就是ipv6的问题，你再输入这个命令，强制使用ipv4看看
wget-ssl -4 --no-check-certificate -O - 'https://easylist-downloads.adblockplus.org/easylistchina+easylist.txt' > /tmp/adnew.conf，现在控制台是不是所有的解析都变成ipv4了，还会不会出现timeout了？

hplee0120, 2020-04-06:
> > 自从添加第三方规则之后的版本，那个Adblock Plus 规则自动更新一直都是失效的。然后这个可以自定义时间也挺好的，之前也有人建议过。然后第三方规则自动更新据大雕所说是防止有人添加过后然后自动更新的时候炸了。
> 
> 确实第三方更新的时候可能会炸 所以感觉要是能有个复选框 自动更新的时候也更新第三方规则的复选框就好了 炸了的话就取消掉 不炸的话就用这种 另外 abp的那个规则 我用着也没问题 你可以看看update的log 看看什么地址 加到ssr试一下

手动更新一直都是没问题的，我指的失效说的是“自动更新”失效，也就是每天6点不会自动更新。为什么这么多人没看明白，有的还特意“我手动更新了一下没问题”

hplee0120, 2020-04-06:
> [image]
> 特意更新了下，还真没失效，失效是你自己的问题了

看清楚说的“自动更新”几个字好么，还“特意更新了下”，你不特意更新。过多久它还是停留在你上次手动更新的规则数

hplee0120, 2020-04-06:
> > 自从添加第三方规则之后的版本，那个Adblock Plus 规则自动更新一直都是失效的。然后这个可以自定义时间也挺好的，之前也有人建议过。然后第三方规则自动更新据大雕所说是防止有人添加过后然后自动更新的时候炸了。
> 
> adbplus block更新不了，是因为ipv6的问题，要么按照L大的建议，lede编译的时候不编译ipv6，或者你会改文件的话，改2个文件，让adbyby强制使用ipv4，就百分之百可以正常更新了，我已经改了，问题解决。

手动点击更新我这也一直没问题阿，我指的是那个6点自动更新失效。每天不自动更新规则。

liyuhaolol, 2020-04-06:
> > > 自从添加第三方规则之后的版本，那个Adblock Plus 规则自动更新一直都是失效的。然后这个可以自定义时间也挺好的，之前也有人建议过。然后第三方规则自动更新据大雕所说是防止有人添加过后然后自动更新的时候炸了。
> > 
> > 
> > adbplus block更新不了，是因为ipv6的问题，要么按照L大的建议，lede编译的时候不编译ipv6，或者你会改文件的话，改2个文件，让adbyby强制使用ipv4，就百分之百可以正常更新了，我已经改了，问题解决。
> 
> 手动点击更新我这也一直没问题阿，我指的是那个6点自动更新失效。每天不自动更新规则。

你通过什么东西判断自动更新是失败的呢？你并没有拿出什么实际证明，你就感觉失败了，就失败了？

hplee0120, 2020-04-06:
> 你通过什么东西判断自动更新是失败的呢？你并没有拿出什么实际证明，你就感觉失败了，就失败了？

你可以尝试下每天不手动去点那个更新规则按钮，然后观察过几天看看它条数有没有变动，是不是停留在你上次手动更新的状态，或者明天6点过后手动去点击更新看下有没有刷新出新规则，比如你6点05分的时候去更新一下有变动，那大概可不可以判断它6点的时候没有自动更新。又或者你今天没手动更新过的话，你可以去计划任务把那个6点的任务调整一下到现在就近的时间让它这个计划任务执行一下然后去看看规则有没有更新再尝试手动点击一下看会不会有不一样的结果。反正我这边遇到的情况就是大雕重写这个插件之后更新和编译过很多遍都是发现失效的，也在群里问过大雕他也没有否认 就说近期不想去动这个插件。

AwesomeSky, 2020-04-06:
> 你可以尝试下每天不手动去点那个更新规则按钮，然后观察过几天看看它条数有没有变动，是不是停留在你上次手动更新的状态，或者明天6点过后手动去点击更新看下有没有刷新出新规则，比如你6点05分的时候去更新一下有变动，那大概可不可以判断它6点的时候没有自动更新。又或者你今天没手动更新过的话，你可以去计划任务把那个6点的任务调整一下到现在就近的时间让它这个计划任务执行一下然后去看看规则有没有更新再尝试手动点击 [...]

确实是没有自动更新成功 这个我是可以确定的  因为每天看的/tmp下的更新日志就是只更新首页的普通规则和视频规则 ABP的规则貌似会检查 但是LUCI显示的条目数量不会更新

> 你通过什么东西判断自动更新是失败的呢？你并没有拿出什么实际证明，你就感觉失败了，就失败了？

确实是没有自动更新成功 这个我是可以确定的  因为每天看的/tmp下的更新日志就是只更新首页的普通规则和视频规则 ABP的规则貌似会检查 但是LUCI显示的条目数量不会更新

liyuhaolol, 2020-04-06:
> 确实是没有自动更新成功 这个我是可以确定的 因为每天看的/tmp下的更新日志就是只更新首页的普通规则和视频规则 ABP的规则貌似会检查 但是LUCI显示的条目数量不会更新

我看代码了，自动更新ABP的代码，就执行了下载，往后全都没有，我修复一下，很容易

liyuhaolol, 2020-04-07:
> 确实是没有自动更新成功 这个我是可以确定的 因为每天看的/tmp下的更新日志就是只更新首页的普通规则和视频规则 ABP的规则貌似会检查 但是LUCI显示的条目数量不会更新

说到做到，改完了，发来给你
修改/usr/share/adbyby/adblock.sh
全文复制就完了
`#!/bin/sh

newcount=0
oldcount=0
echo 开始下载规则
wget-ssl -4 --no-check-certificate -O - 'https://easylist-downloads.adblockplus.org/easylistchina+easylist.txt' > /tmp/adnew.conf
if [ -s "/tmp/adnew.conf" ];then
  /usr/share/adbyby/ad-update
  if [ -s "/tmp/ad.conf" ];then
  newcount=`cat /tmp/ad.conf | wc -l`
  echo 新规则数量为:$newcount
		if [ $newcount -ge 0 ];then
			if [ -s "/usr/share/adbyby/dnsmasq.adblock" ];then
				oldcount=`cat /usr/share/adbyby/dnsmasq.adblock | wc -l`
			else
				oldcount=0
			fi
			if [ $newcount -ne $oldcount ];then
				echo 开始更新规则
				cp -f /tmp/ad.conf /usr/share/adbyby/dnsmasq.adblock
				cp -f /tmp/ad.conf /tmp/etc/dnsmasq-adbyby.d/adblock
				sleep 10
				echo 重启adbyby
				/etc/init.d/adbyby restart
			else
			echo 当前已是最新规则
			fi
		fi
	fi
  rm -f /tmp/ad.conf
  echo 更新结束
fi`
好不好用，你就一句话，完事了

liyuhaolol, 2020-04-07:
> > > > > 自从添加第三方规则之后的版本，那个Adblock Plus 规则自动更新一直都是失效的。然后这个可以自定义时间也挺好的，之前也有人建议过。然后第三方规则自动更新据大雕所说是防止有人添加过后然后自动更新的时候炸了。
> > > > 
> > > > 
> > > > adbplus block更新不了，是因为ipv6的问题，要么按照L大的建议，lede编译的时候不编译ipv6，或者你会改文件的话，改2个文件，让adbyby强制使用ipv4，就百分之百可以正常更新了，我已经改了，问题解决。
> > > 
> > > 
> > > 手动点击更新我这也一直没问题阿，我指的是那个6点自动更新失效。每天不自动更新规则。
> > 
> > 
> > 你通过什么东西判断自动更新是失败的呢？你并没有拿出什么实际证明，你就感觉失败了，就失败了？
> 
> 你可以尝试下每天不手动去点那个更新规则按钮，然后观察过几天看看它条数有没有变动，是不是停留在你上次手动更新的状态，或者明天6点过后手动去点击更新看下有没有刷新出新规则，比如你6点05分的时候去更新一下有变动，那大概可不可以判断它6点的时候没有自动更新。又或者你今天没手动更新过的话，你可以去计划任务把那个6点的任务调整一下到现在就近的时间让它这个计划任务执行一下然后去看看规则有没有更新再尝试手动点击一下看会不会有不一样的结果。反正我这边遇到的情况就是大雕重写这个插件之后更新和编译过很多遍都是发现失效的，也在群里问过大雕他也没有否认 就说近期不想去动这个插件。还有别的地方也看见有人反映过同样的问题，还有很多人把网络问题的手动更新失败和自动更新失效混为一谈

我给你改完了，你怎么说

AwesomeSky, 2020-04-07:
> 说到做到，改完了，发来给你
> 修改/usr/share/adbyby/adblock.sh
> 全文复制就完了
>
> ```
> #!/bin/sh
>
> newcount=0
> oldcount=0
> echo 开始下载规则
> wget-ssl -4 --no-check-certificate -O - [...]

感谢 找时间测试一下 建议pr给大雕看看

swxk521, 2020-04-08:
> 感谢 找时间测试一下 建议pr给大雕看看

感谢大佬，但是大佬忘了加上   rm -f /tmp/adbyby.updated
没这个会导致adbyby自带规则不会更新

hplee0120, 2020-04-08:
> 感谢大佬，但是大佬忘了加上 rm -f /tmp/adbyby.updated
> 没这个会导致adbyby自带规则不会更新

加在 sleep 10
	echo 重启adbyby
上一行吗

swxk521, 2020-04-08:
我也不太懂照着大佬的又简单改了下，将就用了
#!/bin/sh

newcount=0
oldcount=0
echo 开始下载规则
wget-ssl --no-check-certificate -O - 'https://easylist-downloads.adblockplus.org/easylistchina+easylist.txt' > /tmp/adnew.conf
if [ -s "/tmp/adnew.conf" ];then
  /usr/share/adbyby/ad-update
  if [ -s "/tmp/ad.conf" ];then
  newcount=`cat /tmp/ad.conf | wc -l`
  echo 新规则数量为:$newcount
		if [ $newcount -ge 0 ];then
			if [ -s "/usr/share/adbyby/dnsmasq.adblock" ];then
				oldcount=`cat /usr/share/adbyby/dnsmasq.adblock | wc -l`
			else
				oldcount=0
			fi
			if [ $newcount -ne $oldcount ];then
				echo 开始更新规则
				cp -f /tmp/ad.conf /usr/share/adbyby/dnsmasq.adblock
				cp -f /tmp/ad.conf /tmp/etc/dnsmasq-adbyby.d/adblock
			else
			echo 当前已是最新规则
			fi
		fi
	fi
  rm -f /tmp/adbyby.updated
  sleep 10
  echo 重启adbyby
  /etc/init.d/adbyby restart
  rm -f /tmp/ad.conf
  echo 更新结束
fi

swxk521, 2020-04-08:
#!/bin/sh

newcount=0
oldcount=0
echo 开始下载规则
wget-ssl --no-check-certificate -O - 'https://easylist-downloads.adblockplus.org/easylistchina+easylist.txt' > /tmp/adnew.conf
if [ -s "/tmp/adnew.conf" ];then
  /usr/share/adbyby/ad-update
  if [ -s "/tmp/ad.conf" ];then
  newcount=`cat /tmp/ad.conf | wc -l`
  echo 新规则数量为:$newcount
		if [ $newcount -ge 0 ];then
			if [ -s "/usr/share/adbyby/dnsmasq.adblock" ];then
				oldcount=`cat /usr/share/adbyby/dnsmasq.adblock | wc -l`
			else
				oldcount=0
			fi
			if [ $newcount -ne $oldcount ];then
				echo 开始更新规则
				cp -f /tmp/ad.conf /usr/share/adbyby/dnsmasq.adblock
				cp -f /tmp/ad.conf /tmp/etc/dnsmasq-adbyby.d/adblock
			else
			echo 当前已是最新规则
			fi
		fi
	fi
  rm -f /tmp/adbyby.updated
  sleep 10
  echo 重启adbyby
  /etc/init.d/adbyby restart
  rm -f /tmp/ad.conf
  echo 更新结束
fi

liyuhaolol, 2020-04-09:
> 感谢大佬，但是大佬忘了加上 rm -f /tmp/adbyby.updated
> 没这个会导致adbyby自带规则不会更新

1，你说的自带规则，是第一个页面的video rules和lazy rules么？那个倒是真的没注意会被这个文件影响，我就试了试abp规则更新是正常的我就发出来了。
2，adbyby.updated这个文件我个人感觉，只是用来控制页面显示最后更新时间的，并没有发现其他任何作用。所以你说的这个问题，是有什么代码为证，还是你自己试出来的？

swxk521, 2020-04-09:
> 1，你说的自带规则，是第一个页面的video rules和lazy rules么？那个倒是真的没注意会被这个文件影响，我就试了试abp规则更新是正常的我就发出来了。
> 2，adbyby.updated这个文件我个人感觉，只是用来控制页面显示最后更新时间的，并没有发现其他任何作用。所以你说的这个问题，是有什么代码为证，还是你自己试出来的？

加上可以更新的，这是大雕代码
https://github.com/coolsnowwolf/lede/blob/master/package/lean/luci-app-adbyby-plus/root/usr/share/adbyby/adbybyupdate.sh

liyuhaolol, 2020-04-09:
> > > > > > > > > > > 自从添加第三方规则之后的版本，那个Adblock Plus 规则自动更新一直都是失效的。然后这个可以自定义时间也挺好的，之前也有人建议过。然后第三方规则自动更新据大雕所说是防止有人添加过后然后自动更新的时候炸了。
> > > > > > > > > > 
> > > > > > > > > > 
> > > > > > > > > > adbplus block更新不了，是因为ipv6的问题，要么按照L大的建议，lede编译的时候不编译ipv6，或者你会改文件的话，改2个文件，让adbyby强制使用ipv4，就百分之百可以正常更新了，我已经改了，问题解决。
> > > > > > > > > 
> > > > > > > > > 
> > > > > > > > > 手动点击更新我这也一直没问题阿，我指的是那个6点自动更新失效。每天不自动更新规则。
> > > > > > > > 
> > > > > > > > 
> > > > > > > > 你通过什么东西判断自动更新是失败的呢？你并没有拿出什么实际证明，你就感觉失败了，就失败了？
> > > > > > > 
> > > > > > > 
> > > > > > > 你可以尝试下每天不手动去点那个更新规则按钮，然后观察过几天看看它条数有没有变动，是不是停留在你上次手动更新的状态，或者明天6点过后手动去点击更新看下有没有刷新出新规则，比如你6点05分的时候去更新一下有变动，那大概可不可以判断它6点的时候没有自动更新。又或者你今天没手动更新过的话，你可以去计划任务把那个6点的任务调整一下到现在就近的时间让它这个计划任务执行一下然后去看看规则有没有更新再尝试手动点击一下看会不会有不一样的结果。反正我这边遇到的情况就是大雕重写这个插件之后更新和编译过很多遍都是发现失效的，也在群里问过大雕他也没有否认 就说近期不想去动这个插件。
> > > > > > 
> > > > > > 
> > > > > > 确实是没有自动更新成功 这个我是可以确定的 因为每天看的/tmp下的更新日志就是只更新首页的普通规则和视频规则 ABP的规则貌似会检查 但是LUCI显示的条目数量不会更新
> > > > > 
> > > > > 
> > > > > 说到做到，改完了，发来给你
> > > > > 修改/usr/share/adbyby/adblock.sh
> > > > > 全文复制就完了
> > > > > ```
> > > > > #!/bin/sh
> > > > > 
> > > > > newcount=0
> > > > > oldcount=0
> > > > > echo 开始下载规则
> > > > > wget-ssl -4 --no-check-certificate -O - 'https://easylist-downloads.adblockplus.org/easylistchina+easylist.txt' > /tmp/adnew.conf
> > > > > if [ -s "/tmp/adnew.conf" ];then
> > > > >   /usr/share/adbyby/ad-update
> > > > >   if [ -s "/tmp/ad.conf" ];then
> > > > >   newcount=`cat /tmp/ad.conf | wc -l`
> > > > >   echo 新规则数量为:$newcount
> > > > > 		if [ $newcount -ge 0 ];then
> > > > > 			if [ -s "/usr/share/adbyby/dnsmasq.adblock" ];then
> > > > > 				oldcount=`cat /usr/share/adbyby/dnsmasq.adblock | wc -l`
> > > > > 			else
> > > > > 				oldcount=0
> > > > > 			fi
> > > > > 			if [ $newcount -ne $oldcount ];then
> > > > > 				echo 开始更新规则
> > > > > 				cp -f /tmp/ad.conf /usr/share/adbyby/dnsmasq.adblock
> > > > > 				cp -f /tmp/ad.conf /tmp/etc/dnsmasq-adbyby.d/adblock
> > > > > 				sleep 10
> > > > > 				echo 重启adbyby
> > > > > 				/etc/init.d/adbyby restart
> > > > > 			else
> > > > > 			echo 当前已是最新规则
> > > > > 			fi
> > > > > 		fi
> > > > > 	fi
> > > > >   rm -f /tmp/ad.conf
> > > > >   echo 更新结束
> > > > > fi
> > > > > ```
> > > > > 
> > > > > 
> > > > > 好不好用，你就一句话，完事了
> > > > 
> > > > 
> > > > 感谢 找时间测试一下 建议pr给大雕看看
> > > 
> > > 
> > > 感谢大佬，但是大佬忘了加上 rm -f /tmp/adbyby.updated
> > > 没这个会导致adbyby自带规则不会更新
> > 
> > 
> > 1，你说的自带规则，是第一个页面的video rules和lazy rules么？那个倒是真的没注意会被这个文件影响，我就试了试abp规则更新是正常的我就发出来了。
> > 2，adbyby.updated这个文件我个人感觉，只是用来控制页面显示最后更新时间的，并没有发现其他任何作用。所以你说的这个问题，是有什么代码为证，还是你自己试出来的？
> 
> 加上可以更新的，并不是只更新时间。这是大雕代码
> https://github.com/coolsnowwolf/lede/blob/master/package/lean/luci-app-adbyby-plus/root/usr/share/adbyby/adbybyupdate.sh

兄弟，你后来那个东西是画蛇添足，根本没有就是错的，adbyby.updated是用来负责控制显示最后一次更新时间。不论从lua还是你发的L大的adbybyupdate.sh来看，adbyby.updated都不是一个需要被删除的文件，根本就不在判断逻辑中出现，只是一个记录时间的文件。L大的adblock.sh里面的代码本身就是错的，所以你按照一个错的逻辑里的代码，强行rm -rf 删除adbyby.updated的作用无非就是让你的界面显示1970-01-01而已。所以我给的解决方案是没有问题的。你要不再看看吧

swxk521, 2020-04-09:
> 兄弟，你后来那个东西是画蛇添足，根本没有就是错的，adbyby.updated是用来负责控制显示最后一次更新时间。不论从lua还是你发的L大的adbybyupdate.sh来看，adbyby.updated都不是一个需要被删除的文件，根本就不在判断逻辑中出现，只是一个记录时间的文件。L大的adblock.sh里面的代码本身就是错的，所以你按照一个错的逻辑里的代码，强行rm -rf [...]

也许吧，但是我用着没问题。

snakwu, 2020-04-16:
> 说到做到，改完了，发来给你
> 修改/usr/share/adbyby/adblock.sh
> 全文复制就完了
>
> ```
> #!/bin/sh
>
> newcount=0
> oldcount=0
> echo 开始下载规则
> wget-ssl -4 --no-check-certificate -O - [...]

有没有办法,可以更新adblock规则同时,也更新自己订阅的第三方规则?

closed 2020-07-04 by coolsnowwolf