coolsnowwolf/lede #4192: glib-2.58.3 编译失败

fanck0605, 2020-04-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：把 luci-app-samba 替换成 luci-app-sifsd 后，glib-2.58.3 编译出错 `/staging_dir/host/': Permission denied`

### (2) 路由器型号和固件版本

A：移植添加的 nanopi-r2s，2020-04-05 最新源码，这好像和平台无关啊。actions 试了两次都是这里报错。

### (3) 详细日志

A：
```
2020-04-05T17:26:53.5760668Z make[6]: Entering directory '/home/runner/work/friendlywrt-nanopi_r2s/friendlywrt-nanopi_r2s/friendlywrt-rk3328/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/glib-2.58.3/gio'
2020-04-05T17:26:53.5774206Z   GEN      xdp-dbus.c
2020-04-05T17:26:53.6460160Z   GEN      xdp-dbus.h
2020-04-05T17:26:53.6891997Z cp gnextstepsettingsbackend.m gnextstepsettingsbackend.c
2020-04-05T17:26:53.6916305Z cp gosxcontenttype.m gosxcontenttype.c
2020-04-05T17:26:53.6939577Z cp gosxappinfo.m gosxappinfo.c
2020-04-05T17:26:53.6968864Z cp gcocoanotificationbackend.m gcocoanotificationbackend.c
2020-04-05T17:26:53.7026639Z   GEN      gconstructor_as_data.h
2020-04-05T17:26:53.7294283Z   GEN      gioenumtypes.h
2020-04-05T17:26:53.7303340Z /usr/bin/env: '/home/runner/work/friendlywrt-nanopi_r2s/friendlywrt-nanopi_r2s/friendlywrt-rk3328/friendlywrt/staging_dir/host/': Permission denied
2020-04-05T17:26:53.7307051Z make[6]: *** [gioenumtypes.h] Error 126
2020-04-05T17:26:53.7307678Z Makefile:5044: recipe for target 'gioenumtypes.h' failed
2020-04-05T17:26:53.7310855Z make[6]: Leaving directory '/home/runner/work/friendlywrt-nanopi_r2s/friendlywrt-nanopi_r2s/friendlywrt-rk3328/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/glib-2.58.3/gio'
2020-04-05T17:26:53.7312008Z make[5]: *** [all-recursive] Error 1
2020-04-05T17:26:53.7312305Z Makefile:1259: recipe for target 'all-recursive' failed
2020-04-05T17:26:53.7313124Z make[5]: Leaving directory '/home/runner/work/friendlywrt-nanopi_r2s/friendlywrt-nanopi_r2s/friendlywrt-rk3328/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/glib-2.58.3'
2020-04-05T17:26:53.7314112Z make[4]: *** [all] Error 2
2020-04-05T17:26:53.7314550Z Makefile:886: recipe for target 'all' failed
2020-04-05T17:26:53.7315333Z make[4]: Leaving directory '/home/runner/work/friendlywrt-nanopi_r2s/friendlywrt-nanopi_r2s/friendlywrt-rk3328/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/glib-2.58.3'
2020-04-05T17:26:53.7316565Z make[3]: *** [/home/runner/work/friendlywrt-nanopi_r2s/friendlywrt-nanopi_r2s/friendlywrt-rk3328/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/glib-2.58.3/.built] Error 2
2020-04-05T17:26:53.7317031Z Makefile:123: recipe for target '/home/runner/work/friendlywrt-nanopi_r2s/friendlywrt-nanopi_r2s/friendlywrt-rk3328/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/glib-2.58.3/.built' failed
2020-04-05T17:26:53.7318070Z make[3]: Leaving directory '/home/runner/work/friendlywrt-nanopi_r2s/friendlywrt-nanopi_r2s/friendlywrt-rk3328/friendlywrt/feeds/packages/libs/glib2'
2020-04-05T17:26:53.7327463Z make[2]: *** [package/feeds/packages/glib2/compile] Error 2
2020-04-05T17:26:53.7327623Z make[2]: *** Waiting for unfinished jobs....
2020-04-05T17:26:53.7327730Z time: package/feeds/packages/glib2/compile#79.60#13.34#134.38
```

fanck0605, 2020-04-06:
附上完整日志:
[10_Build FriendlyWrt.zip](https://github.com/coolsnowwolf/lede/files/4435653/10_Build.FriendlyWrt.zip)

Promix953, 2020-04-06:
1.看了官方的issue，路径超长应该是No such file or directory，你的错误是Permission denied
2.你项目路径也太深了：/home/runner/work/friendlywrt-nanopi_r2s/friendlywrt-nanopi_r2s/friendlywrt-rk3328/friendlywrt
3.自己移植的包的问题让固件修bug？

fanck0605, 2020-04-06:
1. 原本应该是打开一个文件，长度不够了，所以给的路径正好是个目录，你以打开文件的方式打开目录，所以 Permission denied
官方的issue，路径长度不太正好，连个目录都不是，所以 No such file or directory
2. 你会发现有个个共同的特性,
"/home/runner/work/friendlywrt-nanopi_r2s/friendlywrt-nanopi_r2s/friendlywrt-rk3328/friendlywrt/staging_dir/host/".length == 112
”/home/daniel/Build/openwrt-sdk/cshoredwrt-sdk-19.01-1-brcm2708-bcm2708_gcc-7.4.0_musl_eabi.Linux-x86_64/staging_“.length == 112
3. 我特地把项目名从 friendlywrt-nanopi_r2s -> nanopi_r2s 节约的 24 个字，编译成功!
[openwrt for nanopi r2s](https://github.com/fanck0605/nanopi_r2s)

Promix953, 2020-04-06:
1.ok，我没细看，你这正好卡在目录上
2.长度的确有限制 
>Limited to 127 chars on 99.9% of systems due to kernel compile time buffer limit.
>It's limited in the kernel by , set in .`BINPRM_BUF_SIZE include/linux/binfmts.h`
```
/usr/bin/env: '/home/runner/work/friendlywrt-nanopi_r2s/friendlywrt-nanopi_r2s/friendlywrt-rk3328/friendlywrt/staging_dir/host/'
```
长度的确超了,所以不是玄学修bug
3.最简单的解决方法就是不要放到`/home/runner/work/friendlywrt-nanopi_r2s/friendlywrt-nanopi_r2s/friendlywrt-rk3328/friendlywrt`这么深的目录里面，光这个目录长度就有95字符
4.的确不能说是移植rk3228导致的报错，放到任何超长路径下应该都能复现问题，但也不能说和你的相关操作毫无关系吧

coolsnowwolf, 2020-04-07:
其实就是目录太长了

fanck0605, 2020-04-08:
可以解除目录长度限制吗🤣

Promix953, 2020-04-09:
> 可以解除目录长度限制吗🤣

It's limited in the kernel by , set in .BINPRM_BUF_SIZE include/linux/binfmts.h
编译机系统的linux内核需要改参数编译

fanck0605, 2020-04-11:
> > 可以解除目录长度限制吗🤣
> 
> It's limited in the kernel by , set in .BINPRM_BUF_SIZE include/linux/binfmts.h
> 编译机系统的linux内核需要改参数编译

行吧。谢谢了。

closed 2020-04-11 by fanck0605