coolsnowwolf/lede #4196: 困扰多天小白求助，安装docker-ce以后无法启动

panthole, 2020-04-06:
十分抱歉，但我研究好几天实在搞不定这个问题，我用Actions里面3月25日的包安装了双软路由，然后安装了docker-ce_19.03.3-2_x86_64.ipk这个版本的docker，dockerman也安装了可是docker一直无法启动。我使用dockerd这个命令，一直报如下的错误，求助好心人，这个应该怎么解，网上实在没搜到教程对docker。

DEBU[2020-04-06T08:16:55.462516391Z] received signal                               signal=terminated
failed to start daemon: Error initializing network controller: Error creating default "bridge" network: Failed to program NAT chain: Failed to inject DOCKER in PREROUTING chain: iptables failed: iptables --wait -t nat -A PREROUTING -m addrtype --dst-type LOCAL -j DOCKER: iptables v1.8.4 (legacy): Couldn't load match `addrtype':No such file or directory

docker version
Client:
 Version:           19.03.3
 API version:       1.40
 Go version:        go1.12.1
 Git commit:        09e60feea48388455cb573d80deaa92b53cc3ffc
 Built:             Sat Oct 12 19:48:10 2019
 OS/Arch:           linux/amd64
 Experimental:      false

docker info
Client:
 Debug Mode: false

Server:
ERROR: Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?
errors pretty printing info

coolsnowwolf, 2020-04-07:
内核没有配置docker支持 不是安装ipk就行的

panthole, 2020-04-07:
明白了，谢谢大大

closed 2020-04-07 by coolsnowwolf