coolsnowwolf/lede #4203: 全新的 OpenWrt PCHiFi 数字转盘遥控 已经正式发布，给喜欢欣赏美妙音乐的你~

coolsnowwolf, 2020-04-06:
PCHiFi 数字转盘遥控是一个基于 DAAP (iTunes Remote), MPD (Music Player Daemon) 和 RSP (Roku) media server 的服务器端播放器

后端输出支持 路由器本机声卡输出 /Airplay/ Chromecast

这个播放器在 OpenWrt 里面运行，可以通过 iPhone/Andriod/Web 界面遥控输出到你的音箱/ Ariplay 设备例如 Homepod， 和 Google Chromecast 盒子种

本地音频支持 MP3/AAC/FLAC/APE/ALAC ，支持 32bit / 352Khz 高清音频文件解码（无损音乐需要你下载后放入 路由器里面, 或者将 NAS 的共享目录挂载过来)

给喜欢欣赏音乐的你而做

Enjoy~~

[image]

[image]

[image]

guochan2019, 2020-04-06:
支持L大新功能，正在跑码中

hcym, 2020-04-06:
😂谢谢啦

a13547, 2020-04-06:
@coolsnowwolf 按我自己编译的情况，编译 forked-daapd-27.1 时还需要 libunistring-dev libconfuse-dev libmxml-dev libsqlite3-dev libevent-dev libjson-c-dev libgcrypt-dev libantlr3c-dev libavformat-dev libavcodec-dev libavutil-dev libavfilter-dev libavahi-client-dev libplist-dev libsodium-dev libwebsockets-dev 这些依赖。另外不知道怎么回事，Music Remote Center编译出来没有汉化。

pacinoa, 2020-04-07:
和minidlna冲突，导致编译失败。

shu0802, 2020-04-07:
> 支持L大新功能，正在跑码中

如何选择，PCHiFi才能编译出来，求指导

u0225, 2020-04-07:
昨晚更新源码后编译的x86_64系统，数字转盘和NFS在系统启动项里一直是禁用状态，点启用，启动，重启后一刷新页面就又变禁用了。
数字转盘里选上启用后也是一直的未运行状态。
不知道是和什么冲突了，还是其他什么原因？

zhg-em, 2020-04-07:
为什么我在编译菜单中找不到 luci-app-music-remote-center 呢？
x86系统

shu0802, 2020-04-07:
> 为什么我在编译菜单中找不到luci-app-music-remote-center呢？
> x86系统

要重头编译

coolsnowwolf, 2020-04-07:
启动不了的留意下有没有对应的文件夹

shu0802, 2020-04-07:
> 启动不了的留意下有没有对应的文件夹

怎么查看，我的就是启用不了
[image]

ybist, 2020-04-07:
启动不了有一种情况，可能跟mpd有关，如果有mpd-avahi-service，mpd-full或mpd-mini，停止或删除试试。

u0225, 2020-04-07:
> 启动不了的留意下有没有对应的文件夹

在/opt目录里没有数据库文件songs.db，也没有music这个目录，新建一个music目录也不行。
不知道是不是和没有songs.db这个数据库文件有关？

coolsnowwolf, 2020-04-07:
那你的版本太老了

shu0802, 2020-04-08:
> 那你的版本太老了

编译时是不是只选择luci-app-music-remote-center就可以了。重头编译出来还是没有你说的那个opt文件夹，也没有你图片中的“打开WEB界面”。
[image]。是不是还差什么没有选择上。请指点一下，谢谢

u0225, 2020-04-08:
> 那你的版本太老了

昨天晚上更新的源码还是一样，数字转盘和nfs都还是一直禁用状态，转盘也还是未运行状态。。。

shu0802, 2020-04-08:
> > 那你的版本太老了
> 
> 昨天晚上更新的源码还是一样，数字转盘和nfs都还是一直禁用状态，转盘也还是未运行状态。。。

你的有opt这个文件夹吗？我的没有这个文件夹。

u0225, 2020-04-08:
> > > 那你的版本太老了
> > 
> > 
> > 昨天晚上更新的源码还是一样，数字转盘和nfs都还是一直禁用状态，转盘也还是未运行状态。。。
> 
> 你的有opt这个文件夹吗？我的没有这个文件夹。也没有“打开WEB管理界面”。
> [image]

有opt文件夹（有可能是docker建的这个文件夹），但是没有数据库文件

coolsnowwolf, 2020-04-08:
你有 opt 但没有 music 文件夹，也是起不来的

guochan2019, 2020-04-08:
第一时间编译，wifi音箱播放正常，遗憾的是x86软路由自带的输出启用不了，不知有解决办法无？

cnzd, 2020-04-08:
airplay电视播放正常，iPhone本机播放没有声音

u0225, 2020-04-08:
> 你有 opt 但没有 music 文件夹，也是起不来的

暂时不折腾了。
另 建议L大把 可道云 从x86_32固件插件列表中移除，因为32位固件选上可道云编译的话会报php7错误而编译失败（试了很多次了，都是php7错误而编译失败，64位的就没问题）。

coolsnowwolf, 2020-04-08:
> airplay电视播放正常，iPhone本机播放没有声音

这个是播放器的服务端，除非你的 iPhone 安装了接收器才有声音

agutools, 2020-04-09:
一直：OpenWrt+mpd-full+upmpdcli+USB声卡/音箱=7×24家庭背景音乐，客户端也不少——现在又有新折腾了😘

jinjin327, 2020-04-09:
求助L大打开web管理界面提示Forbidden

xiaoyongdong, 2020-04-09:
@jinjin327 你的IP不是192.168网段吧，没在信任列表里。

coolsnowwolf, 2020-04-09:
> 求助L大打开web管理界面403提示Forbidden

默认只能支持 192.168 网段，后续加入 LAN IP 网段自定义白名单吧

dhlsam, 2020-04-10:
希望大神能加入 LAN IP 网段自定义白名单，10.10.10.X的内网ip 出现403Forbidden

jinjin327, 2020-04-10:
> > 求助L大打开web管理界面403提示Forbidden
> 
> 默认只能支持 192.168 网段，后续加入 LAN IP 网段自定义白名单吧

好的，谢谢L大

hue715, 2020-04-10:
> 希望大神能加入 LAN IP 网段自定义白名单，10.10.10.X的内网ip 出现403Forbidden

临时处理，可以修改/etc/init.d/forked-daapd文件，增加“trusted_networks"
[image]

xpc0938, 2020-04-10:
感谢L大，已测试，airplay到音箱播放正常。

aaaisan, 2020-04-11:
只能用于x86吗?树莓派4无效,自己mkdir了opt和music目录

monsterzzzz, 2020-04-11:
> Request

同样问题

andywx2008, 2020-04-12:
感谢L大，这个功能很是期待，赶紧重新编译来体验下

xayane, 2020-04-13:
> > 希望大神能加入 LAN IP 网段自定义白名单，10.10.10.X的内网ip 出现403Forbidden
> 
> 临时处理，可以修改/etc/init.d/forked-daapd文件，增加“trusted_networks"
> [image]

感谢，正想问10段ip怎么直接就是forbidden的。

u0225, 2020-04-13:
可以打开和播放了，但是软路由本地没声音。
Local Audio Output 选项打不开，可能是没有声卡驱动的原因

andywx2008, 2020-04-15:
编译总是出错，不知道是否需要安装其他依赖，全局代理了。请L大指导：
make[1]: *** [/home/vk-op/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/vk-op/lede'
/home/vk-op/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

u0225, 2020-04-22:
HTTP STREAM (stream.mp3) 电脑手机浏览器里都可以正常播放。
把声卡的驱动都选上后编译，Local Audio Output 可以选上了，但是软路由本地还是没有声音。

aepub, 2020-04-28:
编译进固件后从来没有启动成功过，后来删除了，难道是这两个文件夹没有建立的原因？这个播放器是实现跟volumio或者moodeaudio一样的功能吗？这样的话可以搞一台斐讯N1，做成数播了！

Yrp, 2020-05-08:
启动不起来的可以看看日志，路径
> /var/log/forked-daapd.log

我的报错是
> [  LOG]    httpd: Could not stat() web root directory '/usr/share/forked-daapd/htdocs': No such file or directory

创建这个目录就可以启动了

LiyongLuo, 2020-05-10:
安卓端用什么软件搭配呢，我用了几个mpd客户端，发现都连不上

hue715, 2020-05-10:
> 安卓端用什么软件搭配呢，我用了几个mpd客户端，发现都连不上

可以用Retune
[image]

imkage, 2020-05-13:
netdata 装不上,
opkg update 之后
opkg install netdata 提示 package size mismatch
无法继续。

Yrp, 2020-05-13:
> netdata 装不上，
> opkg update 之后
> opkg install netdata 提示 package size mismatch
> 无法继续。

换成北京外国语大学的源试试，用这个源可以解决mismatch
> https://mirrors.bfsu.edu.cn/openwrt/

btw，问错issues了

imkage, 2020-05-13:
> > netdata 装不上，
> > opkg update 之后
> > opkg install netdata 提示 package size mismatch
> > 无法继续。
> 
> 换成北京外国语大学的源试试，用这个源可以解决mismatch
> 
> > https://mirrors.bfsu.edu.cn/openwrt/
> 
> btw，问错issues了

哈哈，我看这里热闹，借个人气。非常感谢，已经解决了～！

closed 2020-07-04 by coolsnowwolf

chenbaker, 2020-11-21:
遥控端，显示乱码，怎样搞？

soulxiangmu, 2020-11-25:
已经扫描出歌曲了，用什么客户端使用呢

Mixyr, 2021-02-12:
> 第一时间编译，wifi音箱播放正常，遗憾的是x86软路由自带的输出启用不了，不知有解决办法无？
> 在设置里选择Local Audio Output时，会提示错误：
> Request failed (status: 500 Internal Server Error, url: http://192.168.50.1:3689/api/outputs/0/toggle)

请教您连的是哪一类wifi音箱，我加的小米AI音箱连不上，谢谢

VergilGao, 2021-07-23:
关于这个库，后排给有条件的玩家安利一下linuxserver.io维护的这个软件的docker镜像：
https://hub.docker.com/r/linuxserver/daapd
这个docker版的优势是版本更新更频繁，而且docker部署方便快捷，更适合多设备玩家在非路由设备上部署相应服务。
附上我目前的部署配置：
```
version: "2.1"
services:
  daapd:
    image: linuxserver/daapd
    container_name: owntone
    environment:
      - PUID=1000
      - PGID=29
      - TZ=Asia/Shanghai
    volumes:
      - /disk/data1/codebase/docker/owntone/config:/config
      - /disk/data1/media/music:/music
    devices:
      - /dev/snd
    network_mode: host
    restart: unless-stopped
```
缺点是你仍然需要自己手动编辑配置文件，这也是linux系软件的通病了，相信在这个仓库遇到的大家都是喜欢折腾的人，配置文件的注释写的也很清晰，应该不是阻碍。

jsdfhasuh, 2022-02-13:
打开就闪退

VergilGao, 2022-10-26:
我正在尝试给webui添加简体中文翻译
https://github.com/owntone/owntone-server/pull/1548
目前正在施工，如果有错误或者更好的建议可以提出来

fansijou, 2022-11-25:
> 

请问如何挂载NAS的目录给music？