coolsnowwolf/lede #4206: ssr-plus,v2ray http伪装不能用

smallprogram, 2020-04-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
ssr-plus,v2ray http伪装不能用
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
当设置v2ray代理，http伪装，即使勾选Instance也不起作用。这个问题我看之前有很多人提过，但是都没有解决。
A：

### (2) 路由器型号和固件版本
x86软路由，使用最新源码编译的固件
A：

### (3) 详细日志

A：

coolsnowwolf, 2020-04-07:
就这一句话，没有日志，没有节点信息，你觉得不关等什么呢？

smallprogram, 2020-04-07:
> 就这一句话，没有日志，没有节点信息，你觉得不关等什么呢？

#4102 有日志有细节不一样也关了。算了，大佬忙着，叨扰了。

coolsnowwolf, 2020-04-07:
从来没人提交过vmess url

9l, 2020-04-07:
那算啥日志

smallprogram, 2020-04-08:
> 那算啥日志
> 
> PS: 现在还有人用v2的http伪装？

有人用啊。用的人多了。V2ray的使用方式有很多。你不能说用法小众就不支持对吧。

9l, 2020-04-08:
怪不得那么多人说自己用v2然后vps废了，原来是还有这么多人在用http伪装

aiden-5154, 2020-04-12:
还有这个问题比如这个没有套用http混淆的纯tcp节点vmess://eyJhZGQiOiJpdHVuZXMuYXBwbGUuY29tIiwiaG9zdCI6IiIsImlkIjoiQzUwMjU4RjAtREY3Ni04M0Q4LTIyQkItMzk0QjIxMkEzMzE5IiwibmV0IjoidGNwIiwicGF0aCI6IiIsInBvcnQiOiIxMDAwMSIsInBzIjoiSEtULUEiLCJ0bHMiOiIiLCJ2IjoyLCJhaWQiOjAsInR5cGUiOiJhdXRvIn0=
会给自动加上http混淆，base64解码后把hosts内容去掉一样会自动选上http混淆，hosts栏显示undefined如;eyJhZGQiOiJpdHVuZXMuYXBwbGUuY29tIiwiaWQiOiJDNTAyNThGMC1ERjc2LTgzRDgtMjJCQi0zOTRCMjEyQTMzMTkiLCJuZXQiOiJ0Y3AiLCJwYXRoIjoiIiwicG9ydCI6IjEwMDAxIiwicHMiOiJIS1QtQSIsInRscyI6IiIsInYiOjIsImFpZCI6MCwidHlwZSI6ImF1dG8ifQ==

aiden-5154, 2020-04-12:
把http混淆改为none以后etc/config/shadowsocksr文件中关于该节点会多一行	option tcp_guise 'none'，而选择http混淆时则没有option tcp_guise这一行

closed 2020-04-29 by smallprogram