coolsnowwolf/lede #4208: 加入数字转盘遥控器后依赖报错

bin20088, 2020-04-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：加入数字转盘遥控器后依赖报错

### (2) 路由器型号和固件版本

A：X64  N1

### (3) 详细日志

A：
[报错.txt](https://github.com/coolsnowwolf/lede/files/4441982/default.txt)

ybist, 2020-04-07:
大概去lib里去掉libffmpeg-audio-dec就行，就在libffmpeg-full上面。
你可能选了luci-app-minidlna？
先选luci-app-music-remote-center，后选luci-app-minidlna应该就没这个问题了

bin20088, 2020-04-07:
> 大概去的lib里去掉libffmpeg音频-DEC就行，就在libffmpeg全上面。
> 你可能选了荧光素酶-APP-minidlna？
> 先选的luci-APP-音乐远程中心，后选的luci-APP-minidlna应该就没这个问题了

如果我想保留libffmpeg-audio-dec，删除minidlna的话，除了取消选择minidlna外，还应取消哪个依赖呢

ybist, 2020-04-07:
我猜libffmpeg-full 应该包含了libffmpeg-audio-dec。libffmpeg-audio-dec不用单独选了。

bin20088, 2020-04-07:
> 我猜libffmpeg-full 应该包含了libffmpeg-audio-dec。libffmpeg-audio-dec不用单独选了。都选上就冲突了。

明白，多谢。

bin20088, 2020-04-07:
> 我猜libffmpeg-full 应该包含了libffmpeg-audio-dec。libffmpeg-audio-dec不用单独选了。都选上就冲突了。

确实按你的方法解决了  感谢

closed 2020-04-07 by bin20088

bin20088, 2020-04-11:
> 我猜libffmpeg-full 应该包含了libffmpeg-audio-dec。libffmpeg-audio-dec不用单独选了。都选上就冲突了。

大哥，你的办法可以编译成功 但是会导致开不了机，无限重启