coolsnowwolf/lede #4210: 編譯出現錯誤，求問大神原因

bdgsfs, 2020-04-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：問題同標題

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：checking for gperf... no
configure: error: GNU gperf required, please install it.
Makefile:82: recipe for target '/home/mcy/lede/build_dir/target-x86_64_musl/forked-daapd-27.1/.configured_a17fb5ef857664f03cd0ce37cc5ea591' failed
make[3]: *** [/home/mcy/lede/build_dir/target-x86_64_musl/forked-daapd-27.1/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 1
make[3]: Leaving directory '/home/mcy/lede/feeds/packages/sound/forked-daapd'
time: package/feeds/packages/forked-daapd/compile#24.85#3.25#54.61
package/Makefile:111: recipe for target 'package/feeds/packages/forked-daapd/compile' failed
make[2]: *** [package/feeds/packages/forked-daapd/compile] Error 2
make[2]: Leaving directory '/home/mcy/lede'
package/Makefile:107: recipe for target '/home/mcy/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/mcy/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mcy/lede'
Build failed - please re-run with -j1 to see the real error message
/home/mcy/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 1

ybist, 2020-04-07:
这个问题不要麻烦大神了，小白我猜一下，可能是下面这个原因

A：checking for gperf... no
configure: error: GNU gperf required, please install it.

编译命令如下:
首先装好 Ubuntu 64bit，推荐 Ubuntu 18 LTS x64

命令行输入 sudo apt-get update ，然后输入 sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf

bdgsfs, 2020-04-07:
> sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core [...]

好的我試試，謝謝大神

pirately, 2020-04-07:
我也是出错了，用的云编译

lx521603, 2020-04-08:
> 我也是出错了，用的云编译

云编译也是要改里面的依赖的，依赖改了。

pirately, 2020-04-08:
> > 我也是出错了，用的云编译
> 
> 云编译也是要改里面的依赖的，依赖改了。

刚看到，大雕也更新了workflow。谢谢。

lx521603, 2020-04-08:
> > > 我也是出错了，用的云编译
> > 
> > 
> > 云编译也是要改里面的依赖的，依赖改了。
> 
> 刚看到，大雕也更新了workflow。谢谢。

我的是能编译成功，但解压后只有53M，能启动，但所有配置文件都是空的，df -h看了下是/overlay 容量只有几百K，不知道哪里没搞对。

closed 2020-04-08 by coolsnowwolf