coolsnowwolf/lede #4212: 请问一下螃蟹8111H网卡不插网线网口识别为10mb 半双工的原因.

amosly, 2020-04-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：开机后,如果网口插拔过网线,不插网线时网口被识别为10mb/s 半双工.

### (2) 路由器型号和固件版本

A：试过各种版本,网卡型号是8111H

-----------------------------
我想请问一下这个是驱动问题吗?

我搜索的时候发现有人说螃蟹网卡大多都是通用的驱动.

我尝试自己编译,没有找到8111系列网卡只有8169,查看网卡型号发现安装的也是8169的驱动

(能用是能用,就是这个10mb问题一直困扰我,所以发上来麻烦各位大佬解答一下)

coolsnowwolf, 2020-04-07:
本来就是啊

dolphinpaopao, 2020-04-07:
不插网线降为半双工有什么问题吗？

amosly, 2020-04-07:
> 本来就是啊

正常不是应该显示为 - 半双工吗? 

emmm 我试过其他的,就显示为 - 半双工.用ethtool 查询网口的话,会显示为 Unknown.

但是这个8111H就是10mb/s 半双工.

amosly, 2020-04-07:
> 不插网线降为半双工有什么问题吗？

半双工没问题,问题在于 10mb/s,好吧如果大佬们觉得没问题那我也就不纠结了这个了.

但是还是想知道为什么跟其他的有差别.详细可以看楼上.

dolphinpaopao, 2020-04-07:
> > 不插网线降为半双工有什么问题吗？
> 
> 半双工没问题,问题在于 10mb/s,好吧如果大佬们觉得没问题那我也就不纠结了这个了.
> 
> 但是还是想知道为什么跟其他的有差别.详细可以看楼上.

不同的芯片和驱动设计不同，现代化的芯片在空闲状态工作正常的情况下一般会通过降频降速或者关闭来达到节能省电的目的。

closed 2020-04-08 by coolsnowwolf