coolsnowwolf/lede #4218: 最新源码QCA9561+QCA9886无线驱动或者内核问题

keke1023, 2020-04-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：TPLINK WDR6800 V2，硬件等同于WDR6500 V6，主控QCA9561，5G QCA9886，8M spi+64M ddr2，最新源码里内核为4.9.215，9888的驱动是三天前commit的ath10k-ct: bump to 5.4 (#4165)

### (3) 详细日志

A：编译出来的固件刷入可以正常启动，但是一旦需要更改任何设置尤其是无线设置，负载将急剧上升至死机重启，占用最多的进程的是kworker/0:0，最终我采用了早期的内核4.9.182加上2019-05-12-7b4aff22-1的ath10k-firmware-qca9888编译出来，可以正常使用，期间测试过切换到4.14内核，或者单独降级ath10k-firmware为上一次提交的，都没有解决问题，不清楚到底问题出在内核还是无线驱动上

AmadeusGhost, 2020-04-07:
改用`kmod-ath10k-ct-smallbuffers`试试

keke1023, 2020-04-08:
> 改用`kmod-ath10k-ct-smallbuffers`试试

确实取消选中ath10k换成smallbuffers就没问题了，请问这个是什么原因呢？

AmadeusGhost, 2020-04-08:
> 确实取消选中ath10k换成smallbuffers就没问题了，请问这个是什么原因呢？

你机子内存太小了，需要用小内存版本。

keke1023, 2020-04-08:
> > 确实取消选中ath10k换成smallbuffers就没问题了，请问这个是什么原因呢？
> 
> 你机子内存太小了，需要用小内存版本。

明白啦～看来64m确实已经到了淘汰边缘了哈哈哈

closed 2020-04-08 by coolsnowwolf