coolsnowwolf/lede #4221: ACRH17编译更新4月2日以后版本，无线5G疯狂自动掉线

cindzzic, 2020-04-07:
1、ACRH17编译更新4月2日以后版本，无线5G疯狂自动掉线，2.4G不稳定，PC端偶尔ping值1000左右。
Q：是否搜索了issue (使用 "x" 选择)
 [x ] 没有类似的issue（已搜索，无类似issue）
 2、详细叙述
 (1) 具体问题
无线5G疯狂掉线，2.4G不稳定，PC端偶尔ping值1000左右。
 (2) 路由器型号和固件版本
ACRH17，OpenWrt R20.3.19 / LuCI Master (git-20.087.30210-56fe7ae) 
 (3) 详细日志
[系统日志.txt](https://github.com/coolsnowwolf/lede/files/4443880/default.txt)
[内核日志.txt](https://github.com/coolsnowwolf/lede/files/4443881/default.txt)

jackberry, 2020-04-07:
我确实也遇到同样的问题，你编译了airplay2吗？

cindzzic, 2020-04-07:
> 
> 
> 我确实也遇到同样的问题，你编译了airplay2吗？

没有；大雕默认配置，增加了aria2，nps，baidupcs

coolsnowwolf, 2020-04-07:
[image]

OP 官方无线驱动爆炸了。。。

cindzzic, 2020-04-07:
> 
> 
> [image]
> 
> OP 官方无线驱动爆炸了。。。

雕兄，这咋解决？可以在配置文件里选用那个文件名为ath10k-ct的无线驱动不？

cindzzic, 2020-04-07:
> 
> 
> 我确实也遇到同样的问题，你编译了airplay2吗？

我刚刚选用了这个ath10k-ct驱动编译了下，貌似暂时没问题了
[image]

closed 2020-04-07 by coolsnowwolf