coolsnowwolf/lede #4222: 编译固件最后打包的一步出错

lon91ong, 2020-04-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译自用的极路由B70硬改的SPI闪存适用版固件出错
参考了硬件规格类似的Youku-L2和newifi-D1的配置文件，拼凑了个[配置文件](https://github.com/lon91ong/lede/blob/master/target/linux/ramips/dts/mt7621_hiwifi_hc5962_spi.dts)
编译到最后打包的一步出错

### (2) 路由器型号和固件版本

A：极路由B70，硬改16M SPI闪存，W25Q128，拉取最新(2020.4.7)源码编译

### (3) 详细日志

A：
``` bash
sysupgrade-hiwifi_hc5962_spi/
8868864 bytes (8.9 MB, 8.5 MiB) copied, 0.0228074 s, 389 MB/s
sysupgrade-hiwifi_hc5962_spi/CONTROL
tar: Option --mtime: Treating date '@1586248271' as 2020-04-07 08:31:11
sysupgrade-hiwifi_hc5962_spi/kernel
3736+1 records in
3736+1 records out
1913139 bytes (1.9 MB, 1.8 MiB) copied, 0.00881063 s, 217 MB/s
dd: invalid number: ''
make[5]: *** [/home/runner/work/OpenWRT_B70_SPI_16M/OpenWRT_B70_SPI_16M/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-hiwifi_hc5962_spi-squashfs-factory.bin] Error 1
make[4]: *** [install] Error 2
make[3]: *** [install] Error 2
make[2]: *** [target/linux/install] Error 2
make[1]: *** [/home/runner/work/OpenWRT_B70_SPI_16M/OpenWRT_B70_SPI_16M/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
sysupgrade-hiwifi_hc5962_spi/root
make: *** [world] Error 2
echo '{  "metadata_version": "1.0", "supported_devices":["hiwifi,hc5962,spi"], "version": { "dist": "OpenWrt", "version": "SNAPSHOT", "revision": "r0-683634c", "target": "ramips/mt7621", "board": "hiwifi_hc5962_spi" } }' | fwtool -I - /home/runner/work/OpenWRT_B70_SPI_16M/OpenWRT_B70_SPI_16M/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-hiwifi_hc5962_spi-squashfs-sysupgrade.bin
........
Makefile:201: recipe for target '/home/runner/work/OpenWRT_B70_SPI_16M/OpenWRT_B70_SPI_16M/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-hiwifi_hc5962_spi-squashfs-factory.bin' failed
make[5]: Leaving directory '/home/runner/work/OpenWRT_B70_SPI_16M/OpenWRT_B70_SPI_16M/openwrt/target/linux/ramips/image'
Makefile:26: recipe for target 'install' failed
make[4]: Leaving directory '/home/runner/work/OpenWRT_B70_SPI_16M/OpenWRT_B70_SPI_16M/openwrt/target/linux/ramips'
Makefile:13: recipe for target 'install' failed
make[3]: Leaving directory '/home/runner/work/OpenWRT_B70_SPI_16M/OpenWRT_B70_SPI_16M/openwrt/target/linux'
time: target/linux/install#61.94#12.59#71.93
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: Leaving directory '/home/runner/work/OpenWRT_B70_SPI_16M/OpenWRT_B70_SPI_16M/openwrt'
target/Makefile:19: recipe for target '/home/runner/work/OpenWRT_B70_SPI_16M/OpenWRT_B70_SPI_16M/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install' failed
make[1]: Leaving directory '/home/runner/work/OpenWRT_B70_SPI_16M/OpenWRT_B70_SPI_16M/openwrt'
/home/runner/work/OpenWRT_B70_SPI_16M/OpenWRT_B70_SPI_16M/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
##[error]Process completed with exit code 2.
```

pirately, 2020-04-07:
我也是

incartoon, 2020-04-08:
我也是到这里Fail了

lon91ong, 2020-04-08:
拼凑的mt7621.mk配置有问题，照下面修改后编译通过了
```
define Device/hiwifi_hc5962-spi
  IMAGE_SIZE := 16064k
  DEVICE_VENDOR := HiWiFi
  DEVICE_MODEL := HC5962
  DEVICE_PACKAGES := kmod-mt7603 kmod-mt76x2 kmod-usb3 kmod-sdhci-mt7620 \
	kmod-usb-ledtrig-usbport wpad-openssl
endef
TARGET_DEVICES += hiwifi_hc5962-spi
```

closed 2020-04-08 by lon91ong