coolsnowwolf/lede #4227: 挂载 SMB/CIFS 网络共享文件夹，共享名不能带空格

mjyhj, 2020-04-07:
[image]
[image]
[image]

lizhenglin2002, 2020-04-10:
请教一下这个功能怎样设置

mjyhj, 2020-04-10:
> 请教一下这个功能怎样设置
如图所示啊

图1，是小钢炮的SMB共享。设置好后访问N1的IP 192.168.123.254，不就看到我共享的music文件夹吗？

图2，就是填你路由器去访问那个SMB共享的文件夹的参数IP,共享文件夹名，共享服务端的版本，权限，账户名。  再填上要挂载到路由上的什么路径

图3，成功后就出来了

lizhenglin2002, 2020-04-10:
我想添加电脑的为啥会失败😭

lizhenglin2002, 2020-04-11:
好的，谢谢

发送自 Windows 10 版邮件应用

发件人: mjyhj
发送时间: 2020年4月10日 下午 11:15
收件人: coolsnowwolf/lede
抄送: LEE LIN; Comment
主题: Re: [coolsnowwolf/lede] 挂载 SMB/CIFS 网络共享文件夹，共享名不能带空格 (#4227)

请教一下这个功能怎样设置
如图所示啊
图1，是小钢炮的SMB共享。设置好后访问N1的IP 192.168.123.254，不就看到我共享的music文件夹吗？
图2，就是填你路由器去访问那个SMB共享的文件夹的参数IP,共享文件夹名，共享服务端的版本，权限，账户名。 再填上要挂载到路由上的什么路径
图3，成功后就出来了

nbfhscl, 2020-04-30:
请问这个功能需要配置哪些组件

closed 2020-07-04 by coolsnowwolf

kitkw, 2022-03-26:
挂载win11共享目录不成功............
请我哪里设置不对了（共享目录密码已关闭）？？
[image]
[image]
[image]

ocean521, 2022-04-03:
> > 请教一下这个功能怎样设置
> > 如图所示啊
> 
> 图1，是小钢炮的SMB共享。设置好后访问N1的IP 192.168.123.254，不就看到我共享的music文件夹吗？
> 
> 图2，就是填你路由器去访问那个SMB共享的文件夹的参数IP,共享文件夹名，共享服务端的版本，权限，账户名。 再填上要挂载到路由上的什么路径
> 
> 图3，成功后就出来了

楼主，我的问题是将NAS的共享文件夹挂载到OP,挂载目录设置的/mnt/xxx，并将之作为下载目录。但实际下载时，却下载到了/overlay/upper/mnt/xxx，没下载到共享文件夹。是哪里的问题啊