coolsnowwolf/lede #4229: wireguard连接回路由器，无法科学上网

LockeJet, 2020-04-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：路由器用了ssr-plus+trojan，局域网内的机器都可以透明科学上网，用ocserv和shadowsocks连接回家也可以科学上网。可是用wireguard连接回来则无法上国外网；我在内网主机上开启了dns服务器，wireguard客户端DNS指定为该主机就可以科学上网。另外，路由器上测试时谷歌连接失败，但事实上是可以科学上网的。

### (2) 路由器型号和固件版本

A： 主机：3865U软路由 + PROXMOX PVE; 虚拟机：OpenWrt R20.3.19 / LuCI Master (git-20.079.47035-c242d65) ，内核4.19.108。

### (3) 详细日志

A：
#VPS上
```
admin@aws:~$ sudo journalctl -u trojan -f
-- Logs begin at Sun 2020-04-05 15:24:01 CST. --
Apr 05 19:37:59 aws trojan[11383]: [2020-04-05 19:37:59] [INFO] 127.0.0.1:46362 disconnected, 0 bytes received, 517 bytes sent, lasted for 132 seconds
Apr 05 19:38:01 aws trojan[11383]: [2020-04-05 19:38:01] [ERROR] 127.0.0.1:46366 cannot establish connection to remote server 69.63.181.11:443: Connection timed out
Apr 05 19:38:01 aws trojan[11383]: [2020-04-05 19:38:01] [INFO] 127.0.0.1:46366 disconnected, 0 bytes received, 517 bytes sent, lasted for 131 seconds
Apr 05 19:38:01 aws trojan[11383]: [2020-04-05 19:38:01] [INFO] 127.0.0.1:46270 disconnected, 7657 bytes received, 4352 bytes sent, lasted for 266 seconds
Apr 05 19:38:03 aws trojan[11383]: [2020-04-05 19:38:03] [INFO] 127.0.0.1:46410 authenticated as ${TROJAN_PASSWORD}
Apr 05 19:38:03 aws trojan[11383]: [2020-04-05 19:38:03] [INFO] 127.0.0.1:46410 requested connection to 54.243.132.124:443
Apr 05 19:38:04 aws trojan[11383]: [2020-04-05 19:38:04] [INFO] 127.0.0.1:46416 authenticated as ${TROJAN_PASSWORD}
Apr 05 19:38:04 aws trojan[11383]: [2020-04-05 19:38:04] [INFO] 127.0.0.1:46416 requested connection to 23.21.154.58:443
Apr 05 19:38:05 aws trojan[11383]: [2020-04-05 19:38:05] [INFO] 127.0.0.1:46414 authenticated as ${TROJAN_PASSWORD}
Apr 05 19:38:05 aws trojan[11383]: [2020-04-05 19:38:05] [INFO] 127.0.0.1:46414 requested connection to 23.21.154.58:443
Apr 05 19:38:08 aws trojan[11383]: [2020-04-05 19:38:08] [INFO] 127.0.0.1:46422 authenticated as ${TROJAN_PASSWORD}
Apr 05 19:38:08 aws trojan[11383]: [2020-04-05 19:38:08] [INFO] 127.0.0.1:46422 requested connection to 172.217.160.78:443
Apr 05 19:38:09 aws trojan[11383]: [2020-04-05 19:38:09] [INFO] 127.0.0.1:46426 authenticated as ${TROJAN_PASSWORD}
Apr 05 19:38:09 aws trojan[11383]: [2020-04-05 19:38:09] [INFO] 127.0.0.1:46426 requested connection to 172.217.160.74:443
Apr 05 19:38:12 aws trojan[11383]: [2020-04-05 19:38:12] [INFO] 127.0.0.1:46430 authenticated as ${TROJAN_PASSWORD}
Apr 05 19:38:12 aws trojan[11383]: [2020-04-05 19:38:12] [INFO] 127.0.0.1:46430 requested connection to 69.171.247.71:443
Apr 05 19:38:13 aws trojan[11383]: [2020-04-05 19:38:13] [INFO] 127.0.0.1:46434 authenticated as ${TROJAN_PASSWORD}
Apr 05 19:38:13 aws trojan[11383]: [2020-04-05 19:38:13] [INFO] 127.0.0.1:46434 requested connection to 31.13.72.1:443
Apr 05 19:38:13 aws trojan[11383]: [2020-04-05 19:38:13] [INFO] 127.0.0.1:46434 disconnected, 3148 bytes received, 547 bytes sent, lasted for 1 seconds
Apr 05 19:38:19 aws trojan[11383]: [2020-04-05 19:38:19] [INFO] 127.0.0.1:46438 authenticated as ${TROJAN_PASSWORD}
Apr 05 19:38:19 aws trojan[11383]: [2020-04-05 19:38:19] [INFO] 127.0.0.1:46438 requested connection to 31.13.72.1:443
Apr 05 19:38:20 aws trojan[11383]: [2020-04-05 19:38:20] [INFO] 127.0.0.1:46442 authenticated as ${TROJAN_PASSWORD}
Apr 05 19:38:20 aws trojan[11383]: [2020-04-05 19:38:20] [INFO] 127.0.0.1:46442 requested connection to 31.13.72.1:443
Apr 05 19:38:20 aws trojan[11383]: [2020-04-05 19:38:20] [INFO] 127.0.0.1:46416 disconnected, 5902 bytes received, 11569 bytes sent, lasted for 16 seconds
Apr 05 19:38:20 aws trojan[11383]: [2020-04-05 19:38:20] [INFO] 127.0.0.1:46442 disconnected, 3147 bytes received, 547 bytes sent, lasted for 0 seconds
Apr 05 19:38:20 aws trojan[11383]: [2020-04-05 19:38:20] [INFO] 127.0.0.1:46414 disconnected, 5902 bytes received, 1400 bytes sent, lasted for 16 seconds
Apr 05 19:38:21 aws trojan[11383]: [2020-04-05 19:38:21] [INFO] 127.0.0.1:46438 disconnected, 3149 bytes received, 547 bytes sent, lasted for 2 seconds
Apr 05 19:38:21 aws trojan[11383]: [2020-04-05 19:38:21] [INFO] 127.0.0.1:46446 authenticated as ${TROJAN_PASSWORD}
Apr 05 19:38:21 aws trojan[11383]: [2020-04-05 19:38:21] [INFO] 127.0.0.1:46446 requested connection to 69.171.247.71:443
Apr 05 19:38:23 aws trojan[11383]: [2020-04-05 19:38:23] [INFO] 127.0.0.1:46450 authenticated as ${TROJAN_PASSWORD}
Apr 05 19:38:23 aws trojan[11383]: [2020-04-05 19:38:23] [INFO] 127.0.0.1:46450 requested connection to 31.13.72.1:443
Apr 05 19:38:24 aws trojan[11383]: [2020-04-05 19:38:24] [INFO] 127.0.0.1:46450 disconnected, 3147 bytes received, 547 bytes sent, lasted for 1 seconds
```
# OpenWrt输出 (采用pdnsd)
```
root@OpenWrt:~# logread -f
Sun Apr  5 19:39:38 2020 user.notice shadowsocksr: pdnsd tunnel error.restart!
Sun Apr  5 19:39:38 2020 daemon.err pdnsd[3785]: Could not bind tcp socket: Address in use
Sun Apr  5 19:39:38 2020 daemon.err pdnsd[3785]: Could not bind to udp socket: Address in use
Sun Apr  5 19:39:38 2020 daemon.err pdnsd[3785]: tcp and udp initialization failed. Exiting.
```

# 浏览器：
> NET::ERR_CERT_COMMON_NAME_INVALID

# OpenWrt输出 (采用dns2socks)

```
root@OpenWrt:~# logread -f
Sun Apr  5 19:43:13 2020 user.notice shadowsocksr: dns2socks  tunnel error.restart!
Sun Apr  5 19:43:43 2020 user.notice shadowsocksr: dns2socks 208.67.222.222:53 tunnel error.restart!
```

coolsnowwolf, 2020-04-08:
DNS端口冲突了

LockeJet, 2020-04-08:
我看了一下，没有看到其它有在用53、5335等端口的呀。不知道pdnsd和dns2socks相关的进程用什么端口？
如果DNS端口冲突，那么，下面两个事实就无法解释通了吧？
- 无论是ssr-plus+的客户端使用ss还是v2ray或是trojan，局域网内的电脑是可以的科学上网，即使pdnsd和dns2socks仍然会出现类似的出错信息。
- 若ssr-plus+设置为v2ray，用wireguard连回家里却可以科学上网。

closed 2020-07-04 by coolsnowwolf

xuyqing888, 2023-01-02:
ssr的问题，passwall 没问题