coolsnowwolf/lede #4232: 修复luci-app-syncdial中的单线并发多拨功能的请求

waynon, 2020-04-08:
## 1. 关于你要提交的问题
能不能将在新版OpenWRT中失效的单线并发多拨功能修复？

Q：是否搜索了issue (使用 "x" 选择)
类似的issue:#1821 #1673 #1612 #3768 #3178

## 2. 详细叙述
A：luci-app-syncdial的单线并发多拨在新版本中已经失效，目前有以下问题：
1. **当"当前在线接口数量<最低在线接口数量"时不会重新拨号**
2. 只有开机的时候有约50%的概率并发多拨成功，**在Web页面下点击"重新拨号"无法并发多拨成功**，电信并发多拨失败会获得12.x.x.x, 13.x.x.x或14.x.x.x的IP地址，获得这些地址会出现彩虹通道提示拨号超过允许的拨号数。

## 3. 我的一些疑问
我看到L大之前对这个问题的解释是"这个功能的 API 在新版本不存在了"（见 #1821）。

API不存在了但是通过修改pppd的源码来配合luci应该还是能够提供单线并发多拨功能的。我想问的是L大**没有解决这个问题是因为解决这个问题的工作量太大还是有什么其他的难处吗**？

这个功能的issue有好几个了(#1821 #1673 #1612 #3768 #3178)，我觉得单线并发多拨还是有用的，或者有没有人能提供一个解决这个问题的思路。

waynon, 2020-04-08:
L大应该是修改了https://github.com/coolsnowwolf/lede/tree/master/package/network/services/shellsync
https://github.com/coolsnowwolf/lede/tree/master/package/network/services/ppp
来提供并发多拨
[(参考)](https://github.com/Lienol/openwrt/issues/19)

waynon, 2020-04-08:
基于OpenWRT的潘多拉(PandoraBox)固件在某个版本后也删去了luci-app-syncdial中的"最小在线数量"和“断线重连”的功能选项
[参考](https://www.right.com.cn/forum/thread-259950-1-1.html)

waynon, 2020-04-08:
OpenWRT 15.05 Chaos Calmer并发多拨的源码修改方法
http://www.openwrt.org.cn/bbs/thread-45559-1-1.html

hxlls, 2020-04-08:
大概不支持的原因应该是几个，
第一，没有环境测试，所以很多东西靠想象
第二，多拨对于ssrplus 会产生很多兼容性的问题，需要很多精力去维护
第三，很多软件都不是基于多拨来开发的，所以要修改，单一来解决不现实，没这么多精力和实际，人总要恰饭，要等上游去解决
第四，相对ssr plus 来说，多拨的受益群体太小了，2选一，肯定都知道选什么，作为一个开发者不能用个别人的需求来代表大众，这个是最根本的，这个是在平时聊天断断续续集合起来的信息，加上猜测。

coolsnowwolf, 2020-04-08:
原因其实很简单。。。我一次多拨都没成功过。。。你们等大神吧

Promix953, 2020-04-08:
> 原因其实很简单。。。我一次多拨都没成功过。。。你们等大神吧

许多地方已经完全封禁多播了

rshi0212, 2020-04-08:
> 原因其实很简单。。。我一次多拨都没成功过。。。你们等大神吧

我现在是一条联通100M和一条移动200M，联通100M不支持多拨，移动200M支持，用MACVLAN虚拟了3个接移动线路的WAN口，成功WAN(20%)/WANB(20%)/WANC(20%)/WAND(20%)/WANE(20%)用MWAN3负载均衡，并且使用ssr+完全没问题。

cnzd, 2020-04-09:
我的情况是可以多拨，但是一个小时之后就会掉线到只剩一个线路，怀疑是运营商限制，干脆就不折腾了。

waynon, 2020-04-09:
我这里是可以多拨的，使用**高恪固件**或**ROS**是可以完美实现多拨功能的。

我这里多拨如果成功会获得118.x.x.x的IP地址，如果不成功会获得12.x.x.x, 13.x.x.x或14.x.x.x的IP地址，获得这样的IP地址会出现运营商的彩虹通道页面，提示超出拨号数限制。

我这里**当且仅当拨号数>=3时上行速度翻一倍，下行速度不叠加**(但是仅双拨时上行下行都不会叠加)。但是拨号数不管为多大，上行速度都只能翻一倍，而不能更大。

waynon, 2020-04-09:
> 许多地方已经完全封禁多播了

很多地方其实还是可以**并发**多拨的，注意，是并发多拨，而不是普通的多拨。

关于并发多拨的原理可以看一下这个：http://koolshare.cn/thread-90619-1-1.html

waynon, 2020-04-09:
> 我的情况是可以多拨，但是一个小时之后就会掉线到只剩一个线路，怀疑是运营商限制，干脆就不折腾了。

这个就是我说的luci-app-syncdial中存在问题，你可以试一下高恪固件，**高恪固件的并发多拨功能是没有任何问题的**，当一个接口掉线它会重新拨号。

老版本的OpenWRT以及基于OpenWRT的PandoraBox也会在一个接口掉线后自动重新拨号，来确保始终有**你指定的拨号数**个的接口在线。

但是在某一个OpenWRT版本中这个功能失败了，这也是我提出这个issue的原因

waynon, 2020-04-09:
> 大概不支持的原因应该是几个，
> 第一，没有环境测试，所以很多东西靠想象
> 第二，多拨对于ssrplus 会产生很多兼容性的问题，需要很多精力去维护
> 第三，很多软件都不是基于多拨来开发的，所以要修改，单一来解决不现实，没这么多精力和实际，人总要恰饭，要等上游去解决
> 第四，相对ssr plus 来说，多拨的受益群体太小了，2选一，肯定都知道选什么，作为一个开发者不能用个别人的需求来代表大众，这个是最根本的，这个是在平时聊天断断续续集合起来的信息，加上猜测。

我答复一下：
1. 测试环境是需要一个使用PPPoE来拨号的宽带，而且这个宽带要支持并发多拨，还应满足普通多拨会失败的条件。
2. 我在使用OpenWRT多拨成功后使用ssrplus没有遇到任何兼容性问题，楼上也有人说了没有问题。
3. 这个功能以前是可以使用的，只是在一个OpenWRT版本更新后失效了，工程量应该不会很大。
4. 如上述第2点，没有遇到兼容性问题，不存在二选一的情况。

假设确实有与多拨不兼容的插件，也应该提供这个功能，由用户选择是否禁用。多拨这个功能还是一个比较常见的路由器功能，希望OpenWRT支持

高恪固件是完美支持并发多拨相关功能的，不清楚多拨的可以使用高恪的并发多拨。高恪可以提供完全的并发多拨支持，OpenWRT应该也可以

coolsnowwolf, 2020-04-09:
所以楼上有环境的你们，谁来开发呢？。。。

rshi0212, 2020-04-09:
> 所以楼上有环境的你们，谁来开发呢？。。。

哈哈哈哈哈哈哈哈，一针见血。。。。

Promix953, 2020-04-09:
> > 许多地方已经完全封禁多播了
> 
> 很多地方其实还是可以**并发**多拨的，注意，是**并发多拨**，而不是普通的多拨。
> 
> 关于并发多拨的原理以及适用性可以看一下这个：http://koolshare.cn/thread-90619-1-1.html

我知道并发多播，也早就试过了，彩虹通道加定时检测加连续拨号后一定时间完全无法拨号了解一下

waynon, 2020-04-10:
> > > 许多地方已经完全封禁多播了
> > 
> > 
> > 很多地方其实还是可以**并发**多拨的，注意，是**并发多拨**，而不是普通的多拨。
> > 关于并发多拨的原理以及适用性可以看一下这个：http://koolshare.cn/thread-90619-1-1.html
> 
> 我知道并发多播，也早就试过了，彩虹通道加定时检测加连续拨号后一定时间完全无法拨号了解一下

高恪是没有任问题的，它在尝试多拨失败时应该会给PPPoE认证服务器发送一个结束PPPoE请求的报文（PADT），这样就没有你说的那个问题了。

closed 2020-04-10 by waynon

wingswing1, 2023-03-31:
所以这个问题现在也没有人解决吗？？

BenjaminX, 2024-07-17:
修改usr/lib/lua/luci/model/cbi/syncdial.lua

#第三行

local e="mwan3 status | grep -c \"is online and tracking is active\""

#修改为
local e="mwan3 status | grep -c \"is online\""​