coolsnowwolf/lede #4236: DOCKER 下怎么安装Openwrt？

lianzhaowen, 2020-04-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

dolphinpaopao, 2020-04-08:
https://www.right.com.cn/forum/

martin0607, 2020-04-09:
下载镜像-》安装

lianzhaowen, 2020-04-09:
> 
> 
> 下载镜像-》安装

有没有做好的，可以pull？x86的，docker hub 里面是arm64

martin0607, 2020-04-09:
docker pull raymondwong/openwrt_r9

lianzhaowen, 2020-04-09:
> 
> 
> docker pull raymondwong/openwrt_r9

我看到这个了，就是嫌它有点旧，哈

我去试试

martin0607, 2020-04-09:
那就自己编译好x86的再转docker img的
https://zorz.cc/post/n1-compile-openwrt-docker-image.html

lianzhaowen, 2020-04-09:
raymondwong/openwrt_r9  刚好有 20.1.24-x86_64 版本
但是有新问题，我是在qnap docker 运行的

Failed to resize receive buffer: Operation not permitted                                                                                    
/etc/preinit: line 6: can't create /sys/devices/system/cpu/microcode/reload: Read-only file system                                          
RTNETLINK answers: Operation not permitted                                                                                                  
RTNETLINK answers: Operation not permitted                                                                                                  
Press the [f] key and hit [enter] to enter failsafe mode                                                                                    
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level                                                                
Failed to send flush request: Operation not permitted                                                                                       
RTNETLINK answers: Operation not permitted   

提示这个，不知道怎么解决

closed 2020-07-04 by coolsnowwolf