coolsnowwolf/lede #4244: [R20.4.8]LAN接口设置DNS为局域网自建DNS服务后会造成大量垃圾请求

rhettor, 2020-04-09:
系统：ESXi 6.7u3
CPU：Intel(R) Core(TM) i3-7100U CPU @ 2.40GHz : 1 Core 2 Thread
固件版本：OpenWrt R20.4.8 / LuCI Master (git-20.088.29068-cebe120)
场景：lede作为网关（10.10.10.252），局域网DNS服务（10.10.10.240）

当LAN接口设置自定义DNS为局域网自建DNS服务后会造成大量垃圾请求，dnsmasq的CPU使用率至少在20%以上，甚至会达到99%，但是在自定义DNS设置为公共DNS则不会出现相应问题。

截图1：
[dnsmasq CPU使用率](https://i.loli.net/2020/04/09/9oSEGlhyzpfsHYq.png)
截图2:
[dns请求记录](https://i.loli.net/2020/04/09/BXhYpeV4mqfDcNd.png)
截图3:
[dns请求域名](https://i.loli.net/2020/04/09/FCfKjYnXRJVIc1z.png)

Promix953, 2020-04-09:
```
|-->lede防火墙将dns请求拦截转发到本机53端口，本机将dns请求发到240，240发送dns请求-->|
^                                                                              v
|--<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<--|
```

coolsnowwolf, 2020-04-10:
因为不是这么玩的，自环了

rhettor, 2020-04-11:
@coolsnowwolf 但是240并没把252设置为网关，他们的网关都是主路由251

1049066154, 2020-04-11:
你是不是在240机器上设置了上游dns为10.10.10.240?

rhettor, 2020-04-11:
@1049066154 上游是公共dns

1049066154, 2020-04-11:
> @1049066154 上游是公共dns
你网络中负责DHCP的那台机器中配置的下发DNS是哪个IP?

1049066154, 2020-04-11:
> @1049066154 上游是公共dns

你描述一下你网络中各个设备的连接配置吧

rhettor, 2020-04-12:
> > @1049066154 上游是公共dns
> 
> 你描述一下你网络中各个设备的连接配置吧

[image]

Promix953, 2020-04-12:
[https://www.google.com/search?q=lb._dns-sd._udp](url)

1049066154, 2020-04-12:
> > > @1049066154 上游是公共dns
> > 
> > 
> > 你描述一下你网络中各个设备的连接配置吧
> 
> [image]

[image]
你这里不能设置127.0.0.1,上游服务器是自己的ip这样就死循环了,而且你的dns程序应该没有完全接管53端口,导致dnsmasq这类自带的服务会使用接口上配置的127.0.0.1,但是接受到请求的还是dnsmasq,然后就陷入死循环

hue715, 2020-04-19:
> > > @1049066154 上游是公共dns
> > 
> > 
> > 你描述一下你网络中各个设备的连接配置吧
> 
> [image]

试试把Lede防火墙规则中的iptables重定向注释掉
[image]

closed 2020-07-04 by coolsnowwolf