coolsnowwolf/lede #4246: 请问什么时候会升级luci19

lloyd18, 2020-04-09:
会升级吗？

hcym, 2020-04-09:
这个有什么意思，蹄子硬才是王道

coolsnowwolf, 2020-04-10:
luci19 的好处是？

lloyd18, 2020-04-11:
> luci19 的好处是？

具体有什么好处我也不懂，不过既然官方会用肯定有一定的道理，现在很多插件开发者都跟进适配了，比如：
https://github.com/brvphoenix/luci-app-wrtbwmon
https://github.com/lisaac/luci-app-dockerman/issues/29
https://github.com/pymumu/smartdns
https://github.com/jerrykuku/luci-theme-argon
要同时兼容两个luci版本经常会遇到一些麻烦，要是统一跟官方我相信会给开发者解决不少麻烦

9l, 2020-04-11:
需要重写所有的插件代码哦

tty228, 2020-04-14:
我就用的 luci19 ，目前来说小 bug 有点多，而且老插件会遇到很多问题
如果切换 luci19 目测 issues 要爆炸

最大的优点也就 JavaScript 不占用路由器资源，概况页面不用担心 cpu占用狂飙，速度上并没有感觉很优秀
还有就是概况页可以直接禁用设备 WiFi 连接，但是其实很少用

bug 的话，目前遇到的有这些
1.chrome 浏览器顶部红字错误
https://user-images.githubusercontent.com/33397881/76483627-7d452200-6452-11ea-909e-3e85255e22db.png
详见：https://github.com/openwrt/luci/issues/3747

2.部分插件网页按钮无法启动

3.部分页面翻译错误：
  (1)无线页面无连接识别为禁用，修复于 https://github.com/openwrt/luci/commit/b572c6d82eed487b9e48b7537921d967787c83f1
19.07.2 未修复
[image]
  (2)软件包页面关闭识别为取消，暂未看到修复
[image]

closed 2020-04-20 by lloyd18