coolsnowwolf/lede #4255: 竟斗云 2.4g 掉线 STA-OPMODE-N_SS-CHANGED

sososoyoung, 2020-04-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：手机连接 WiFi 后出现大量 `STA-OPMODE-N_SS-CHANGED` 日志，测速只有 10Mbps, (正常可以跑 40~60)
注：2.4g同时连接好几个设备，一台手机，两个只能插排和网关, 目前仅发现手机有这个问题

### (2) 路由器型号和固件版本

A：路由：竟斗云2，固件R20.4.8  ，最后提交 ae7a5d706026c70ff5236eef6d72211cfcac5642 ：luci-app-ssr-plus: OpenWrt SNAPSHOT compatibility (#4154 

### (3) 详细日志

A：
```
Fri Apr 10 00:33:17 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED d8:63:75:a9:54:31
Fri Apr 10 00:33:17 2020 daemon.info hostapd: wlan0: STA d8:63:75:a9:54:31 IEEE 802.11: authenticated
Fri Apr 10 00:33:17 2020 daemon.info hostapd: wlan0: STA d8:63:75:a9:54:31 IEEE 802.11: associated (aid 4)
Fri Apr 10 00:33:17 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED d8:63:75:a9:54:31
Fri Apr 10 00:33:17 2020 daemon.info hostapd: wlan0: STA d8:63:75:a9:54:31 RADIUS: starting accounting session 39DBFF69BB7B50B1
Fri Apr 10 00:33:17 2020 daemon.info hostapd: wlan0: STA d8:63:75:a9:54:31 WPA: pairwise key handshake completed (RSN)
Fri Apr 10 00:33:17 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:17 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:18 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:18 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:18 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:18 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:19 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:19 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:20 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:20 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:20 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:20 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:21 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:21 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:23 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:23 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:23 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:23 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:24 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:24 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:24 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:25 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:26 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:26 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:28 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:28 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:28 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:28 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:29 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:30 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:33 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:33 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:33 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:34 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:34 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:34 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:35 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:35 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:35 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:35 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:36 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:36 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:36 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 1
Fri Apr 10 00:33:37 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED d8:63:75:a9:54:31 2
Fri Apr 10 00:33:38 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED d8:63:75:a9:54:31
Fri Apr 10 00:33:38 2020 daemon.info hostapd: wlan1: STA d8:63:75:a9:54:31 IEEE 802.11: authenticated
Fri Apr 10 00:33:38 2020 daemon.info hostapd: wlan1: STA d8:63:75:a9:54:31 IEEE 802.11: associated (aid 2)
Fri Apr 10 00:33:38 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED d8:63:75:a9:54:31
Fri Apr 10 00:33:38 2020 daemon.info hostapd: wlan1: STA d8:63:75:a9:54:31 RADIUS: starting accounting session D84F6468B6F008B0
Fri Apr 10 00:33:38 2020 daemon.info hostapd: wlan1: STA d8:63:75:a9:54:31 WPA: pairwise key handshake completed (RSN)
Fri Apr 10 00:33:38 2020 daemon.err odhcpd[1426]: Failed to send to ::%wan6@eth0.2 (Permission denied)
```

stock169, 2020-04-10:
见过同样的日志，隔一段时间会出现。

1049066154, 2020-04-10:
小米r3g同样会出现这个问题，等待解决办法

coolsnowwolf, 2020-04-10:
肯固定信道就没这么多事情了

sososoyoung, 2020-04-15:
> 肯固定信道就没这么多事情了

大雕，感觉应该还是 mu-mimo 的问题，换了个小米 6 也是这个问题，后来小米 mix2 从高速模式切换到兼容模式没事了，应该兼容模式禁用mu-mimo了吧，我看链接速率从 144 降到 72Mbps了

1049066154, 2020-04-15:
> > 肯固定信道就没这么多事情了
> 
> 大雕，感觉应该还是 mu-mimo 的问题，换了个小米 6 也是这个问题，后来小米 mix2 从高速模式切换到兼容模式没事了，应该兼容模式禁用mu-mimo了吧，我看链接速率从 144 降到 72Mbps了

感觉不是mu-mimo的问题，我不管开没开都会出现

sososoyoung, 2020-04-15:
> > > 肯固定信道就没这么多事情了
> > 
> > 
> > 大雕，感觉应该还是 mu-mimo 的问题，换了个小米 6 也是这个问题，后来小米 mix2 从高速模式切换到兼容模式没事了，应该兼容模式禁用mu-mimo了吧，我看链接速率从 144 降到 72Mbps了
> 
> 感觉不是mu-mimo的问题，我不管开没开都会出现

切换后速率有变化么？

1049066154, 2020-04-16:
> > > > 肯固定信道就没这么多事情了
> > > 
> > > 
> > > 大雕，感觉应该还是 mu-mimo 的问题，换了个小米 6 也是这个问题，后来小米 mix2 从高速模式切换到兼容模式没事了，应该兼容模式禁用mu-mimo了吧，我看链接速率从 144 降到 72Mbps了
> > 
> > 
> > 感觉不是mu-mimo的问题，我不管开没开都会出现
> 
> 切换后速率有变化么？

没有变化

closed 2020-07-04 by coolsnowwolf