coolsnowwolf/lede #4257: x86编译出错，内核mac80211错误

KnowledegAndAction, 2020-04-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：x86已经成功编译过若干次，昨天看到新增了音乐播放器功能所以尝试重新编译新版。重新git pull了新源码，编译失败。网络环境应该没有问题，路由器全局。

### (2) 路由器型号和固件版本

A：x86 64bit  固件版本为最新

### (3) 详细日志

A：Applying ./patches/subsys/300-mac80211-add-stop-start-logic-for-software-TXQs.patch using plaintext: 
patching file include/net/mac80211.h
Hunk #1 FAILED at 1504.
Hunk #2 succeeded at 1631 with fuzz 2 (offset 103 lines).
1 out of 2 hunks FAILED -- saving rejects to file include/net/mac80211.h.rej
patching file net/mac80211/ieee80211_i.h
Hunk #1 FAILED at 818.
Hunk #2 FAILED at 1226.
Hunk #3 FAILED at 2046.
3 out of 3 hunks FAILED -- saving rejects to file net/mac80211/ieee80211_i.h.rej
patching file net/mac80211/main.c
Hunk #1 succeeded at 710 with fuzz 2 (offset 24 lines).
patching file net/mac80211/tx.c
Hunk #1 FAILED at 3485.
Hunk #2 FAILED at 3586.
2 out of 2 hunks FAILED -- saving rejects to file net/mac80211/tx.c.rej
patching file net/mac80211/util.c
Hunk #1 succeeded at 345 with fuzz 2 (offset 105 lines).
Hunk #2 FAILED at 401.
Hunk #3 succeeded at 443 with fuzz 2 (offset -1 lines).
Hunk #4 FAILED at 456.
2 out of 4 hunks FAILED -- saving rejects to file net/mac80211/util.c.rej
Patch failed!  Please fix ./patches/subsys/300-mac80211-add-stop-start-logic-for-software-TXQs.patch!
Makefile:538: recipe for target '/home/mark/Desktop/x86lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.4.27-1/.prepared_6e4ee8fbe6195d47bddfcfe74c0a1f1e_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: *** [/home/mark/Desktop/x86lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.4.27-1/.prepared_6e4ee8fbe6195d47bddfcfe74c0a1f1e_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/mark/Desktop/x86lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#0.86#0.31#1.11
package/Makefile:111: recipe for target 'package/kernel/mac80211/compile' failed
make[2]: *** [package/kernel/mac80211/compile] Error 2
make[2]: Leaving directory '/home/mark/Desktop/x86lede'
package/Makefile:107: recipe for target '/home/mark/Desktop/x86lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/mark/Desktop/x86lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mark/Desktop/x86lede'
/home/mark/Desktop/x86lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

waynon, 2020-04-10:
看一下这个https://stackoverflow.com/questions/14282617/hunk-1-failed-at-1-whats-that-mean
更新一下feeds然后make download试试？

lyin888, 2020-04-10:
libffmpeg-audio-dec
去掉这个模块

KnowledegAndAction, 2020-04-14:
最后make clean重新编译解决。没有改动配置。

closed 2020-04-14 by KnowledegAndAction