coolsnowwolf/lede #4258: 关于自动切换的一些实际体验和建议（不是bug）没有log

norikohxb, 2020-04-10:
首先感谢lean大的自动切换。自动切换的目的是为了让翻墙服务器不稳定的时候让你切换到另外一个。
我为了所谓的自动切换更顺滑修改了，我本来土土的想，如果路由器cpu资源足够，60秒检查一次不是更好吗？切换检查超时时间缩短为2秒就判定一个vps不能用，切换检查超时时间1次。就自动切换，不是会让网络更顺畅吗。

自动切换检查周期（秒）=60秒
切换检查超时时间（秒）=2
切换检查重试次数=1
结果让整个翻墙网络更不稳定。不稳定的原因我用我的苍白的知识分析。
我的主翻墙服务器是谷歌云的trojan香港节点。而备用是微软云美国节点。
频繁的切换会导致你youtube的app在上60秒的时候还是用香港节点解析出来的youtube的流媒体服务器的ip进行访问，下60秒线路切换到美国节点，变成美国ip，访问之前缓存在手机，平板内的app的流媒体服务器仍然用的是上一个不同地区节点的。发现youtube你在往下刷的时候每个视频的封面打开都没有问题，但是点击进去视频，都是转圈圈。（因为youtube的视频的首页的logo图是图片格式，而youtube的每个图片点击进去是流媒体的格式，两个分属于不同服务器）
感觉实际用下来特别是切换检查超时时间和切换重试次数给我设置少了以后。很容易造成一个现象。
一些老的安卓5.0的平板。（感觉新的安卓1.0的手机不会或者症状不明显）必须禁用wifi，退出youtube的app清理出内存，重新连接wifi，重新启用youtube的app才可以正常访问。

有可能我这个分析还不是很对还希望大神分析一下。
昨天甚至我的台式机，chrome打开youtube的网页端，准备测速几乎打不开，流媒体没有任何流量。或者流量很微弱，最后运行的ipconfig /flushdns才解决问题。

norikohxb, 2020-04-10:
另外有一个疑问，编译出Build EFI GRUB images的固件，和编译出 Build GRUB images 的固件在esxi虚拟机上面使用op会不会有性能上的提升。还是都一样。

hxlls, 2020-04-10:
很简单啊，你设置不切换就行了啊，前提你保证你梯子的稳定，这些东西肯定是建立在一定的稳定性上说的，不稳定，10包掉3包的，自己都判断不了，更别说要机器去判断了

norikohxb, 2020-04-10:
我现在就是不切换，手动的。如果确实不行了察觉了再去更新，只是说当时以为好用，我就去修改了检测时间，结果频繁更换发现最后连那个谷歌连接性检查也都乱了。我只是发帖告诉其他还没有用的人一下，其实就是切换一次意味着你的地理位置也变化了，跟着所有国外域名还是要再重新解析一次。

hxlls, 2020-04-10:
实际上并没有你所说的问题，我从毛子的梯子切换到韩国，在切换到越南，DNS都是跟随设置的，具体表现就是youtube 后面的地址会变化，也就是左上角的标准是跟随变化的,标志后面是跟随着国家，比如ru kr VN等，也就是证明DNS 解析到了对应的国家，你的现象应该是你做了DNS缓存造成的
[image]
[image]

norikohxb, 2020-04-11:
对，我也是认为dns缓存造成的，而且安卓10的手机比如三星s10是不需要网卡禁用再启用，安卓5.0的早期的安卓平板就不行。应该和客户端有关系吧。最新的手机就挺好的。但是我发誓我没有做任何dns缓存，我的要翻墙的手机平板用的哦都市op作为dns。

hxlls, 2020-04-11:
实际上，重启的脚本上就有kill 翻墙的DNS 的命令，也就是你每次重启PDNS 都会重新更新，SSRP 的DNS 不是用dnsmasq 查询的，走的PDNS 或者哪个DNS2SOCKE(大概是这个名称）,所以，和openwrt 自身没关系，这个问题根本就不是openwrt 的问题，更多是你自己设备的问题，具体设备怎么设置的DNS 还有设备的缓存是怎么设置的，那就自己解决咯

closed 2020-04-11 by norikohxb

norikohxb, 2020-04-11:
> 实际上并没有你所说的问题，我从毛子的梯子切换到韩国，在切换到越南，DNS都是跟随设置的，具体表现就是youtube 后面的地址会变化，也就是左上角的标准是跟随变化的,标志后面是跟随着国家，比如ru kr VN等，也就是证明DNS 解析到了对应的国家，你的现象应该是你做了DNS缓存造成的
> [image]
> [image]

能否请教你一个问题，以lede为dns服务器，这个lede（不编译smartdns或者adguardhome进去）就是原生的lede，自己是否有dns缓存？如果有dns缓存他的缓存时间是多长？