coolsnowwolf/lede #4260: 勾选上Ipv6helper后编译失败

Omoinemie, 2020-04-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：勾选上Ipv6helper后编译失败

### (2) 路由器型号和固件版本

A：R3G

### (3) 详细日志

A：https://github.com/Omoinemie/action-lede/commit/cfbdc44ccafe69299e0f36499cca290efb82f138#annotation_162684458

lon91ong, 2020-04-11:
你这个**详细日志**自己看过吗!!!

closed 2020-04-11 by coolsnowwolf

lyin888, 2020-04-11:
Collected errors:
Configuring kmod-mac80211.
 * check_data_file_clashes: Package odhcpd wants to install file /home/runner/work/action-lede/action-lede/openwrt/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/odhcpd
Configuring kmod-mt76-core.
	But that file is already provided by package  * odhcpd-ipv6only
Configuring kmod-mt76x02-common.
 * check_data_file_clashes: Package odhcpd wants to install file /home/runner/work/action-lede/action-lede/openwrt/build_dir/target-mipsel_24kc_musl/root-ramips/etc/uci-defaults/odhcpd.defaults
Configuring kmod-mt76x2-common.
	But that file is already provided by package  * odhcpd-ipv6only
Configuring kmod-ipt-offload.
 * check_data_file_clashes: Package odhcpd wants to install file /home/runner/work/action-lede/action-lede/openwrt/build_dir/target-mipsel_24kc_musl/root-ramips/usr/sbin/odhcpd
Configuring kmod-tcp-bbr.
	But that file is already provided by package  * odhcpd-ipv6only
Configuring luci-app-flowoffload.
 * check_data_file_clashes: Package odhcpd wants to install file /home/runner/work/action-lede/action-lede/openwrt/build_dir/target-mipsel_24kc_musl/root-ramips/usr/sbin/odhcpd-update
Configuring luci-i18n-flowoffload-zh-cn.
	But that file is already provided by package  * odhcpd-ipv6only
Configuring dropbear.
 * opkg_install_cmd: Cannot install package odhcpd.
Configuring luci-theme-argon.
make[2]: *** [package/install] Error 255
Configuring mtd.

日志里面写的明明白白，模块冲突