coolsnowwolf/lede #4262: 编译环境出错

505799690, 2020-04-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：按照README上sudo apt-get -y install build-essen...后make menuconfig报错
scripts/config/mconf: error while loading shared libraries: libncursesw.so.6: cannot open shared object file: No such file or directory
/home/zhangxin/lede/include/toplevel.mk:141: recipe for target 'menuconfig' failed
make: *** [menuconfig] Error 127
ubuntu-16.04.6-server-amd64
ubuntu18 server 也试过了同样报这个错误，新装的系统，新git的项目，之前是不会的，不知道哪次出现了问题

### (3) 详细日志

A：

guyezi, 2020-04-10:
# 查找此文件
locate libncursesw.so.6
# 查看lib库位置
cat /etc/ld.so.conf
cat /etc/ld.so.conf.d/*.conf
# 指定的/usr/local/lib等均为空，只有一个有文件
# 链接至/usr/local/lib仍无效，需退出环境再进入才生效
sudo ln -s /conda/lib/libncursesw.so.6 /usr/local/lib
sudo ln -s /conda/lib/libncursesw.so.6 /usr/lib/x86_64-linux-gnu/

9l, 2020-04-11:
可能是你网络环境不行，搭上梯子update feed

lon91ong, 2020-04-11:
为什么不用Github Action, **除了效率低些**, 其它方面都方便太多了!

closed 2020-07-04 by coolsnowwolf