coolsnowwolf/lede #4263: 路由器翻墙 Android 端telegram,play商店 不稳定分析

z826540272, 2020-04-10:
使用路由器翻墙，安卓手机 谷歌服务和TG经常不稳定，打不开等。
1，我使用 绕过中国大陆ip 模式
2,   旁路由模式
        常见于睡了一觉或者手机待机一段时间后，打开手机无法访问play商店或者无法加载图像。
通过测试得知，dns没有发现解析错误，所以不存在dns污染，单纯连接超时。
非 手机问题，因为从安卓6-10都可以复现。
非 ipv6问题，我手机无ipv6地址。
非 翻墙服务器不稳定，因手机端直接翻墙稳定

谷歌服务访问故障的时候，
有两种解决办法，开关wifi后瞬间恢复
已知开关wifi会清空dns缓存，断开所有连接
或此时在手机端使用dns lookup等工具查询一下相关域名 play.google.com 的dns A记录，发现地址发生变化，稍后自动恢复连接

所以得知手机必须要重新查询dns才可以恢复连接
手机放置一段时间后用之前的dns记录直接连接就会不通。
有没有可能是dns处理，ipset等的问题？

我这次尝试直接在手机端设置dns为8.8.8.8继续测试

hcym, 2020-04-10:
一直用的手机杂牌，除了第一次登录容易出错，需要重启啊改全局模式啊的，后面使用基本正常，感觉还是dns侧漏吧

mjyhj, 2020-04-10:
小米MIUI11，从没有遇到访问不了的情况

txdywy, 2020-04-10:
查出来的dns是污染后的吧

9l, 2020-04-11:
services.googleapis.cn

这个域名加入到强制走代理

lizhenglin2002, 2020-04-14:
> services.googleapis.cn
> 
> 这个域名加入到强制走代理

其实添加这个域名有啥作用的？

martin0607, 2020-04-15:
iOS端正常
安卓端也是像你说的，长时间闲置需要重新开关Wifi才能正常

lizhenglin2002, 2020-05-13:
我把DNS设置成8.8.8.8无解，iOS正常

closed 2020-07-04 by coolsnowwolf

z826540272, 2020-07-20:
> 小米MIUI11，从没有遇到访问不了的情况。
> 好明显是你用的那种伪旁路带来的问题

我确实是用来作为网关的，那么怎么解决

z826540272, 2021-02-17:
> iOS端正常
> 安卓端也是像你说的，长时间闲置需要重新开关Wifi才能正常

请问你是不是用的网关模式, 这个问题可能和网关模式有关

chanho0, 2022-01-02:
安卓+旁路由网关模式就会这样，ios+旁路由网关正常