coolsnowwolf/lede #4264: xiaomi mini 16m固件默认配置编译报错

xwbchn, 2020-04-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
xiaomi mini 默认配置编译到最后一部分报错

### (2) 路由器型号和固件版本

A：Xiaomi mini

### (3) 详细日志

A：Configuring shadowsocksr-libev-alt.
Configuring kmod-rt2800-soc.
Configuring kmod-mt76x2.
Configuring luci-i18n-upnp-zh-cn.
Configuring libustream-openssl20150806.
Configuring urngd.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-ssr-plus:
 * 	v2ray-plugin
 * 	v2ray
 * 	trojan
 * 	ipt2socks
 * 	redsocks2
 * 	kcptun-client
 * 	shadowsocksr-libev-server
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
make[2]: *** [package/install] Error 255

AmadeusGhost, 2020-04-10:
`make package/luci-app-ssr-plus/{clean,compile} V=s`或者直接`make clean`

fuokgfw, 2020-04-10:
不就是说少了这些个文件夹啊，找到并放在一起就行了啊

9l, 2020-04-11:
怕不是你编译环境网络不行，有些东西没下全吧

lon91ong, 2020-04-11:
你确定16M的固件放得下v2ray? 我折腾了两天, v2ray太大了, 放弃了!

xwbchn, 2020-04-11:
我根本就没选v2 就默认的配置 结果编译到这里就是报缺依赖 没搞懂

從我的iPhone傳送

xwbchn, 2020-04-11:
反复执行了10+次make download V=s，确认库全了，原因不明。

從我的iPhone傳送

xwbchn, 2020-04-11:
不是少文件夹 是少依赖 关键是我没选这几个功能 不应该这样的

從我的iPhone傳送

xwbchn, 2020-04-11:
好的，我尝试一下，谢谢。

從我的iPhone傳送

closed 2020-07-04 by coolsnowwolf