coolsnowwolf/lede #4269: 新固件版本内核是不是升级了，报错

fuokgfw, 2020-04-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

make[3]: Entering directory '/home/sc0000000/Desktop/openwrt/package/lede/luci-app-flowoffload'
bash: % 86400: syntax error: operand expected (error token is "% 86400")
bash: % 86400: syntax error: operand expected (error token is "% 86400")
/home/sc0000000/Desktop/openwrt/feeds/luci/luci.mk:277: *** Package/luci-app-flowoffload is missing the VERSION field.  Stop.
make[3]: Leaving directory '/home/sc0000000/Desktop/openwrt/package/lede/luci-app-flowoffload'
time: package/lede/luci-app-flowoffload/compile#0.66#0.37#1.40
package/Makefile:111: recipe for target 'package/lede/luci-app-flowoffload/compile' failed
make[2]: *** [package/lede/luci-app-flowoffload/compile] Error 2
make[2]: Leaving directory '/home/sc0000000/Desktop/openwrt'
package/Makefile:107: recipe for target '/home/sc0000000/Desktop/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/sc0000000/Desktop/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sc0000000/Desktop/openwrt'
/home/sc0000000/Desktop/openwrt/include/toplevel.mk:225: recipe for target 'world' failed

garypang13, 2020-04-11:
你用的是官方master分支的luci吧

fuokgfw, 2020-04-11:
> 你用的是官方master分支的luci吧

是的

garypang13, 2020-04-11:
https://github.com/openwrt/luci/commit/59fe4086f3a93f919547154373ac3d5e975e7db2#diff-2d8a7a9b9acd99058ff6f4d496ea6ebe 只支持在feeds.conf通过git添加的插件了, 把luci.mk文件降级吧,

vbzc, 2020-04-14:
请问 有没有办法 自己加这个 VERSION field 呀？

a-kibou, 2020-04-14:
> 请问 有没有办法 自己加这个 VERSION field 呀？

將luci-app-xxx拷貝到openwrt/feeds/luci/application下，終端下輸入 ./scripts/feeds install -a 安裝就可以了。

fuokgfw, 2020-04-14:
> > 请问 有没有办法 自己加这个 VERSION field 呀？
> 
> 將luci-app-xxx拷貝到openwrt/feeds/luci/application下，終端下輸入 ./scripts/feeds install -a 安裝就可以了。

感谢

fuokgfw, 2020-04-14:
> > 请问 有没有办法 自己加这个 VERSION field 呀？
> 
> 將luci-app-xxx拷貝到openwrt/feeds/luci/application下，終端下輸入 ./scripts/feeds install -a 安裝就可以了。

这个操作并不能在菜单中出现选项啊

garypang13, 2020-04-14:
需要先 ./scripts/feeds update -a,但这个方法我测试过了,并不能解决 VERSION field的问题.

a-kibou, 2020-04-14:
> 需要先 ./scripts/feeds update -a,但这个方法我早就试过了,并不能解决 VERSION field的问题.

確實不行，應是缺少VERSION欄位的問題，不過最新的luci.mk已經修正了。
https://github.com/openwrt/luci/issues/3911

snakwu, 2020-04-15:
> > 需要先 ./scripts/feeds update -a,但这个方法我早就试过了,并不能解决 VERSION field的问题.
> 
> 確實不行，應是缺少VERSION欄位的問題，不過最新的luci.mk已經修正了。
> [openwrt/luci#3911](https://github.com/openwrt/luci/issues/3911)

请不要两边相互引用问题,这样会搞混乱!

closed 2020-07-04 by coolsnowwolf