coolsnowwolf/lede #4272: 编译错误net/http: TLS handshake timeout

linxuankent, 2020-04-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：Finding targets
go: github.com/json-iterator/go@v1.1.9: Get "https://proxy.golang.org/github.com/json-iterator/go/@v/v1.1.9.mod": net/http: TLS handshake timeout

Building targets
go: github.com/json-iterator/go@v1.1.9: Get "https://proxy.golang.org/github.com/json-iterator/go/@v/v1.1.9.mod": net/http: TLS handshake timeout

Cleaning module download cache (golang/go#27455)

Makefile:65: recipe for target '/home/lin/Desktop/openwrt/lede/build_dir/target-x86_64_musl/UnblockNeteaseMusicGo/.built' failed
make[3]: *** [/home/lin/Desktop/openwrt/lede/build_dir/target-x86_64_musl/UnblockNeteaseMusicGo/.built] Error 1
make[3]: Leaving directory '/home/lin/Desktop/openwrt/lede/package/lean/UnblockNeteaseMusicGo'
time: package/lean/UnblockNeteaseMusicGo/compile#0.31#0.13#20.42
package/Makefile:111: recipe for target 'package/lean/UnblockNeteaseMusicGo/compile' failed
make[2]: *** [package/lean/UnblockNeteaseMusicGo/compile] Error 2
make[2]: Leaving directory '/home/lin/Desktop/openwrt/lede'
package/Makefile:107: recipe for target '/home/lin/Desktop/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/lin/Desktop/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lin/Desktop/openwrt/lede'
/home/lin/Desktop/openwrt/lede/include/toplevel.mk:225: recipe for target 'world' failed

Potterli20, 2020-04-11:
哈哈哈我也有哦
go: github.com/v2ray/v2ray-core@v4.22.1+incompatible: reading https://goproxy.cn/github.com/v2ray/v2ray-core/@v/v4.22.1+incompatible.mod: 404 Not Found
        server response: not found: github.com/v2ray/v2ray-core@v4.22.1+incompatible: invalid version: +incompatible suffix not allowed: module contains a go.mod file, so semantic import versioning is required

linxuankent, 2020-04-11:
> 哈哈哈我也有哦
> go：github.com/v2ray/v2ray-core@v4.22.1+incompatible：阅读[https://goproxy.cn/github.com/v2ray/v2ray-core/@v/v4.22.1+incompatible .mod](https://goproxy.cn/github.com/v2ray/v2ray-core/@v/v4.22.1+incompatible.mod)：404未找到
> 服务器响应：找不到：github.com/v2ray/v2ray-core@v4.22.1+incompatible：无效版本：不允许使用+ incompatible后缀：模块包含go.mod文件，因此语义导入版本控制为需要
go: github.com/json-iterator/go@v1.1.9: Get "https://proxy.golang.org/github.com/json-iterator/go/@v/v1.1.9.mod": dial tcp 172.217.160.113:443: i/o timeout

Building targets
go: github.com/json-iterator/go@v1.1.9: Get "https://proxy.golang.org/github.com/json-iterator/go/@v/v1.1.9.mod": dial tcp 172.217.160.113:443: i/o timeout

多次尝试就成这样了

Potterli20, 2020-04-11:
> 哈哈哈我也有哦
> go: github.com/v2ray/v2ray-core@v4.22.1+incompatible: reading https://goproxy.cn/github.com/v2ray/v2ray-core/@v/v4.22.1+incompatible.mod: 404 Not Found
> server response: not found: [...]

@ @coolsnowwolf 
l大，请问go的代理如何关闭？？
我一直有个问题

linxuankent, 2020-04-13:
> > 哈哈哈我也有哦
> > go：github.com/v2ray/v2ray-core@v4.22.1+incompatible：阅读[https://goproxy.cn/github.com/v2ray/v2ray-core/@v/v4.22.1+incompatible .mod](https://goproxy.cn/github.com/v2ray/v2ray-core/@v/v4.22.1+incompatible.mod)：404未找到
> > 服务器响应：找不到：github.com/v2ray/v2ray-core@v4.22.1+incompatible：无效版本：不允许使用+ incompatible后缀：模块包含go.mod文件，因此语义导入版本控制为需要
> 
> @ @coolsnowwolf
> l大，请问 go的代理如何关闭？？
> 我一直有个问题
更新到最新版本编译就没报错

linxuankent, 2020-04-13:
> @ @coolsnowwolf
> l大，请问 go的代理如何关闭？？
> 我一直有个问题

更新到最新版本，我就没这个问题了

closed 2020-04-13 by linxuankent

ClayStan, 2021-02-20:
> > 新版本，我就没这个问题了
请问下是如何更新到最新版本的，我今天编译也一直出现这个问题

linxuankent, 2021-03-29:
> > > 新版本，我就没这个问题了
> 
> 请问下是如何更新到最新版本的，我今天编译也一直出现这个问题

git pull 
./scripts/feeds update -a && ./scripts/feeds install -a