coolsnowwolf/lede #4280: X64最新源码无限重启

bin20088, 2020-04-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
X64最新源码编译的固件刷机后无限重启
Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：X64最新源码编译的固件刷机后无限重启

### (2) 路由器型号和固件版本

A：X64

### (3) 详细日志

A：
[image]

ckxt, 2020-04-12:
我昨晚编译了 没问题，会不会是优盘/硬盘有坏块？

bin20088, 2020-04-12:
我是包含的插件多，之前没问题，就是加入了PCHIFI.CIFS-MOUNT之后出问题了，我还选了netdata，minidlna，应该是这四个之间有点冲突。可以成功编译，就是刷完机无限重启进不去系统。

bin20088, 2020-04-12:
已解决

closed 2020-04-12 by bin20088

bang88, 2020-05-11:
怎么解决的 是不是可以说一下帮助后面有需要的人。

bin20088, 2020-05-11:
我的问题你们一般不会遇到，是因为我添加的插件过多，超出了默认的内存值，导致刷机失败。

bang88, 2020-05-11:
制作PE盘中... 我的分区被搞坏了，然后无限重启。

Gongzigao, 2020-07-10:
大佬，我也遇到这个问题了，请问可以告知从哪里改内存大小吗？

bin20088, 2020-07-10:
> 大佬，我也遇到这个问题了，请问可以知道从哪里改内存大小吗？

target images----root filesystem