coolsnowwolf/lede #4281: 小米3Pro更新后，WIFI无法启动

hcwang, 2020-04-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：从两周前开始，在更新ROM或者重启后，WIFI无法启动，需要通过有线方式进入后台，对WIFI进行重新设置（保存并应用）后方可启动。

### (2) 路由器型号和固件版本

A：小米3Pro

### (3) 详细日志

A：

```
Sun Apr 12 07:07:56 2020 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Sun Apr 12 07:07:56 2020 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sun Apr 12 07:07:56 2020 daemon.err hostapd: Using interface wlan0 with hwaddr 78:11:dc:04:4d:f6 and ssid "1601"
Sun Apr 12 07:07:57 2020 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ENABLED
Sun Apr 12 07:07:57 2020 daemon.notice hostapd: wlan0: AP-ENABLED
Sun Apr 12 07:07:58 2020 user.notice root: ip link set dev wlan0 up
Sun Apr 12 07:07:58 2020 daemon.info hostapd: wlan0: STA 78:11:dc:05:85:7b IEEE 802.11: authenticated
Sun Apr 12 07:07:58 2020 daemon.notice netifd: Network device 'wlan0' link is up
Sun Apr 12 07:07:59 2020 daemon.info hostapd: wlan0: STA 78:11:dc:61:79:62 IEEE 802.11: authenticated
Sun Apr 12 07:07:59 2020 daemon.info hostapd: wlan0: STA 78:11:dc:61:79:62 IEEE 802.11: associated (aid 1)
Sun Apr 12 07:07:59 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 78:11:dc:61:79:62
Sun Apr 12 07:07:59 2020 daemon.info hostapd: wlan0: STA 78:11:dc:61:79:62 WPA: pairwise key handshake completed (RSN)
Sun Apr 12 07:07:59 2020 daemon.info dnsmasq-dhcp[6326]: DHCPREQUEST(br-lan) 192.168.88.195 78:11:dc:61:79:62
Sun Apr 12 07:07:59 2020 daemon.info dnsmasq-dhcp[6326]: DHCPNAK(br-lan) 192.168.88.195 78:11:dc:61:79:62 address not available
Sun Apr 12 07:08:00 2020 daemon.info dnsmasq-dhcp[6326]: DHCPREQUEST(br-lan) 192.168.88.195 78:11:dc:61:79:62
Sun Apr 12 07:08:00 2020 daemon.info dnsmasq-dhcp[6326]: DHCPNAK(br-lan) 192.168.88.195 78:11:dc:61:79:62 address not available
Sun Apr 12 07:08:02 2020 daemon.info hostapd: wlan0: STA 64:09:80:2f:fd:79 IEEE 802.11: authenticated
Sun Apr 12 07:08:02 2020 daemon.info hostapd: wlan0: STA 64:09:80:2f:fd:79 IEEE 802.11: associated (aid 2)
Sun Apr 12 07:08:02 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 64:09:80:2f:fd:79
Sun Apr 12 07:08:02 2020 daemon.info hostapd: wlan0: STA 64:09:80:2f:fd:79 WPA: pairwise key handshake completed (RSN)
Sun Apr 12 07:08:02 2020 daemon.info hostapd: wlan0: STA f0:b4:29:07:cc:cc IEEE 802.11: authenticated
Sun Apr 12 07:08:02 2020 daemon.info hostapd: wlan0: STA f0:b4:29:07:cc:cc IEEE 802.11: associated (aid 3)
Sun Apr 12 07:08:02 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED f0:b4:29:07:cc:cc
Sun Apr 12 07:08:02 2020 daemon.info hostapd: wlan0: STA f0:b4:29:07:cc:cc WPA: pairwise key handshake completed (RSN)
Sun Apr 12 07:08:02 2020 daemon.notice hostapd: wlan0: STA 78:11:dc:05:85:7b IEEE 802.11: did not acknowledge authentication response
Sun Apr 12 07:08:02 2020 daemon.info hostapd: wlan0: STA 78:11:dc:05:85:7b IEEE 802.11: associated (aid 4)
Sun Apr 12 07:08:02 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 78:11:dc:05:85:7b
Sun Apr 12 07:08:02 2020 daemon.info hostapd: wlan0: STA 78:11:dc:05:85:7b WPA: pairwise key handshake completed (RSN)
```

hcwang, 2020-04-12:
内核日志如下：

```
[ 1000.590785] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
[ 1005.813258] mtk_soc_eth 1e100000.ethernet eth0: port 1 link up
[ 1096.161079] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
[ 1099.098789] mtk_soc_eth 1e100000.ethernet eth0: port 1 link up
[ 1137.428645] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
[ 1141.514497] mtk_soc_eth 1e100000.ethernet eth0: port 1 link up
[ 1154.404096] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
[ 1157.509303] mtk_soc_eth 1e100000.ethernet eth0: port 1 link up
[ 1220.374173] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
[ 1226.817159] mtk_soc_eth 1e100000.ethernet eth0: port 1 link up
[ 1228.170416] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
[ 1231.223217] mtk_soc_eth 1e100000.ethernet eth0: port 1 link up
[ 1330.133874] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
[ 1333.192076] mtk_soc_eth 1e100000.ethernet eth0: port 1 link up
[ 1750.547722] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
[ 1755.735304] mtk_soc_eth 1e100000.ethernet eth0: port 1 link up
[ 1832.802811] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
[ 1835.721233] mtk_soc_eth 1e100000.ethernet eth0: port 1 link up
[ 1837.436668] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
[ 1840.343445] mtk_soc_eth 1e100000.ethernet eth0: port 1 link up
[ 2428.301412] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
[ 2430.240800] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[ 8704.826195] ------------[ cut here ]------------
[ 8704.831040] WARNING: CPU: 0 PID: 0 at target-mipsel_24kc_musl/linux-ramips_mt7621/backports-5.4.27-1/net/mac80211/sta_info.c:1937 ieee80211_sta_update_pending_airtime+0x104/0x204 [mac80211]
[ 8704.847857] STA 00:db:70:65:20:84 AC 2 txq pending airtime underflow: 4294967248, 48
[ 8704.847863] Modules linked in: xt_FULLCONENAT pppoe ppp_async pppox ppp_mppe ppp_generic mt7615e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_bpf xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm slhc nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv4 nf_conntrack_rtcache
[ 8704.926127]  nf_conntrack_netlink macvlan iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat fuse tcp_bbr ledtrig_usbport cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 nf_nat nf_conntrack nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 sit tunnel4 ip_tunnel tun vfat fat ntfs hfsplus hfs cifs nls_utf8 nls_iso8859_1 nls_cp437 sha256_generic sha1_generic
[ 8704.997130]  md5 md4 hmac ecb des_generic authenc usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd sd_mod scsi_mod gpio_button_hotplug f2fs ext4 mbcache jbd2 usbcore nls_base usb_common crc32c_generic crc32_generic
[ 8705.017547] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.14.172 #0
[ 8705.023635] Stack : 00000000 8ee30d80 00000000 8006c638 805b0000 80557e40 00000000 00000000
[ 8705.031995]         80522788 8fc09ca4 80592d3c 805928c7 8051d5c0 00000001 8fc09c48 1cc2824c
[ 8705.040345]         00000000 00000000 806f0000 00008d88 00000000 000001b7 00000008 00000000
[ 8705.048692]         00000000 80590000 0000448b 00000000 00000000 805b0000 00000000 8ea877d8
[ 8705.057038]         00000009 00000791 00000001 8ee30d80 00000000 802a4908 00000000 806f0000
[ 8705.065384]         ...
[ 8705.067826] Call Trace:
[ 8705.070298] [<8000c6f0>] show_stack+0x58/0x100
[ 8705.074766] [<8045b754>] dump_stack+0xa4/0xe0
[ 8705.079117] [<8002aec8>] __warn+0xe0/0x114
[ 8705.083201] [<8002af2c>] warn_slowpath_fmt+0x30/0x3c
[ 8705.088311] [<8ea877d8>] ieee80211_sta_update_pending_airtime+0x104/0x204 [mac80211]
[ 8705.096103] [<8ea81d38>] ieee80211_report_low_ack+0xac/0x43c [mac80211]
[ 8705.102775] [<8ea820e0>] ieee80211_free_txskb+0x18/0x30 [mac80211]
[ 8705.108996] [<8ea04f58>] mt76_tx_complete_skb+0x5c/0xb0 [mt76]
[ 8705.114832] [<8ea26f24>] mt7615_txp_skb_unmap+0x1f0/0x22c [mt7615e]
[ 8705.121275] ---[ end trace 6cd6b42f8d84e082 ]---
```

Twinzo1, 2020-04-13:
mini也是

jxph1123, 2020-04-13:
MTK7621好像也是这毛病，重启后5G WIFI连不上，一直获取IP地址，无响应，我换回旧版本了。

AmadeusGhost, 2020-04-14:
http://tmp.link/f/5e95a7a8c4630
试一下这个sysupgrade固件？不过WiFi可能不会自启

hcwang, 2020-04-14:
非常感谢，我目前/etc/rc.local文件中增加了一行代码，延时60秒调用/sbin/wifi up，就可以在重启或者固件更新后，大约120秒左右使WIFI启动起来。

```
nohup sleep 60 && /sbin/wifi up &
```

不过还是不明白为什么需要调用一次wifi up，在我的/etc/config/wireless当中并没有disable呀。

感觉系统日志中以下可能是关键所在，不过google了几天，没有找到解决方案。只能等上游更新了。

```
Sun Mar  8 10:20:01 2020 daemon.notice netifd: radio1 (17980): Command failed: Request timed out
Sun Mar  8 10:20:01 2020 daemon.notice netifd: radio1 (17980): Command failed: Not found
Sun Mar  8 10:20:01 2020 daemon.notice netifd: radio1 (17980): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Sun Mar  8 10:20:01 2020 daemon.notice netifd: radio1 (17980): Command failed: Invalid argument
```

jackberry, 2020-04-15:
小米r3p先回官方再刷breed后，再刷自己编译的固件（恩山有教程），不要用xwrt过渡刷其他op，我的就是这么好的，现在怎么刷无线都没事，应该不是源码问题。pb-boot下刷xwrt过渡的可能会有这个问题

hcwang, 2020-04-15:
我刷两周前的固件正常，两周后的就不正常，就是两周前签入的mac80211代码的事情。

AmadeusGhost, 2020-04-23:
可以试下我的pr：#4420

closed 2020-04-24 by hcwang