coolsnowwolf/lede #4288: luci页面修改配置后保存&应用无效

ilongli, 2020-04-12:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
想把默认的uhttpd换成nginx，所以编译的时候把Network -> Web Servers/Proxies -> nginx-mod-luci-ssl勾上了（勾上这个会把相关的自动勾上）。
能正常进入路由的管理页面，**第一次**保存&应用的时候是正常的，此时看了下luci相关的进程：
`root@OpenWrt:~# ps -ef | grep luci`
` 6235 root      1064 S    lock /var/run/luci-reload`
` 6561 root      1072 S    grep luci`

然后**第二次**保存&应用一个配置，发现无效，此时的进程：
`root@OpenWrt:~# ps -ef | grep luci`
` 6235 root      1064 S    lock /var/run/luci-reload`
` 6641 root      1144 S    /bin/sh /sbin/luci-reload adbyby`
` 6642 root      1064 S    lock /var/run/luci-reload`
` 6664 root      1072 S    grep luci`

自己分析了下，感觉是文件锁没有释放？所以自己手动把锁的进程杀掉了：
`root@OpenWrt:~# kill -9 6235`

杀掉6235进程后**第二次**的保存&应用则生效了，此时的进程：
`root@OpenWrt:~# ps -ef | grep luci`
` 6723 root      1064 S    lock /var/run/luci-reload`
` 6883 root      1072 S    grep luci`

我目前临时的解决办法是把`/sbin/luci-reload`的36和45行代码注释掉了：
`#lock "/var/run/luci-reload"`
`#lock -u "/var/run/luci-reload"`

说个题外话，刚刷完固件第一次进系统nginx是正常启动的，但是重启后发现uhttpd启动了，并占用了端口，nginx无法正常启动，并且此时路由管理页面是进不去的。不过手动`service uhttpd disable`就好了，也不算大问题。
不知道是不是我编译的配置有误。编译的时候如何配置才能正确地把uhttpd换成nginx？还请各位不吝赐教。

### (2) 路由器型号和固件版本

A：
Intel(R) Celeron(R) CPU N3150 @ 1.60GHz
X86_64 OpenWrt R20.4.8 / LuCI Master (git-20.088.29068-cebe120)

### (3) 详细日志

A：
[op.log](https://github.com/coolsnowwolf/lede/files/4466058/op.log)

lloyd18, 2020-04-12:
我也想知道

garypang13, 2020-04-12:
https://github.com/garypang13/Actions-OpenWrt-Nginx/ 我的就是把uhttpd换成了nginx,没有编译uhttpd,并没有你说的这些问题,可以参考下.

closed 2020-07-04 by coolsnowwolf

liwx, 2020-07-06:
我也遇到了，是因为断电意外关机，再重启后系统文件变成了只读，我用下面办法解决了
ssh登录后执行命令

e2fsck -y /dev/sda2

然后reboot一下就可以了，goodluck

freenubc, 2020-07-11:
我也遇到这个奇葩问题了，保存并应用按钮无效，一点就恢复之前设置了，但点保存，然后未保存的配置，点应用，就成功保存了，但未保存的配置提示还在，这时候再点一次里面的保存并应用，就恢复之前的设置了

WROIATE, 2020-09-19:
> https://github.com/garypang13/Actions-OpenWrt-Nginx/ 我的就是把uhttpd换成了nginx,没有编译uhttpd,并没有你说的这些问题,可以参考下.

我把uhttpd取消掉了也还是有这问题

WROIATE, 2020-09-19:
> 我也遇到了，是因为断电意外关机，再重启后系统文件变成了只读，我用下面办法解决了
> ssh登录后执行命令
> 
> e2fsck -y /dev/sda2
> 
> 然后reboot一下就可以了，goodluck

```
root@OpenWrt:~# e2fsck -y /dev/sda2
e2fsck 1.44.5 (15-Dec-2018)
/dev/sda2 is in use.
e2fsck: Cannot continue, aborting.
```
好像没用啊

mis1042, 2021-03-20:
> 我也遇到这个奇葩问题了，保存并应用按钮无效，一点就恢复之前设置了，但点保存，然后未保存的配置，点应用，就成功保存了，但未保存的配置提示还在，这时候再点一次里面的保存并应用，就恢复之前的设置了

解决了嘛大佬，我也碰到了

isher, 2021-04-09:
看样子我不是第一例遇到这额问题的，同样也是安装了nginx后发现点击uhttpd应用无效，不过重刷了一遍ow之后，安装了luci-app-uhttpd之后，通过管理面板应用，依然无效.....

TangZhongham, 2021-06-08:
解决了嘛。。。我现在页面配置啥都失败了。。。

isher, 2021-06-10:
已经解决，php配置文件问题

WaterSeekingFish, 2021-12-16:
> 

请问大佬具体怎么解决###