coolsnowwolf/lede #4293: PUBG电脑和手机版都不走UDP

liliass, 2020-04-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [* ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：PUBG电脑和手机版都不走UDP. (用SSTAP和iOS Shadowrockets,Quantumult UDP正常，节点也正常)

如果在 “增强游戏模式客户端LAN IP” 添加我的手机/电脑IP。 游戏都进不去。 

### (2) 路由器型号和固件版本

A：OpenWrt R20.4.8_dev_Len yu / LuCI Master (git-20.088.29068-cebe120)

luci-app-ssr-plus | 176-2
-- | --

### (3) 详细日志

`A：root@OpenWrt:~# iptables -t mangle -L
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination         
SS_SPEC_TPROXY  udp  --  anywhere             anywhere             /* _SS_SPEC_RULE_ */

Chain INPUT (policy ACCEPT)
target     prot opt source               destination         

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination         
TCPMSS     tcp  --  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination         

Chain SS_SPEC_TPROXY (1 references)
target     prot opt source               destination         
RETURN     udp  --  anywhere             anywhere             udp dpt:https
RETURN     udp  --  anywhere             anywhere             udp dpt:80
RETURN     udp  --  anywhere             anywhere             udp dpt:domain
RETURN     udp  --  anywhere             0.0.0.0/8           
RETURN     udp  --  anywhere             10.0.0.0/8          
RETURN     udp  --  anywhere             127.0.0.0/8         
RETURN     udp  --  anywhere             169.254.0.0/16      
RETURN     udp  --  anywhere             172.16.0.0/12       
RETURN     udp  --  anywhere             192.168.0.0/16      
RETURN     udp  --  anywhere             base-address.mcast.net/4 
RETURN     udp  --  anywhere             240.0.0.0/4         
RETURN     udp  --  anywhere             167.238.92.34.bc.googleusercontent.com  udp dpt:!domain
RETURN     udp  --  anywhere             anywhere             match-set bplan src
TPROXY     udp  --  anywhere             anywhere             multiport dports ssh,domain,submission,465,pop3s,imaps,imap2,80,https match-set fplan src TPROXY redirect 0.0.0.0:1234 mark 0x1/0x1
TPROXY     udp  --  anywhere             anywhere             match-set gmlan src ! match-set china dst TPROXY redirect 0.0.0.0:1234 mark 0x1/0x1
TPROXY     udp  --  anywhere             anywhere             multiport dports ssh,domain,submission,465,pop3s,imaps,imap2,80,https ! match-set ss_spec_wan_ac dst TPROXY redirect 0.0.0.0:1234 mark 0x1/0x1
root@OpenWrt:~# 
`

`root@OpenWrt:~# lsmod | grep TPROXY
nf_defrag_ipv4         16384  3 nf_conntrack,xt_socket,xt_TPROXY
nf_defrag_ipv6         16384  3 nf_conntrack,xt_socket,xt_TPROXY
nf_tproxy_ipv4         16384  1 xt_TPROXY
nf_tproxy_ipv6         16384  1 xt_TPROXY
x_tables               24576 35 ipt_MASQUERADE,xt_state,xt_nat,xt_conntrack,xt_REDIRECT,xt_FLOWOFFLOAD,xt_CT,ipt_REJECT,xt_time,xt_tcpudp,xt_socket,xt_quota,xt_policy,xt_pkttype,xt_owner,xt_multiport,xt_mark,xt_mac,xt_limit,xt_esp,xt_comment,xt_addrtype,xt_TPROXY,xt_TCPMSS,xt_LOG,iptable_raw,iptable_mangle,iptable_filter,ipt_ah,ip_tables,xt_set,ip6table_mangle,ip6table_filter,ip6_tables,ip6t_REJECT
xt_TPROXY              16384  3 
root@OpenWrt:~# 
`

liliass, 2020-04-12:
`root@OpenWrt:~# iptables -t mangle -L
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination         
SS_SPEC_TPROXY  udp  --  anywhere             anywhere             /* _SS_SPEC_RULE_ */

Chain INPUT (policy ACCEPT)
target     prot opt source               destination         

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination         
TCPMSS     tcp  --  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination         

Chain SS_SPEC_TPROXY (1 references)
target     prot opt source               destination         
RETURN     udp  --  anywhere             anywhere             udp dpt:https
RETURN     udp  --  anywhere             anywhere             udp dpt:80
RETURN     udp  --  anywhere             anywhere             udp dpt:domain
RETURN     udp  --  anywhere             0.0.0.0/8           
RETURN     udp  --  anywhere             10.0.0.0/8          
RETURN     udp  --  anywhere             127.0.0.0/8         
RETURN     udp  --  anywhere             169.254.0.0/16      
RETURN     udp  --  anywhere             172.16.0.0/12       
RETURN     udp  --  anywhere             192.168.0.0/16      
RETURN     udp  --  anywhere             base-address.mcast.net/4 
RETURN     udp  --  anywhere             240.0.0.0/4         
RETURN     udp  --  anywhere             167.238.92.34.bc.googleusercontent.com  udp dpt:!domain
RETURN     udp  --  anywhere             anywhere             match-set bplan src
TPROXY     udp  --  anywhere             anywhere             multiport dports ssh,domain,submission,465,pop3s,imaps,imap2,80,https match-set fplan src TPROXY redirect 0.0.0.0:1234 mark 0x1/0x1
TPROXY     udp  --  anywhere             anywhere             match-set gmlan src ! match-set china dst TPROXY redirect 0.0.0.0:1234 mark 0x1/0x1
TPROXY     udp  --  anywhere             anywhere             multiport dports ssh,domain,submission,465,pop3s,imaps,imap2,80,https ! match-set ss_spec_wan_ac dst TPROXY redirect 0.0.0.0:1234 mark 0x1/0x1
root@OpenWrt:~# 
`

1049066154, 2020-04-14:
把pubg服务器的ip段添加到黑名单里

liliass, 2020-04-14:
```
Tue Apr 14 13:21:08 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Apr 14 13:21:08 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Apr 14 13:21:40 2020 user.notice shadowsocksr: ssr redir udp error.restart!
Tue Apr 14 13:21:41 2020 cron.info crond[20593]: crond (busybox 1.30.0) started, log level 8
Tue Apr 14 13:21:42 2020 daemon.info /usr/bin/ss-redir[20890]: initializing ciphers... chacha20
Tue Apr 14 13:21:42 2020 daemon.info /usr/bin/ss-redir[20890]: listening at 0.0.0.0:1234
Tue Apr 14 13:21:42 2020 daemon.info /usr/bin/ss-redir[20890]: UDP relay enabled
Tue Apr 14 13:21:42 2020 daemon.info /usr/bin/ss-redir[20890]: udp port reuse enabled
Tue Apr 14 13:21:42 2020 daemon.info /usr/bin/ss-redir[20890]: TCP relay disabled
Tue Apr 14 13:21:42 2020 daemon.info /usr/bin/ss-redir[20890]: running from root user
Tue Apr 14 13:21:42 2020 daemon.info pdnsd[20929]: pdnsd-1.2.9b-par starting.
Tue Apr 14 13:21:42 2020 daemon.warn pdnsd[20929]: EOF encountered while reading entry count from disk cache file.
Tue Apr 14 13:22:13 2020 user.notice shadowsocksr: ssr redir udp error.restart!
Tue Apr 14 13:22:13 2020 cron.info crond[24119]: crond (busybox 1.30.0) started, log level 8
Tue Apr 14 13:22:14 2020 daemon.info /usr/bin/ss-redir[24469]: initializing ciphers... chacha20
Tue Apr 14 13:22:14 2020 daemon.info /usr/bin/ss-redir[24469]: listening at 0.0.0.0:1234
Tue Apr 14 13:22:14 2020 daemon.info /usr/bin/ss-redir[24469]: UDP relay enabled
Tue Apr 14 13:22:14 2020 daemon.info /usr/bin/ss-redir[24469]: udp port reuse enabled
Tue Apr 14 13:22:14 2020 daemon.info /usr/bin/ss-redir[24469]: TCP relay disabled
Tue Apr 14 13:22:14 2020 daemon.info /usr/bin/ss-redir[24469]: running from root user
Tue Apr 14 13:22:15 2020 daemon.info pdnsd[24509]: pdnsd-1.2.9b-par starting.
Tue Apr 14 13:22:15 2020 daemon.warn pdnsd[24509]: EOF encountered while reading entry count from disk cache file.
Tue Apr 14 13:22:22 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Apr 14 13:22:22 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Apr 14 13:22:22 2020 cron.info crond[25470]: crond (busybox 1.30.0) started, log level 8
Tue Apr 14 13:22:23 2020 daemon.info /usr/bin/ss-redir[25760]: initializing ciphers... chacha20
Tue Apr 14 13:22:23 2020 daemon.info /usr/bin/ss-redir[25760]: listening at 0.0.0.0:1234
Tue Apr 14 13:22:23 2020 daemon.info /usr/bin/ss-redir[25760]: UDP relay enabled
Tue Apr 14 13:22:23 2020 daemon.info /usr/bin/ss-redir[25760]: udp port reuse enabled
Tue Apr 14 13:22:23 2020 daemon.info /usr/bin/ss-redir[25760]: TCP relay disabled
Tue Apr 14 13:22:23 2020 daemon.info /usr/bin/ss-redir[25760]: running from root user
Tue Apr 14 13:22:23 2020 daemon.info pdnsd[25801]: pdnsd-1.2.9b-par starting.
Tue Apr 14 13:22:23 2020 daemon.warn pdnsd[25801]: EOF encountered while reading entry count from disk cache file.
Tue Apr 14 13:22:56 2020 user.notice shadowsocksr: ssr redir udp error.restart!
Tue Apr 14 13:22:56 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Apr 14 13:22:56 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
```

**### Tue Apr 14 13:22:13 2020 user.notice shadowsocksr: ssr redir udp error.restart!**

看到UDP error 不知道正常吗。

feheadls, 2020-04-15:
我使用的第三方基于LEAN大佬编译的固件也是这样，UDP流量不能通过SSR PLUS+或者PASSWALL转发，PC NAT工具检查为udpblocked。已经好几个版本都这样了，也不知道是不是一直就不行，太早的版本没试过。

coolsnowwolf, 2020-04-15:
ssr redir udp error.restart 请问作者了，这里不能解决

feheadls, 2020-04-15:
据说是固件集成了docker的锅，单纯卸载还不行，编译时就不能将docker编译进去的固件才行，有人已经这样解决了，我找时间验证一下。

feheadls, 2020-04-16:
> ssr redir udp error.restart 请问作者了，这里不能解决

的确，装了个未集成DOCKER的固件，UDP转发失败的问题解决了，还请大佬看下这个问题。

liliass, 2020-04-16:
> > ssr redir udp error.restart 请问作者了，这里不能解决
> 
> 的确，装了个未集成DOCKER的固件，UDP转发失败的问题解决了，还请大佬看下这个问题。

你是用什么工具测试UDP的 ??? 
我的版本没集成Docker， 也不行

feheadls, 2020-04-16:
> > > ssr redir udp error.restart 请问作者了，这里不能解决
> > 
> > 
> > 的确，装了个未集成DOCKER的固件，UDP转发失败的问题解决了，还请大佬看下这个问题。
> 
> 你是用什么工具测试UDP的 ???
> 我的版本没集成Docker， 也不行

PC用NAT类型测试工具，很好找。

feheadls, 2020-04-16:
> > > ssr redir udp error.restart 请问作者了，这里不能解决
> > 
> > 
> > 的确，装了个未集成DOCKER的固件，UDP转发失败的问题解决了，还请大佬看下这个问题。
> 
> 你是用什么工具测试UDP的 ???
> 我的版本没集成Docker， 也不行

另外，首先你要确认你的节点是否支持UDP，节点不支持，在客户端怎么折腾都没用。

liliass, 2020-04-16:
> > > > ssr redir udp error.restart 请问作者了，这里不能解决
> > > 
> > > 
> > > 的确，装了个未集成DOCKER的固件，UDP转发失败的问题解决了，还请大佬看下这个问题。
> > 
> > 
> > 你是用什么工具测试UDP的 ???
> > 我的版本没集成Docker， 也不行
> 
> 另外，首先你要确认你的节点是否支持UDP，节点不支持，在客户端怎么折腾都没用。

确定节点能UDP.   
你是在SSR+ 和passwall 都可以吗？
你的UDP 是什么协议？  SS,SSR,V2,Tor?

feheadls, 2020-04-16:
SSR+和PASSWALL都可以。我用的SS和SSR的节点都可以。如果你不可以，看看自己分流设置对了没，不确定可以用大陆白名单模式，这个NAT类型测试工具的流量就会通过插件节点，另外注意一下设置为全局端口转发，不然非常规端口不会通过插件节点转发。

liliass, 2020-04-16:
> SSR+和PASSWALL都可以。我用的SS和SSR的节点都可以。如果你不可以，看看自己分流设置对了没，不确定可以用大陆白名单模式，这个NAT类型测试工具的流量就会通过插件节点，另外注意一下设置为全局端口转发，不然非常规端口不会通过插件节点转发。

好吧，我再试试

bzzjh, 2020-04-19:
我也遇到这个问题，主要是玩switch游戏的时候，之前NAT类型都是A，最近我升级了openwrt源代码编译了最新的固件导致NAT类型变成了B，于是，我测试了，用老大release 4月8号(https://github.com/coolsnowwolf/lede/releases/tag/20200408)的代码编译或者直接用x86的img文件虚拟机测试了，没有问题，我用Nattype tester 测试了，ssr plus 开启UDP模式与主服务器相同，
[image]

然后端口可以暴力一点使用全部端口，测试了所有模式 包括trojan节点,socks5节点模式下，Nattype tester 都可以显示我VPS节点IP和NAT 类型fullcone.
[image]
这边大家也可以测试一下看看！！！

bzzjh, 2020-04-19:
ssr plus ipk版本：
[image]

bzzjh, 2020-04-19:
目前感觉大雕的4月8号的代码用起来还是比较稳定的，之前因为socks5代理节点https://github.com/coolsnowwolf/lede/issues/4299
无法使用我回退过一次，看来这一次又要回退一次了

bzzjh, 2020-04-19:
非常奇怪，同样的配置同样的版本，为啥x86没问题，N1的arm有问题呢？

bzzjh, 2020-04-19:
> 非常奇怪，同样的配置同样的版本，为啥x86没问题，N1的arm有问题呢？

原因找到了，X86接口没有启用br-lan,N1启用了，删除N1的br-lan保证是eth0就没问题了。

feheadls, 2020-04-19:
> 我也遇到这个问题，主要是玩switch游戏的时候，之前NAT类型都是A，最近我升级了openwrt源代码编译了最新的固件导致NAT类型变成了B，于是，我测试了，用老大release 4月8号
>
>
> 的代码编译或者直接用x86的img文件虚拟机测试了，没有问题，我用Nattype tester 测试了，ssr plus 开启UDP模式与主服务器相同，
> [image] [...]

楼主反馈的问题，和你说的问题并不一样，这里我和楼主反馈的是UDP流量无法通过SSR+的节点转发，开启插件的UDP转发也同样是UDP BLOCKED的状态，我自己测试的结果不光是SSR+无法转发，另一个插件PASSWALL也一样，所以我认为这个问题跟插件本身没有太大关系，后来偶然得知编译固件不要把docker整合到固件中去，就可解决这个问题，我就用了个不含docker的固件版本，的确解决了问题。猜测两者存在冲突，或两者同时存在的时候，防火墙的配置不满足两者同时存在的情况。

liliass, 2020-04-29:
我重新编制了一个新固件。 没把Docker加入， 还是不行啊。。。 很奇怪。
难道你们说的docker不是在luci里applications 》 luci-app-dockerman ?

Luci > Libraries > Luci-Lib-Docker 我也没打勾啊。

liliass, 2020-05-01:
可以了， 原来在”增强游戏模式客户端LAN IP“ 需要添加IP才能用。

closed 2020-07-04 by coolsnowwolf