coolsnowwolf/lede #4295: 求助，希望搞个能在qnap威联通453bmini上用的rtl8156网卡驱动。谢谢了

qingxuan111, 2020-04-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
求救各位大神，我是威联通的的453bmini，买了2.5g的绿联的用这个rtl8156芯片的usb网卡，在威联通上不能用，不过我看到有群晖上的，还有linux的源码的，有高手可以帮忙编译下吗。根本不会编程。买了两个网卡划了三百，不想浪费啊。谢谢高手了

qingxuan111, 2020-04-12:
哎。。。找了好久，找到这里，不知道会否有人帮忙，有人能看到不能。谢谢了。谢谢

hxlls, 2020-04-13:
走错门了，这是openwrt 和威联通没啥关系，请出门右转，找威联通客服

qingxuan111, 2020-04-13:
可是我再这个社区看到群晖的驱动啊。。。不是这个社区吗。。。汗。帮帮忙

Promix953, 2020-04-13:
> 可是我再这个社区看到群晖的驱动啊。。。不是这个社区吗。。。汗。帮帮忙

找错地方了，不是这个地方，买前不看兼容列表的吗？

qingxuan111, 2020-04-14:
https://github.com/bb-qq/r8152   我不懂。不过这个网址里面就有8156可以用在群晖上的驱动啊。这个不是这个版块吗。。。谢谢中国同胞了。哎。。

Aes64X, 2020-04-15:
> https://github.com/bb-qq/r8152 我不懂。不过这个网址里面就有8156可以用在群晖上的驱动啊。这个不是这个版块吗。。。谢谢中国同胞了。哎。。

老哥，你真的走错门了，去找威联通的人。

closed 2020-04-15 by coolsnowwolf