coolsnowwolf/lede #4296: 订阅节点重复

chinassr, 2020-04-12:
订阅节点出现重复现象，总共37个节点，每点一次订阅他就会多几个重复节点，
[image]
[image]
[image]

chinassr, 2020-04-12:
只有订阅里有一个节点，群组名不一样就会出现重复节点，我用别的科学上网插件，没有这个问题

chinassr, 2020-04-12:
我把所有节点群组名改成一样还是重复
[image]

chinassr, 2020-04-13:
用其他插件并没有出现重复节点的情况

coolsnowwolf, 2020-04-15:
这个问机场了，机场的订阅端输出的顺序每次都不同

例如第一次为 server = xx ，port =yy

第二次为 port =yy, server =xx 

这样就是两个不同的节点了

closed 2020-07-04 by coolsnowwolf