coolsnowwolf/lede #4304: x86 legacy 编译amule和kodexplorer

soundghost, 2020-04-13:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
A：
编译x86 legacy版本的amule和kodexplorer都会失败，尝试了无数次了。
**amule的失败日志为：**
touch /home/my/lede/build_dir/target-i386_pentium_musl/aMule-2.3.2/省略部分代码_check
./home/my/lede/include/shell.sh; bzcat /home/my/lede/dl/aMule-2.3.2-省略部分代码.tar.bz2 | tar -C /home/my/lede/build_dir/target-i386_pentium_musl/aMule-2.3.2/.. -xf -
**tar：this does not look like a tar archieve
tar：exiting with failure status due to previous errors**
**kodexplorer的错误为：**
make[3]: *** [ext/curl/curl.la] Error 1
make[3]: Leaving directory '/home/my/lede/build_dir/target-i386_pentium_musl/php-7.2.9'
make[2]: *** [/home/my/lede/build_dir/target-i386_pentium_musl/php-7.2.9/.built] Error 2
make[2]: Leaving directory '/home/my/lede/feeds/packages/lang/php7'
time: 省略
make[1]: *** [packages/feeds/packages/php7/compile] Error 2
make[1]: Leaving directory '/home/my/lede'
make: *** [package/lean/luci-app-kodexplorer/compile] Error 2

kodexplorer的错误我在[https://github.com/coolsnowwolf/lede/issues/4203#issuecomment-610875119](https://github.com/coolsnowwolf/lede/issues/4203#issuecomment-610875119)也看到有网友留言，但未见lean大回复。

### (2) 路由器型号和固件版本

A：x86-legacy。最新2020-04-13源码（其实这两个app的错误一直都存在未修复）

hyird, 2020-04-13:
文件残了 make clean

soundghost, 2020-04-15:
> 文件残了 make clean
谢谢您的回复。
make clean 这一步骤有做过，并且每一次编译固件前都会做这个步骤，可是依然出现问题。

coolsnowwolf, 2020-04-15:
梯子挂了，删除 dl 目录里面的文件

soundghost, 2020-04-16:
谢谢L大回复，刚刚查验了一下aMule的大小果然不对劲，删除aMule后重下载后的大小为4.1M，这下正常了。
也谢谢hyird。

Blueskyhou, 2020-04-16:
x86 action编译 加 kodexplorer 模块 也失败，最后去除了，就可以了。

soundghost, 2020-04-18:
> x86 action编译 加 kodexplorer 模块 也失败，最后去除了，就可以了。

确实是这样的。kodexplorer目前编译还是会失败，而且好像在mtk平台的762x下也会失败，不仅仅是x86

xynu617, 2020-04-27:
很长时间了，32位X86只要编译php就会报错

hyird, 2020-04-28:
32位 编译php铁定报错

closed 2020-07-04 by coolsnowwolf