coolsnowwolf/lede #4308: 添加后服务器后，check google失败，2020-4-13（r2448-b97c4e8b）

shizhai, 2020-04-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
有类似的issue，但是方案均已经测试过没有效果
类似的issue均指向为dns代理存在问题，我看了log与ps，dns代理正常

## 2. 详细叙述

### (1) 具体问题

A：添加服务器与订阅之后，订阅可以正常拉取，并且服务器均可以ping通
B：服务器在手机与电脑上测试正常，使用的是v2rayN与shadowsocket

### (2) 路由器型号和固件版本

A：路由器型号为raspberry pi 3b+
B：固件版本为git仓库中最新版本2020-4-13（r2448-b97c4e8b）
[log.txt](https://github.com/coolsnowwolf/lede/files/4470132/log.txt)

### (3) 详细日志

A：参见附件

coolsnowwolf, 2020-04-14:
5335 端口被dnsforward 占用了

shizhai, 2020-04-14:
你好，那条日志是我有pdnsd换成dnsforword出现的，后面重启了就没有那个错误了，但是还是不能FQ。我测试过更换多个服务器，订阅也可以正常过工作。以及页面上的选项组合，最好的是可以更新GWFlist成功，check google仍然失败

1049066154, 2020-04-14:
> 你好，那条日志是我有pdnsd换成dnsforword出现的，后面重启了就没有那个错误了，但是还是不能FQ。我测试过更换多个服务器，订阅也可以正常过工作。以及页面上的选项组合，最好的是可以更新GWFlist成功，check google仍然失败

清除dns缓存

shizhai, 2020-04-14:
嗨，你好，我直接在openwrt里面ping ip地址也不通呢，dns我已经将openwrt的dns从pdnsd换成dnsforword了。
请问你的意思是两家里的光猫的dns清除一下吗？

hxlls, 2020-04-14:
SSRP +是独立的DNS 机制，由pdns进行查询的和openwrt 机制不一样，openwrt 是用的dnsmasq ,你这种都没明白就乱改，出问题正常啊

shizhai, 2020-04-14:
哦哦，谢谢，我最开始按照readme直接编译出来的不能工作，按照网上的方案手动重启SSR，换机场，都测试了不得行，最后看到有人讲将pdnsd换成foword可以，我才换的。空了我把dnsmasq等其他的都杀掉看看

1049066154, 2020-04-14:
ping不通是正常的,因为ping是icmp协议,一般不会经过代理,换回Pdnsd,然后电脑上在cmd下运行 ipconfig /flushdns,手机开关一下飞行模式

shizhai, 2020-04-18:
问题已经解决，感谢。解决方案，先同步时间再添加服务器，再清dns缓存

closed 2020-07-04 by coolsnowwolf