coolsnowwolf/lede #4309: 假设谷歌云vps抽风半小时。失去了翻墙的能力，lede能否保持原有未被污染的国外域名dns缓存不被国内dns污染

norikohxb, 2020-04-14:
假设谷歌云vps抽风半小时。失去了翻墙的能力，lede能否保持原有未被污染的国外域名dns缓存不被国内dns污染

我都没有用第三方的dns工具就用lean的lede的dns因为他已经提供了足够强大的抗国内dns服务器对于youtube这些域名的污染的能力。
当然这一切的前提都要架构在你的vps是好的，比如你的trojan是无故障的。
我现在想要说的，是当短期的vps抽风，大家都用过谷歌云，其实他偶尔也会抽风的，抽风的时候trojan连接应该就不是很通畅了，这个时候可能会造成lede的dns给国内isp的dns同化污染掉。因为没有强大的8.8.4.4的vps直连的防污染。

我在想lean大的lede能否在你的vps在这次翻墙连接故障到下一期连接故障之前，已经缓存在dnsmasq的大陆白名单外的国外域名的ip不受任何国内的dns影响。

不知道能否做到？

0x5826, 2020-04-14:
你的意思是在国外不返回结果的情况下，就不解析了？

1049066154, 2020-04-14:
都失去翻墙能力了还在乎啥污染?

norikohxb, 2020-04-14:
> 都失去翻墙能力了还在乎啥污染?
既然用了lede肯定是80%的时间要保持翻墙的状态，但是谷歌云vps又不是一直都没有问题，假设有10分钟-30分钟的抽风时候，我希望保持国外dns不受污染是因为如果中间受到污染了一些老旧的安卓5.0的设备，可能会去调用污染的域名所对应的ip地址，当半个小时谷歌云不抽风的时候。那些老旧的安卓设备可能还用之前污染的域名是，变成要关闭网卡再启用网卡来清空缓存。意思就是这半个小时扛住，不要让国内的dns污染了这些被墙网站的ip

norikohxb, 2020-04-14:
> 你的意思是在国外不返回结果的情况下，就不解析了？

我的意思是当谷歌vps短暂性失去作用的时候比如10分钟到1个小时，已经缓存在lede里面的国外的域名已经有ip地址的就不再用国内的域名刷新（因为一旦用国内的114或者是运营商的，马上就给污染了）仅仅是已经访问过的有在缓存的大陆白名单的国外域名，在翻墙渠道失灵以后不再更新，保持到下一次翻墙渠道通畅。不知道这样子建议会不会对其他人的正常使用造成什么影响。

norikohxb, 2020-04-14:
其实能给政府墙掉的域名，给他们曲解了ip的域名，其实也是访问不了的。不然政府为什么处心积虑要墙壁掉，既然暂时失去了翻墙的能力（我说是暂时，如果是永久是去，当然有没有给污染都无所谓），反正那些网站都是看不了的，横竖都无法访问，就保留失效之前已经访问过得没有给污染的那些域名对应的ip，不是个好主意吗，至少下一次恢复的时候。一些比较老旧的安卓5.0的平板不会因为污染变成还要关闭网卡再启用一次。我大体上是这个意思了。

hxlls, 2020-04-14:
保持时间太长是会有问题的，你缓存48小时，着48小时内哪个地址IP变了呢？你又打算怎么处理？

norikohxb, 2020-04-14:
缓存的那个ip至少是上一次翻墙能力健在的时候准确的正确的ip，保持到下一次翻墙渠道重新畅通为止再刷新。这样，对于一些老旧设备，或者电视盒子不会因为中间的污染而污染到缓存了，这个缓存已经访问过的百名点的ip缓存到下一次翻墙能力恢复重新更新。或者重启。

norikohxb, 2020-04-14:
> 保持时间太长是会有问题的，你缓存48小时，着48小时内哪个地址IP变了呢？你又打算怎么处理？

反正翻墙能力失去了其实就好像一个性能能的人，只是说已经有的缓存就不要给污染了。至少还有下次的黎明 。这个就好像是风雨过后总有彩虹，我是感觉有时候如果国内防火墙力度好的时候谷歌云也会有间歇性失灵。就是不想要在失灵的时候lede上面的国外域名就给趁机玷污了，这个好像就可以理解，你就算暂时阳痿也不希望已经娶过门的小娘们给国内的防火墙玷污了。下次恢复元气的时候可以继续享受。

hxlls, 2020-04-14:
首先PDNS 支持TCP 查询，并且ssrp 是用pdns 出外去查询DNS 服务器并缓存，TCP被劫持的可能性很低所以污染大部分源自于你的PDNS 查询服务器，其次，你要保存这个解析，直接设定host 文件不就完事了？几年都行

norikohxb, 2020-04-14:
> 首先PDNS 支持TCP 查询，并且ssrp 是用pdns 出外去查询DNS 服务器并缓存，TCP被劫持的可能性很低所以污染大部分源自于你的PDNS 查询服务器，其次，你要保存这个解析，直接设定host 文件不就完事了？几年都行

首先PDNS 支持TCP 查询，并且ssrp 是用pdns 出外去查询DNS 服务器并缓存，TCP被劫持的可能性很低所以污染大部分源自于你的PDNS 查询服务器

我其实是同意你的说法，只要vps正常，根本就不会存在问题，就是怕某个阶段因为国内的什么会议召开，防火墙紧缩。造成翻墙不流畅的时候，跟着给污染嘛，自己加host来解决或者加固定也是一个办法，但是如果说当我们在ssrplus里面填写的翻墙服务器没有一个是连接正常的时候，已经访问的国外域名的缓存就顺势不做更新，不是会更容易在代码实现嘛。你想把，访问一个youtube打开一个网页和视频，其实是很多不同域名的，比如那个videogoogle的域名，是媒体流调用的域名。而且不同分辨率的youtube视频还放在不同域名的服务器上。
如果说ssr里面填写的翻墙工具暂时没有一个可以用，已经缓存的已经访问过的国外域名就不去做刷新了也不用用国内的dns从上游服务器刷新，。

我只是提出自己的看法，如果很难实现那就暂时先不考虑，毕竟其实如你说的用其他方式，比如写入host也是可以实现的

norikohxb, 2020-04-14:
> 保持时间太长是会有问题的，你缓存48小时，着48小时内哪个地址IP变了呢？你又打算怎么处理？

感谢你咯，很多疑问帖子都是你在回复的

1049066154, 2020-04-14:
> > 都失去翻墙能力了还在乎啥污染?
> > 既然用了lede肯定是80%的时间要保持翻墙的状态，但是谷歌云vps又不是一直都没有问题，假设有10分钟-30分钟的抽风时候，我希望保持国外dns不受污染是因为如果中间受到污染了一些老旧的安卓5.0的设备，可能会去调用污染的域名所对应的ip地址，当半个小时谷歌云不抽风的时候。那些老旧的安卓设备可能还用之前污染的域名是，变成要关闭网卡再启用网卡来清空缓存。意思就是这半个小时扛住，不要让国内的dns污染了这些被墙网站的ip

被墙的基本上连IP都给你阻断了,如果只是DNS污染,那么只需要解决DNS污染就行了,根本不需要经过代理

norikohxb, 2020-04-14:
> > > 都失去翻墙能力了还在乎啥污染?
> > > 既然用了lede肯定是80%的时间要保持翻墙的状态，但是谷歌云vps又不是一直都没有问题，假设有10分钟-30分钟的抽风时候，我希望保持国外dns不受污染是因为如果中间受到污染了一些老旧的安卓5.0的设备，可能会去调用污染的域名所对应的ip地址，当半个小时谷歌云不抽风的时候。那些老旧的安卓设备可能还用之前污染的域名是，变成要关闭网卡再启用网卡来清空缓存。意思就是这半个小时扛住，不要让国内的dns污染了这些被墙网站的ip
> 
> 被墙的基本上连IP都给你阻断了,如果只是DNS污染,那么只需要解决DNS污染就行了,根本不需要经过代理
gcp只是暂时抽风，我并不是说解决dns污染，，因为如果一旦你的ssr里面的trojan或者v2ray都失效的话，这个时候你要解决dns污染恐怕要用另外一种协议的dns解析。我不是说让lede解决dns污染，都已经丧失了vps的支持，你如何解决污染，我的意思是说之前解析的域名（代表之前你经常用的那些国外的网站）的ip，仍然保留在缓存里面，等待下一次gcp恢复

norikohxb, 2020-04-14:
> > > 都失去翻墙能力了还在乎啥污染?
> > > 既然用了lede肯定是80%的时间要保持翻墙的状态，但是谷歌云vps又不是一直都没有问题，假设有10分钟-30分钟的抽风时候，我希望保持国外dns不受污染是因为如果中间受到污染了一些老旧的安卓5.0的设备，可能会去调用污染的域名所对应的ip地址，当半个小时谷歌云不抽风的时候。那些老旧的安卓设备可能还用之前污染的域名是，变成要关闭网卡再启用网卡来清空缓存。意思就是这半个小时扛住，不要让国内的dns污染了这些被墙网站的ip
> 
> 被墙的基本上连IP都给你阻断了,如果只是DNS污染,那么只需要解决DNS污染就行了,根本不需要经过代理

或者可以这么理解，局域网设备的dns基本上都是lede这个op，在gcp短暂性时效的时候，比如10分钟到2个小时，这个时间段内，已经缓存的非大陆地区的域名的ip就不要去从国内isp那边解析了，之前的可以用的就暂时保留。不会去让114.114.114.114来覆盖。是这个意思，，诚如你说的，翻墙都翻不过去了，解析正确有啥用。换个角度理解，翻墙都翻不过去了，原来的缓存的域名解析的ip是正确的也没有必要再去114索取，虽然你现在gcp暂时断开了，你就算去114索取不管得到的最新的是污染过的，还是和之前一样的没有给政府染指，反正横竖都上不去对吧。既然都上不去，国外域名这块就暂时不要解析了啊。除非你关闭ssr对吗。因为我们ssr如果设置了大陆白名单，如果说gcp暂时挂了，其实国外网络也都上不去了，既然上不去，你去找本地isp解析有意义吗。正因为都没有意义了就不去更新他了。也省的多做一次运算。到了下一次gcp、vps恢复以后再更新。

1049066154, 2020-04-14:
> 都失去翻墙能力了还在乎啥污染? [...]

解决dns污染的方法多了去了,另外,在GFW列表或域名黑名单里的域名都是通过SSR中设置的DNS程序去解析和缓存的,不管服务器能不能访问都不会经过国内dns和运营商dns去解析,在服务器挂了的情况下只会出现无法解析的情况,不会出现被污染的情况,除非你的dns设置不对
[image]

Promix953, 2020-04-14:
> 解决dns污染的方法多了去了,另外,在GFW列表或域名黑名单里的域名都是通过SSR中设置的DNS程序去解析和缓存的,不管服务器能不能访问都不会经过国内dns和运营商dns去解析,在服务器挂了的情况下只会出现无法解析的情况,不会出现被污染的情况,除非你的dns设置不对
> [image]

的确是这个情况，所以我都懒得回复，浪费时间和精力

norikohxb, 2020-04-14:
> 的确是这个情况，所以我都懒得回复，浪费时间和精力
感谢大佬的热心解答，顿开茅厕，谢谢啦

norikohxb, 2020-04-14:
> 解决dns污染的方法多了去了,另外,在GFW列表或域名黑名单里的域名都是通过SSR中设置的DNS程序去解析和缓存的,不管服务器能不能访问都不会经过国内dns和运营商dns去解析,在服务器挂了的情况下只会出现无法解析的情况,不会出现被污染的情况,除非你的dns设置不对
> [image]

感谢大佬的热心解答，顿开茅厕，谢谢啦

closed 2020-04-14 by norikohxb