coolsnowwolf/lede #4312: qBittorrent 提示：无效的用户名或密码

u0225, 2020-04-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

qBittorrent Web UI 提示：无效的用户名或密码，可是都是和以前一样的设置，以前的就可以，昨天更新后编译的就登录不了

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

n270
OpenWrt R20.4.8 / LuCI Master (git-20.088.29068-cebe120)
4.19.108

A：

### (3) 详细日志

A：

stock169, 2020-04-14:
新版不支持了，设置子网白名单用吧

u0225, 2020-04-14:
原来是这样，那就用其他的吧

closed 2020-04-14 by u0225

huangjackey1111, 2020-05-27:
在白名单填 0.0.0.0/0 就全部ipv4允许了

zjt2005, 2020-07-25:
> ## 在白名单填 0.0.0.0/0 就全部ipv4允许了
> 这样是全部免登陆，然后进入到qBittorrent，修改密码，在把白名单不要勾，保存。再进去就可以用那个密码登陆了。应该是config中保存的密码用了新的加密方式。

不止一次遇到了相同的问题，每一次都是删光配置文件重新来过，这次无伤解决问题，谢谢！

miyouzi, 2021-10-14:
主要注意：用户名固定 `admin` ，因此修改用户名是无效的，仅能修改密码

steamch, 2022-05-10:
> ## 在白名单填 0.0.0.0/0 就全部ipv4允许了
> 这样是全部免登陆，然后进入到qBittorrent，修改密码，在把白名单不要勾，保存。再进去就可以用那个密码登陆了。应该是config中保存的密码用了新的加密方式。

最新版https+域名访问webui这样操作会出现无法保存程序偏好选项，可能是无法连接到 qBttorrent。
请教这种又要怎么解决

ccjackcong, 2023-04-10:
> 这样是全部免登陆，然后进入到qBittorrent，修改密码，在把白名单不要勾，保存。再进去就可以用那个密码登陆了。应该是config中保存的密码用了新的加密方式。

学到了。谢谢

morpuct, 2023-11-19:
在容器里面把QB重启也能登陆。