coolsnowwolf/lede #4317: ubuntu18.04 使用默认配置 编译x86_64 失败

jiatongwu, 2020-04-14:
ubuntu18.04 使用默认配置 编译x86_64 失败

g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0/gcc -I/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0/gcc/. -I/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0/gcc/../include -I/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0/gcc/../libcpp/include -I/home/ubuntu/lede/staging_dir/host/include -I/home/ubuntu/lede/staging_dir/host/include -I/home/ubuntu/lede/staging_dir/host/include  -I/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0/gcc/../libdecnumber -I/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0/gcc/../libbacktrace   -o insn-emit.o -MT insn-emit.o -MMD -MP -MF ./.deps/insn-emit.TPo insn-emit.c

g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1099: recipe for target 'insn-emit.o' failed
make[5]: *** [insn-emit.o] Error 4
make[5]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-initial/gcc'
Makefile:4209: recipe for target 'all-gcc' failed
make[4]: *** [all-gcc] Error 2
make[4]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-initial'
Makefile:36: recipe for target '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-initial/.built' failed
make[3]: *** [/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-initial/.built] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#618.66#854.19#8053.39
toolchain/Makefile:98: recipe for target 'toolchain/gcc/initial/compile' failed
make[2]: *** [toolchain/gcc/initial/compile] Error 2
make[2]: Leaving directory '/home/ubuntu/lede'
toolchain/Makefile:96: recipe for target '/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
/home/ubuntu/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

jiatongwu, 2020-04-14:
网上说是内存不足造成的 、

jiatongwu, 2020-04-14:
增加swap文件，解决，ok
 31  dd if=/dev/zero of=/swap4G bs=1024 count=4000000
   32  chown 0600 /swap4G
   33  mkswap /swap4G 
   34  swapon /swap4G
   35  free -h

ckxt, 2020-04-15:
既然解决了 就把这个issue关掉呗

closed 2020-04-15 by jiatongwu

simplerick-simplefun, 2022-06-19:
> 增加swap文件，解决，ok 31 dd if=/dev/zero of=/swap4G bs=1024 count=4000000 32 chown 0600 /swap4G 33 mkswap /swap4G 34 swapon /swap4G 35 free -h

谢谢。
第二行应该是chmod吧？