coolsnowwolf/lede #4319: 最近的版本7621又没无线了

z826540272, 2020-04-14:
每次开机必须手动开启才行
可能和mac 802.11 包更新有关

hcym, 2020-04-14:
7261还是毛子做AP，就看hanwckf那边的trojan搞好后，最后一个新三也刷毛子

hcwang, 2020-04-15:
在luci的启动项那个菜单中，在最下面的文件中，在exit 0前增加如下一行就可以启动wifi了。

```
nohup sleep 60 && /sbin/wifi up &
```

hcwang, 2020-04-15:
使用以上命令后，在你重启结束后，LED灯变绿后，大约90－100秒左右WIFI信号会重新出现。

coolsnowwolf, 2020-04-16:
有日志么

hcwang, 2020-04-16:
> 有日志么

应该就是mac80211模块的问题，我更新到最新版就正常了。

z826540272, 2020-04-16:
怎么更新

hcwang, 2020-04-16:
> 怎么更新
> […](#)

去 https://github.com/openwrt/openwrt 上将package/kernel 中的mac80211最新的模块放在本地编译目录中试试。

Aes64X, 2020-04-17:
> 有日志么

关键日志：
[  126.624461] device wlan0 left promiscuous mode
[  126.629105] br-lan: port 3(wlan0) entered disabled state
[  127.693333] netlink: 4 bytes leftover after parsing attributes in process `iw'.
[  133.615640] device wlan1 left promiscuous mode
[  133.620252] br-lan: port 2(wlan1) entered disabled state
[  134.840376] netlink: 4 bytes leftover after parsing attributes in process `iw'.
[  187.219821] netlink: 4 bytes leftover after parsing attributes in process `iw'.
[  237.267184] netlink: 4 bytes leftover after parsing attributes in process `iw'.
[  256.717447] netlink: 4 bytes leftover after parsing attributes in process `iw'.

有些指定信道不出现，有些信道制定就会出现类似日志。
具体表现为：无线页面出现Wifi相关的信道、速率、强度、加密方式等信息，但客户端在实际搜索时无法搜索到wifi SSID
测试设备：Newifi D1

AmadeusGhost, 2020-04-23:
可以试下我的pr：#4420

hcwang, 2020-04-23:
> 可以试下我的pr：#4420

感谢，坐等merge.

kk3375, 2020-04-24:
> 可以试下我的pr：[＃4420](https://github.com/coolsnowwolf/lede/pull/4420)

这个更新之前，新3无线可以正常开启，今天重新编译后，新3扫描无线加入后，无线无法开启，停止后再启用还是无法开启无线

specp, 2020-04-26:
> > 可以试下我的pr：[＃4420](https://github.com/coolsnowwolf/lede/pull/4420)
> 
> 这个更新之前，4.17号编译的新3无线可以正常开启，今天重新编译后，新3扫描无线加入后，无线无法开启，停止后再启用还是无法开启无线

是作为客户端吗？我这里新编译的作为AP没问题，但是客户端扫描加入无线就是一直不启动

kk3375, 2020-04-26:
> > > 可以试下我的pr：[＃4420](https://github.com/coolsnowwolf/lede/pull/4420)
> > 
> > 
> > 这个更新之前，4.17号编译的新3无线可以正常开启，今天重新编译后，新3扫描无线加入后，无线无法开启，停止后再启用还是无法开启无线
> 
> 是作为客户端吗？我这里新编译的作为AP没问题，但是客户端扫描加入无线就是一直不启动

对的，就是这个更新后，新3重新编译后无线最为客户端扫描加入无线就是一直不启动，等大佬回退或修复

AmadeusGhost, 2020-04-26:
> 对的，就是这个更新后，新3重新编译后无线最为客户端扫描加入无线就是一直不启动，等大佬回退或修复

你可以试下我在树莓派那个issue那提到的那个commit：https://github.com/AmadeusGhost/lede/commit/4a60fe4464e19b6ee29b03b1be96fa8571ccfe59

specp, 2020-04-26:
> > > > 可以试下我的pr：[＃4420](https://github.com/coolsnowwolf/lede/pull/4420)
> > > 
> > > 
> > > 这个更新之前，4.17号编译的新3无线可以正常开启，今天重新编译后，新3扫描无线加入后，无线无法开启，停止后再启用还是无法开启无线
> > 
> > 
> > 是作为客户端吗？我这里新编译的作为AP没问题，但是客户端扫描加入无线就是一直不启动
> 
> 对的，就是这个更新后，新3重新编译后无线最为客户端扫描加入无线就是一直不启动，等大佬回退或修复

我自己回退本地克隆到merge之前，能用是能用了，但是速度极其慢，链路速度虽然显示很快，但实际接入连上级路由的管理页都打不开，一直加载

Aes64X, 2020-04-26:
> > > > > 可以试下我的pr：[＃4420](https://github.com/coolsnowwolf/lede/pull/4420)
> > > > 
> > > > 
> > > > 这个更新之前，4.17号编译的新3无线可以正常开启，今天重新编译后，新3扫描无线加入后，无线无法开启，停止后再启用还是无法开启无线
> > > 
> > > 
> > > 是作为客户端吗？我这里新编译的作为AP没问题，但是客户端扫描加入无线就是一直不启动
> > 
> > 
> > 对的，就是这个更新后，新3重新编译后无线最为客户端扫描加入无线就是一直不启动，等大佬回退或修复
> 
> 我自己回退本地克隆到merge之前，能用是能用了，但是速度极其慢，链路速度虽然显示很快，但实际接入连上级路由的管理页都打不开，一直加载

更新后我使用R3G是没什么问题。。。但这个5.4版的MAC80211相较于4.19版性能上实在是。。。。我不清楚是不是因为驱动的原因，速率不高，在ipq4019上测试时甚至能低到180+带宽

specp, 2020-04-26:
> > 对的，就是这个更新后，新3重新编译后无线最为客户端扫描加入无线就是一直不启动，等大佬回退或修复
> 
> 你可以试下我在树莓派那个issue那提到的那个commit：[AmadeusGhost@4a60fe4](https://github.com/AmadeusGhost/lede/commit/4a60fe4464e19b6ee29b03b1be96fa8571ccfe59)

试了，依旧没法作为客户端加入网络

kk3375, 2020-04-26:
> > > 对的，就是这个更新后，新3重新编译后无线最为客户端扫描加入无线就是一直不启动，等大佬回退或修复
> > 
> > 
> > 你可以试下我在树莓派那个issue那提到的那个commit：[AmadeusGhost@4a60fe4](https://github.com/AmadeusGhost/lede/commit/4a60fe4464e19b6ee29b03b1be96fa8571ccfe59)
> 
> 试了，依旧没法作为客户端加入网络

新3编译了确实不能用，4.17号编译的固件一切正常没有问题
[image]

PR #4518 (2020-05-02): netifd: add dynamic wireless reconfiguration
Set new option 'reconf' in 'wifi-device' section to enable dynamic re-configuration on that radio. Also fix wifi relay and 'netifd: radio1 (9654): Command failed'.
Run-tested-by: @hyird 
fix #4319 , fix #4390 , fix #4474 , fix #4503 , fix #4516
closed 2020-05-02 by coolsnowwolf