coolsnowwolf/lede #4320: 取消无线驱动就会编译失败

lloyd18, 2020-04-14:
默认编译了不少无线驱动，因为用不着，所以把 Wireless Drivers  --->里的驱动全部取消，结果编译失败，大佬看看怎么解决，下面是日志
  make[4]: warning: -jN forced in submake: disabling jobserver mode.
/home/lan/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/7.5.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lan/lede/tmp/cc33DHCO.ltrans5.ltrans.o: in function `br_addif':
<artificial>:(.text.br_addif+0x61): undefined reference to `linux_br_add_if'
/home/lan/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/7.5.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lan/lede/tmp/cc33DHCO.ltrans5.ltrans.o: in function `vlan_get_bridge':
<artificial>:(.text.vlan_get_bridge+0x7b): undefined reference to `linux_br_add'
/home/lan/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/7.5.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lan/lede/tmp/cc33DHCO.ltrans5.ltrans.o: in function `br_delif':
<artificial>:(.text.br_delif+0x62): undefined reference to `linux_br_del_if'
/home/lan/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/7.5.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lan/lede/tmp/cc33DHCO.ltrans5.ltrans.o: in function `vlan_put_bridge':
<artificial>:(.text.vlan_put_bridge+0xa9): undefined reference to `linux_br_del'
/home/lan/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/7.5.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lan/lede/tmp/cc33DHCO.ltrans12.ltrans.o: in function `ap_sta_add':
<artificial>:(.text.ap_sta_add+0x1d3): undefined reference to `sta_track_claim_taxonomy_info'
collect2: error: ld returned 1 exit status
Makefile:596: recipe for target '/home/lan/lede/build_dir/target-x86_64_musl/hostapd-wpad-full-internal/hostapd-2019-08-08-ca8c2bd2/.built' failed
make[3]: *** [/home/lan/lede/build_dir/target-x86_64_musl/hostapd-wpad-full-internal/hostapd-2019-08-08-ca8c2bd2/.built] Error 1
make[3]: Leaving directory '/home/lan/lede/package/network/services/hostapd'
time: package/network/services/hostapd/wpad-full-internal/compile#15.68#1.51#17.42
package/Makefile:111: recipe for target 'package/network/services/hostapd/compile' failed
make[2]: *** [package/network/services/hostapd/compile] Error 2
make[2]: Leaving directory '/home/lan/lede'
package/Makefile:107: recipe for target '/home/lan/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/lan/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lan/lede'
Build failed - please re-run with -j1 to see the real error message
/home/lan/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 1

lxwq, 2020-04-15:
把和无线有关的组件全部去掉就可以了,像上面提示的hostapd

guoben1982, 2020-04-15:
https://www.right.com.cn/forum/thread-1892775-1-1.html
我按照这个帖子的操作，一切正常

lloyd18, 2020-04-15:
> https://www.right.com.cn/forum/thread-1892775-1-1.html
> 我按照这个帖子的操作，一切正常

OK，一切正常了。谢谢。

closed 2020-04-15 by lloyd18