coolsnowwolf/lede #4325: 无法安装kmod-usb-net-rndis 实现android usb网络共享 ，之前我那个rom是可以安装上的，用你这个源码 ，默认配置编译后不能安装这个软件了？

jiatongwu, 2020-04-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：无法安装kmod-usb-net-rndis 实现android usb网络共享 ，之前我那个rom是可以安装上的，用你这个源码 ，默认配置编译后不能安装这个软件了？
### (2) 路由器型号和固件版本

A：gl-iNet 6408A v1 openwrt

### (3) 详细日志

A：
root@192.168.1.1's password: 

BusyBox v1.30.0 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r2452-3ad7a35c
 -----------------------------------------------------
root@OpenWrt:~# opkg update
Downloading https://openwrt.proxy.ustclug.org/snapshots/targets/ar71xx/generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
root@OpenWrt:~# opkg install kmod-usb-net kmod-usb-net-cdc-ether kmod-usb-net-rndis
Installing kmod-usb-net (4.14.125-1) to root...
Downloading https://openwrt.proxy.ustclug.org/snapshots/targets/ar71xx/generic/packages/kmod-usb-net_4.14.125-1_mips_24kc.ipk
Installing kmod-usb-net-cdc-ether (4.14.125-1) to root...
Downloading https://openwrt.proxy.ustclug.org/snapshots/targets/ar71xx/generic/packages/kmod-usb-net-cdc-ether_4.14.125-1_mips_24kc.ipk
Installing kmod-usb-net-rndis (4.14.125-1) to root...
Downloading https://openwrt.proxy.ustclug.org/snapshots/targets/ar71xx/generic/packages/kmod-usb-net-rndis_4.14.125-1_mips_24kc.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-usb-net:
 *      kernel (= 4.14.125-1-4798db7ce04dcb4337e44dbaa2a958a0)
 * opkg_install_cmd: Cannot install package kmod-usb-net.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-usb-net-cdc-ether:
 *      kernel (= 4.14.125-1-4798db7ce04dcb4337e44dbaa2a958a0)
 * opkg_install_cmd: Cannot install package kmod-usb-net-cdc-ether.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-usb-net-rndis:
 *      kernel (= 4.14.125-1-4798db7ce04dcb4337e44dbaa2a958a0)
 * opkg_install_cmd: Cannot install package kmod-usb-net-rndis.
root@OpenWrt:~# uname -a
Linux OpenWrt 4.9.215 #0 Tue Apr 14 01:25:59 2020 mips GNU/Linux
root@OpenWrt:~#

EnnawYang, 2020-04-15:
直接编译进去不行吗？

jiatongwu, 2020-04-15:
刚才 重新 进行make menuconfig了

jiatongwu, 2020-04-15:
root@OpenWrt:~# opkg install kmod-usb-net_4.9.215-1_mips_24kc.ipk 
Package kmod-usb-net (4.9.215-1) installed in root is up to date.
root@OpenWrt:~# rm -rf kmod-usb-net_4.9.215-1_mips_24kc.ipk 
root@OpenWrt:~# opkg install kmod-usb-net-cdc-ether_4.9.215-1_mips_24kc.ipk 
Package kmod-usb-net-cdc-ether (4.9.215-1) installed in root is up to date.
root@OpenWrt:~# opkg install kmod-usb-net-cdc-ether_4.9.215-1_mips_24kc^C
root@OpenWrt:~# rm -rf kmod-usb-net-
kmod-usb-net-cdc-ether_4.9.215-1_mips_24kc.ipk  kmod-usb-net-rndis_4.9.215-1_mips_24kc.ipk
root@OpenWrt:~# rm -rf kmod-usb-net-cdc-ether_4.9.215-1_mips_24kc.ipk 
root@OpenWrt:~# opkg install kmod-usb-net-rndis_4.9.215-1_mips_24kc.ipk 
Package kmod-usb-net-rndis (4.9.215-1) installed in root is up to date.
root@OpenWrt:~# 

安装自己编译的kmod ipk 解决了

closed 2020-04-15 by jiatongwu

zzzop, 2021-09-26:
同样问题，请问是如何编译ipk呢，只会改参数一键编译固件。。

> root@OpenWrt:~# opkg install kmod-usb-net_4.9.215-1_mips_24kc.ipk Package kmod-usb-net (4.9.215-1) installed in root is up to date. root@OpenWrt:~# rm -rf kmod-usb-net_4.9.215-1_mips_24kc.ipk
> root@OpenWrt:~# opkg install kmod-usb-net-cdc-ether_4.9.215-1_mips_24kc.ipk Package kmod-usb-net-cdc-ether (4.9.215-1) installed in root is up to date. root@OpenWrt:~# opkg install kmod-usb-net-cdc-ether_4.9.215-1_mips_24kc^C
> root@OpenWrt:~# rm -rf kmod-usb-net- kmod-usb-net-cdc-ether_4.9.215-1_mips_24kc.ipk kmod-usb-net-rndis_4.9.215-1_mips_24kc.ipk root@OpenWrt:~# rm -rf kmod-usb-net-cdc-ether_4.9.215-1_mips_24kc.ipk
> root@OpenWrt:~# opkg install kmod-usb-net-rndis_4.9.215-1_mips_24kc.ipk Package kmod-usb-net-rndis (4.9.215-1) installed in root is up to date. root@OpenWrt:~#
> 
> 安装自己编译的kmod ipk 解决了
> 
> root@OpenWrt:~# ip link show 1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1 link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00 2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN mode DEFAULT group default qlen 1000 link/ether 00:1b:c5:09:a2:2e brd ff:ff:ff:ff:ff:ff 3: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel master br-lan state DOWN mode DEFAULT group default qlen 1000 link/ether 00:1b:c5:09:a2:2e brd ff:ff:ff:ff:ff:ff 4: teql0: mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 100 link/void 6: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default qlen 1000 link/ether 00:1b:c5:09:a2:2e brd ff:ff:ff:ff:ff:ff 20: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DORMANT group default qlen 1000 link/ether 00:1b:c5:09:a2:2e brd ff:ff:ff:ff:ff:ff 21: wlan0-1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP mode DEFAULT group default qlen 1000 link/ether 02:1b:c5:09:a2:2e brd ff:ff:ff:ff:ff:ff 22: usb0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000 link/ether ba:14:04:bf:5d:be brd ff:ff:ff:ff:ff:ff root@OpenWrt:~#