coolsnowwolf/lede #4327: rinetd移植到openwrt实现

jiatongwu, 2020-04-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：rinetd软件比较好用 能不能帮忙加到官方源里面去
https://www.cnblogs.com/Silence-man/articles/7478850.html
http://jp2.wujiatong.group/rinetd.tar.gz
软件安装包在这里下载，我已经下载好了,linux下编译我可以，但是openwrt编译不会，哪位大神帮忙添加到官方源里边去？

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
https://www.cnblogs.com/Silence-man/articles/7478850.html
http://jp2.wujiatong.group/rinetd.tar.gz

coolsnowwolf, 2020-04-15:
不是很理解，这个用 iptables 的重定向就能实现啊？

jiatongwu, 2020-04-16:
iptables 好像不能重定向到 任意的 公网ip去
rinetd可以

coolsnowwolf, 2020-04-16:
iptables  可以重定向到公网ip 的，网易云音乐云解锁就是这样实现的

jiatongwu, 2020-04-16:
你说的是对的
我测试了一下 又加深了我对iptables的理解,感谢 @coolsnowwolf
root@OpenWrt:~#  iptables -t nat -A PREROUTING -s 192.168.1.175 -p tcp -m multiport --destination-ports 60081 -j DNAT --to 34.92.73.199:81
root@OpenWrt:~# 34.92.73.199^C
root@OpenWrt:~# iptables -t nat -A POSTROUTING -s 192.168.1.175 -d 34.92.73.199 -j MASQUERADE

closed 2020-04-16 by jiatongwu