coolsnowwolf/lede #4331: 单独编译adbybyplus等插件时显示no such file or directory

swhoro, 2020-04-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译adbybyplus等软件时出错，显示no such file or directory。在issues里面搜了一下，目录名不能超过127长度，但是我的目录应该没有超过

### (2) 路由器型号和固件版本

A：路由器N1，

固件
OpenWrt R20.4.8 / LuCI Master (git-20.088.29068-cebe120)
 5.4.32-flippy-34+o

### (3) 详细日志

A：
aiden@ubuntu:~/lede$ make package/luci-app-adbyby-plus/compile V=s
make[1]: Entering directory '/home/aiden/lede'
make[2]: Entering directory '/home/aiden/lede/feeds/luci/modules/luci-base'
make[2]: Leaving directory '/home/aiden/lede/feeds/luci/modules/luci-base'
time: package/feeds/luci/luci-base/host-compile#1.32#0.72#1.82
make[2]: Entering directory '/home/aiden/lede/package/libs/toolchain'
rm -rf /home/aiden/lede/build_dir/target-aarch64_generic_musl/toolchain/.pkgdir/libc.installed /home/aiden/lede/build_dir/target-aarch64_generic_musl/toolchain/.pkgdir/libc
mkdir -p /home/aiden/lede/build_dir/target-aarch64_generic_musl/toolchain/.pkgdir/libc
install -d -m0755 /home/aiden/lede/build_dir/target-aarch64_generic_musl/toolchain/.pkgdir/libc/lib /home/aiden/lede/build_dir/target-aarch64_generic_musl/toolchain/.pkgdir/libc/usr/bin
cp -fpR /home/aiden/lede/staging_dir/toolchain-aarch64_generic_gcc-7.5.0_musl/lib/ld-musl-*.so* /home/aiden/lede/build_dir/target-aarch64_generic_musl/toolchain/.pkgdir/libc/lib/
cp: cannot stat '/home/aiden/lede/staging_dir/toolchain-aarch64_generic_gcc-7.5.0_musl/lib/ld-musl-*.so*': No such file or directory
Makefile:618: recipe for target '/home/aiden/lede/build_dir/target-aarch64_generic_musl/toolchain/.pkgdir/libc.installed' failed
make[2]: *** [/home/aiden/lede/build_dir/target-aarch64_generic_musl/toolchain/.pkgdir/libc.installed] Error 1
make[2]: Leaving directory '/home/aiden/lede/package/libs/toolchain'
time: package/libs/toolchain/compile#0.13#0.06#0.18
package/Makefile:111: recipe for target 'package/libs/toolchain/compile' failed
make[1]: *** [package/libs/toolchain/compile] Error 2
make[1]: Leaving directory '/home/aiden/lede'
/home/aiden/lede/include/toplevel.mk:225: recipe for target 'package/luci-app-adbyby-plus/compile' failed
make: *** [package/luci-app-adbyby-plus/compile] Error 2
[image]

coolsnowwolf, 2020-04-16:
你的 工具链都还没编译，不能编译包

swhoro, 2020-04-16:
> 你的 工具链都还没编译，不能编译包

谢谢，新手接触这个，不太理解这个东西。编译了一次固件就好了。

closed 2020-04-16 by coolsnowwolf