coolsnowwolf/lede #4336: 日志里有大量的 mwan3track[3880]: Check (ping) failed日志

ckxt, 2020-04-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最近一周的源码编译出来的X86一堆这样的错误，不知道怎么解决，但不影响使用

### (2) 路由器型号和固件版本

A：X86 源码版本 OpenWrt R20.4.8 / LuCI Master (git-20.088.29068-cebe120) 

### (3) 详细日志

Wed Apr 15 23:32:19 2020 kern.notice kernel: [    1.338456] sd 2:0:0:0: [sda] Write Protect is off
Wed Apr 15 23:32:19 2020 kern.debug kernel: [    1.339848] sd 2:0:0:0: [sda] Mode Sense: 00 3a 00 00
Wed Apr 15 23:32:19 2020 kern.notice kernel: [    1.339990] sd 2:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.343597]  sda: sda1 sda2
Wed Apr 15 23:32:19 2020 kern.notice kernel: [    1.345121] sd 2:0:0:0: [sda] Attached SCSI disk
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.348122] VFS: Mounted root (squashfs filesystem) readonly on device 8:2.
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.351131] Freeing unused kernel image memory: 1068K
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.387625] Write protecting the kernel read-only data: 14336k
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.390796] Freeing unused kernel image memory: 2024K
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.392337] Freeing unused kernel image memory: 4K
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.393711] Run /sbin/init as init process
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.437417] usb 1-1: new high-speed USB device number 2 using ehci-pci
Wed Apr 15 23:32:19 2020 user.info kernel: [    1.453374] init: Console is alive
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.670863] input: QEMU QEMU USB Tablet as /devices/pci0000:00/0000:00:07.7/usb1/1-1/1-1:1.0/0003:0627:0001.0001/input/input4
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.674754] hid-generic 0003:0627:0001.0001: input,hidraw0: USB HID v0.01 Mouse [QEMU QEMU USB Tablet] on usb-0000:00:07.7-1/input0
Wed Apr 15 23:32:19 2020 user.info kernel: [    1.724687] kmodloader: loading kernel modules from /etc/modules-boot.d/*
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.735534] pps_core: LinuxPPS API ver. 1 registered
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.737020] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.740477] PTP clock support registered
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.742178] Button Hotplug driver version 0.4.1
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.744138] exFAT: Version 1.2.9
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.745824] usbcore: registered new interface driver uas
Wed Apr 15 23:32:19 2020 user.info kernel: [    1.747453] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.747458] tsc: Refined TSC clocksource calibration: 2294.781 MHz
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.751202] clocksource: tsc: mask: 0xffffffffffffffff max_cycles: 0x2113f3155eb, max_idle_ns: 440795252171 ns
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.755203] clocksource: Switched to clocksource tsc
Wed Apr 15 23:32:19 2020 user.info kernel: [    1.757218] init: - preinit -
Wed Apr 15 23:32:19 2020 kern.notice kernel: [    1.834812] random: jshn: uninitialized urandom read (4 bytes read)
Wed Apr 15 23:32:19 2020 kern.notice kernel: [    1.840623] random: jshn: uninitialized urandom read (4 bytes read)
Wed Apr 15 23:32:19 2020 kern.notice kernel: [    1.844107] random: jshn: uninitialized urandom read (4 bytes read)
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.889755] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
Wed Apr 15 23:32:19 2020 kern.info kernel: [    1.891326] 8021q: adding VLAN 0 to HW filter on device eth0
Wed Apr 15 23:32:19 2020 kern.info kernel: [    2.067560] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Wed Apr 15 23:32:19 2020 kern.warn kernel: [    5.911676] urandom_read: 4 callbacks suppressed
Wed Apr 15 23:32:19 2020 kern.notice kernel: [    5.911678] random: jshn: uninitialized urandom read (4 bytes read)
Wed Apr 15 23:32:19 2020 user.info kernel: [    5.928446] mount_root: loading kmods from internal overlay
Wed Apr 15 23:32:19 2020 user.info kernel: [    5.936949] kmodloader: loading kernel modules from //etc/modules-boot.d/*
Wed Apr 15 23:32:19 2020 user.info kernel: [    5.939133] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
Wed Apr 15 23:32:19 2020 user.info kernel: [    5.978675] block: attempting to load /tmp/overlay/upper/etc/config/fstab
Wed Apr 15 23:32:19 2020 user.err kernel: [    5.980626] block: unable to load configuration (fstab: Entry not found)
Wed Apr 15 23:32:19 2020 user.info kernel: [    5.982459] block: attempting to load /tmp/overlay/etc/config/fstab
Wed Apr 15 23:32:19 2020 user.err kernel: [    5.984192] block: unable to load configuration (fstab: Entry not found)
Wed Apr 15 23:32:19 2020 user.info kernel: [    5.985922] block: attempting to load /etc/config/fstab
Wed Apr 15 23:32:19 2020 user.err kernel: [    5.987820] block: unable to load configuration (fstab: Entry not found)
Wed Apr 15 23:32:19 2020 user.err kernel: [    5.989575] block: no usable configuration
Wed Apr 15 23:32:19 2020 kern.notice kernel: [    5.997452] random: procd: uninitialized urandom read (4 bytes read)
Wed Apr 15 23:32:19 2020 user.info kernel: [    6.067812] mount_root: rootdisk overlay filesystem has not been formatted yet
Wed Apr 15 23:32:19 2020 kern.notice kernel: [    6.458226] random: mkfs.ext4: uninitialized urandom read (16 bytes read)
Wed Apr 15 23:32:19 2020 kern.info kernel: [    6.660735] EXT4-fs (loop0): mounted filesystem with ordered data mode. Opts: (null)
Wed Apr 15 23:32:19 2020 user.info kernel: [    6.666727] block: attempting to load /tmp/overlay/upper/etc/config/fstab
Wed Apr 15 23:32:19 2020 user.err kernel: [    6.668576] block: unable to load configuration (fstab: Entry not found)
[... 564 lines cut ...]
Thu Apr 16 13:00:34 2020 user.info mwan3track[3880]: Check (ping) failed for target "114.114.115.115" on interface wan (pppoe-wan)
Thu Apr 16 13:00:39 2020 user.info mwan3track[3880]: Lost 1 ping(s) on interface wan (pppoe-wan)
Thu Apr 16 13:11:39 2020 user.info mwan3track[3880]: Check (ping) failed for target "114.114.115.115" on interface wan (pppoe-wan)
Thu Apr 16 13:11:44 2020 user.info mwan3track[3880]: Lost 1 ping(s) on interface wan (pppoe-wan)
Thu Apr 16 13:15:37 2020 user.info mwan3track[3880]: Check (ping) failed for target "114.114.115.115" on interface wan (pppoe-wan)
Thu Apr 16 13:15:42 2020 user.info mwan3track[3880]: Lost 1 ping(s) on interface wan (pppoe-wan)
Thu Apr 16 13:19:10 2020 user.info mwan3track[3880]: Check (ping) failed for target "114.114.115.115" on interface wan (pppoe-wan)
Thu Apr 16 13:19:15 2020 user.info mwan3track[3880]: Lost 1 ping(s) on interface wan (pppoe-wan)
Thu Apr 16 13:30:15 2020 user.info mwan3track[3880]: Check (ping) failed for target "114.114.115.115" on interface wan (pppoe-wan)
Thu Apr 16 13:30:20 2020 user.info mwan3track[3880]: Lost 1 ping(s) on interface wan (pppoe-wan)
Thu Apr 16 13:34:03 2020 user.info mwan3track[3880]: Check (ping) failed for target "114.114.115.115" on interface wan (pppoe-wan)
Thu Apr 16 13:34:08 2020 user.info mwan3track[3880]: Lost 1 ping(s) on interface wan (pppoe-wan)
Thu Apr 16 13:45:08 2020 user.info mwan3track[3880]: Check (ping) failed for target "114.114.115.115" on interface wan (pppoe-wan)
Thu Apr 16 13:45:13 2020 user.info mwan3track[3880]: Lost 1 ping(s) on interface wan (pppoe-wan)
Thu Apr 16 13:52:42 2020 user.info mwan3track[3880]: Check (ping) failed for target "114.114.115.115" on interface wan (pppoe-wan)
Thu Apr 16 13:52:47 2020 user.info mwan3track[3880]: Lost 1 ping(s) on interface wan (pppoe-wan)
Thu Apr 16 14:03:42 2020 user.info mwan3track[3880]: Check (ping) failed for target "114.114.115.115" on interface wan (pppoe-wan)
Thu Apr 16 14:03:47 2020 user.info mwan3track[3880]: Lost 1 ping(s) on interface wan (pppoe-wan)
Thu Apr 16 14:11:27 2020 user.info mwan3track[3880]: Check (ping) failed for target "114.114.115.115" on interface wan (pppoe-wan)
Thu Apr 16 14:11:32 2020 user.info mwan3track[3880]: Lost 1 ping(s) on interface wan (pppoe-wan)
Thu Apr 16 14:15:10 2020 user.info mwan3track[3880]: Check (ping) failed for target "114.114.115.115" on interface wan (pppoe-wan)
Thu Apr 16 14:15:17 2020 user.info mwan3track[3880]: Check (ping) failed for target "114.114.115.115" on interface wan (pppoe-wan)
Thu Apr 16 14:15:22 2020 user.info mwan3track[3880]: Lost 2 ping(s) on interface wan (pppoe-wan)
Thu Apr 16 14:17:19 2020 user.info mwan3track[3880]: Check (ping) failed for target "114.114.114.114" on interface wan (pppoe-wan)
Thu Apr 16 14:17:24 2020 user.info mwan3track[3880]: Lost 1 ping(s) on interface wan (pppoe-wan)

ckxt, 2020-04-16:
pppoe-wan那边以及DNS解析记录里我也看到，没有重新拨号的情况产生

goodffd, 2020-04-16:
如果只是单线接入的话，关闭这个服务，包括负载均衡，否则会出现频繁域名解析不了的情况。我之前就是频繁出现国内域名无法解析，尤其到了晚上。后来关闭这个nwan3服务，一切都正常了。

fuokgfw, 2020-04-17:
以前我漏过忘记勾选ipip这个，你看看有没有漏，现在的版本需不需要不知道，我一直会去勾选

ckxt, 2020-04-17:
> 以前我漏过忘记勾选ipip这个，你看看有没有漏，现在的版本需不需要不知道，我一直会去勾选

把固件重新刷进优盘，没问题了，很诡异的问题

closed 2020-04-17 by ckxt

dalu-test, 2021-08-31:
我也是一堆这个日志，而且经常无故上不了网，只能重启路由器。看了下，默认开启了负载均衡，刚才已经关闭了，观察几天我再来反馈是否解决。

dalu-test, 2021-09-05:
> 我也是一堆这个日志，而且经常无故上不了网，只能重启路由器。看了下，默认开启了负载均衡，刚才已经关闭了，观察几天我再来反馈是否解决。

果然解决了