coolsnowwolf/lede #4341: 设置WiFi加密，一直无法生效，求大佬指点

kuole-o, 2020-04-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
设置WiFi加密，一直无法生效，求大佬指点
a.我第一种方法，在sh脚本里修改wireless配置  结果不生效；
b.后来直接在 yml里改wireless配置，还是不生效

我的wifi密码放在了仓库的secrets里，作为变量引用的。密码是一个11位的字母+数字+小数点。而且，每次刷上包，etc/config下没有 wireless文件，在ssh里用命令导出WiFi配置，也是一个空的wireless文件。手动放一个，设置权限后，也无法生效；ssh中，uci set wireless.@wifi-iface[0].key=xxx 也不能生效（操作完有提交）

    - name: 设置WiFi密码
      env:
        WIFI_PASSWD: ${{ secrets.WIFI_PASSWD }}
      if: env.WIFI_PASSWD
      run: |
        sed -i '$i uci set wireless.@wifi-iface[0].encryption=psk2' openwrt/package/lean/default-settings/files/zzz-default-settings
        sed -i '$i uci set wireless.@wifi-iface[0].key=${{ env.WIFI_PASSWD }}' openwrt/package/lean/default-settings/files/zzz-default-settings
        sed -i '$i uci set wireless.@wifi-iface[1].encryption=psk2' openwrt/package/lean/default-settings/files/zzz-default-settings
        sed -i '$i uci set wireless.@wifi-iface[1].key=${{ env.WIFI_PASSWD }}' openwrt/package/lean/default-settings/files/zzz-default-settings
        sed -i '$i uci commit wireless' openwrt/package/lean/default-settings/files/zzz-default-settings

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
我的配置文件：https://github.com/kuole-o/Actions-OpenWrt/blob/master/.github/workflows/build-openwrt.yml

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：斐讯k2p A1

### (3) 详细日志
无

kuole-o, 2020-04-16:
另一个问题：添加koolpoxyR插件，刷完包，web里也找不到，求解

sh文件里：https://github.com/kuole-o/Actions-OpenWrt/blob/master/diy.sh
cd openwrt
git clone https://github.com/tzxiaozhen88/luci-app-koolproxyR.git ./package/feeds/luci/applications 

yml文件里：https://github.com/kuole-o/Actions-OpenWrt/blob/master/.github/workflows/build-openwrt.yml
   - name: 增加koolproxyR插件        
      run: | 
           cat >> .config <<EOF
           CONFIG_PACKAGE_luci-app-koolproxyr=y
           EOF

AmadeusGhost, 2020-04-16:
k2p默认用的是闭源驱动，需要用mtwifi控制，还有kpr k2p跑得动吗...

kuole-o, 2020-04-16:
> k2p预设用的是闭源驱动，需要用mtwifi控制，还有kpr k2p跑得动吗...

求详细点，mtwifi控制 怎么改密码呢？上面的方法，需要怎么改动吗。。不想直接改源码。。

kuole-o, 2020-04-20:
mtwifi 不能通过上述方式更改密码。使用files大法，将mt下的两个 dat文件放到源码中后，问题解决了（注意删除原来的，否则会冲突报错）  有遇到一样问题的朋友，可以看我的yml文件，希望能帮到你~

设备 K2P

closed 2020-07-04 by coolsnowwolf