coolsnowwolf/lede #4342: SSR plus独立后，修改feeds后menuconfig未显示

marcosteam, 2020-04-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

SSR plus独立为helloworld以后，更新最新源码，发现根目录的`feeds.conf.default`有一行额外的内容：
[image]
将helloworld有关代码去掉注释以后，亦可正常feeds update和feeds install，但是menuconfig里面没有看到有关编译选项。
还要修改哪些额外内容？

martin0607, 2020-04-16:
没有发现你说的问题

kuole-o, 2020-04-16:
看到另一条评论里说，先rm -rf ./tmp; 接着再改注释；然后update 和install feeds就有了。我也正在编译尝试

mjyhj, 2020-04-16:
加了个#，注释了。当然不显示

js882829, 2020-04-16:
cd lede
1.git pull
2.去掉#号
3. rm ./tmp -fr
4.    ./scripts/feeds update -a && ./scripts/feeds install -a
5.  make menuconfig
6.  make V=s

closed 2020-04-16 by coolsnowwolf

kktt007, 2020-04-19:
大屌表示,我就静静的看着你们不说话~~

ddcatgg, 2020-04-23:
我还以为L大要开始主推passwall或者小猫咪什么的了