coolsnowwolf/lede #4346: 酸酸乳被删除了吗？

liubangjain, 2020-04-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
酸酸乳是不是取消了？
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
 升级到OpenWrt R20.4.8 / LuCI Master (git-20.088.29068-cebe120) 发现没有了酸酸乳。
### (1) 具体问题

A：

### (2) 路由器型号和固件版本
 OpenWrt R20.4.8 / LuCI Master (git-20.088.29068-cebe120) 
A：

### (3) 详细日志

A：

wnn, 2020-04-16:
被[fw876](https://github.com/coolsnowwolf/lede/commits?author=fw876)在11小时之前删除了，不知道为什么？

https://github.com/coolsnowwolf/lede/commit/988acf9daa2c80858c1fd8c097d0dffe49df858d

hahach, 2020-04-16:
同问，make menuconfig里面没有luci-app-ssr-plus了。

kevinlo0909, 2020-04-16:
1.cd lede
2.vi feeds.conf.default 去掉#号 
3.
./scripts/feeds clean
./scripts/feeds update -a
./scripts/feeds install -a

xpc0938, 2020-04-16:
难道大佬被约谈了？

closed 2020-04-16 by coolsnowwolf