coolsnowwolf/lede #4347: 编译失败，求指点

yyan502, 2020-04-16:
[image]
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：

Repobor, 2020-04-16:
apt-get -y install gperf

acehong, 2020-04-17:
同样问题
po2lmo ./po/zh-cn/ssr-plus.po /home/bing/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/luci-app-ssr-plus/.pkgdir/luci-app-ssr-plus/usr/lib/lua/luci/i18n/ssr-plus.zh-cn.lmo
bash: po2lmo: command not found
Makefile:120: recipe for target '/home/bing/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/luci-app-ssr-plus/.pkgdir/luci-app-ssr-plus.installed' failed
make[3]: *** [/home/bing/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/luci-app-ssr-plus/.pkgdir/luci-app-ssr-plus.installed] Error 127
make[3]: Leaving directory '/home/bing/lede/feeds/helloworld/luci-app-ssr-plus'
time: package/feeds/helloworld/luci-app-ssr-plus/compile#0.12#0.01#0.17
package/Makefile:111: recipe for target 'package/feeds/helloworld/luci-app-ssr-plus/compile' failed
make[2]: *** [package/feeds/helloworld/luci-app-ssr-plus/compile] Error 2
make[2]: Leaving directory '/home/bing/lede'
package/Makefile:107: recipe for target '/home/bing/lede/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/bing/lede/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bing/lede'
/home/bing/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

lxwq, 2020-04-17:
如何处理?同样失败在这个位置,网上查资料好像要改makefile文件,

jw10126121, 2020-04-17:
同样的问题

lxwq, 2020-04-17:
很奇怪,先取消$$R编译一次,第二次再选上,就通过了,不知道什么原理

yyan502, 2020-04-17:
> apt-get -y install gperf

感谢，按你的指点，编译成功了

jw10126121, 2020-04-17:
安装了golang，不然会提示golang timeout。然后重新删除build_dir和st**_dir，重新编译，好像是可以了

DanielChen008, 2020-04-17:
> apt-get -y install gperf

试过，依旧 recipe for target 'world' failed 
晚上回去把整个目录 rm 掉重新试试

cnclg, 2020-04-17:
> 同样的问题，gperf也安装了的

make package/feeds/luci/luci-base/compile V=99 再编译，我试了可以解决这个问题。

jw10126121, 2020-04-17:
> > 同样的问题，gperf也安装了的
> 
> make package/feeds/luci/luci-base/compile V=99 再编译，我试了可以解决这个问题。

非常感谢，已经解决了，也知道为什么出这错了

blacknesswing, 2020-04-17:
> > > 同样的问题，gperf也安装了的
> > 
> > 
> > make package / feeds / luci / luci-base / compile V = 99再编译，我试了可以解决这个问题。
> 
> 非常感谢，已经解决了，也知道为什么出这错了

同样问题 请问怎么解决的 谢谢🙏

jw10126121, 2020-04-17:
> > > > 同样的问题，gperf也安装了的
> > > 
> > > 
> > > make package / feeds / luci / luci-base / compile V = 99再编译，我试了可以解决这个问题。
> > 
> > 
> > 非常感谢，已经解决了，也知道为什么出这错了
> 
> 同样问题 请问怎么解决的 谢谢🙏

上面不是写着的吗？

DanielChen008, 2020-04-17:
找到原因了，是L大更新了编译固件所需的组件
`sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf`

我就是先报 gperf 错，又报了 antlr3 错，才发现这里更新了…

blizzardster, 2020-04-17:
apt-get -y install gperf

make package/feeds/luci/luci-base/compile V=99

即可，完美搞定，感谢两位，

gc89925, 2020-04-18:
> apt-get -y install gperf

输入make package/feeds/luci/luci-base/compile V=99后提示
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'microsocks', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'tcpping', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'redsocks2', which does not exist

blizzardster, 2020-04-19:
> 输入make package/feeds/luci/luci-base/compile V=99后提示
> WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'microsocks', which does not exist
> WARNING: Makefile [...]

你缺少个ssr-plus

gfpyc, 2020-04-23:
> > 同样的问题，gperf也安装了的
> 
> make package/feeds/luci/luci-base/compile V=99 再编译，我试了可以解决这个问题。

3Q 同样问题 解决了

closed 2020-07-04 by coolsnowwolf