coolsnowwolf/lede #4353: ssr-plus 换名字么了？package/lean里面怎么找不到了

azazjjyjjy, 2020-04-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
[x] 没有类似的issue

## 2. 详细叙述
编译找不到luci对应包选项
### (1) 具体问题

A：上次编译还有的，这次pull&update了之后就没了，是不是最近做了什么大改动？

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

dolphinpaopao, 2020-04-17:
issue问这样的问题，只有被关闭一个结果。没有了就回退用老版本或者等待。

norikohxb, 2020-04-17:
估计是elan大的压力很大了。所以ssrplus以后ean大不能再开发了，否则可能会受到zf的威胁。所以删除了。建议用一下helloworld

ZqinKing, 2020-04-17:
有地方能开的

norikohxb, 2020-04-17:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
> [x] 没有类似的issue
> 
> ## 2. 详细叙述
> 编译找不到luci对应包选项
> 
> ### (1) 具体问题
> A：上次编译还有的，这次pull&update了之后就没了，是不是最近做了什么大改动？
> 
> ### (2) 路由器型号和固件版本
> A：
> 
> ### (3) 详细日志
> A：

cd /home/你的id名字/lede/package/lean
rm -rf luci-app-ssr-plus
git clone https://github.com/fw876/helloworld.git
然后就都有了。

感觉在linux这种开放平台下，政府是封杀不干净的，公道自在人心，谁说翻墙出去的就一定是反动的。

norikohxb, 2020-04-17:
> 有地方能开的
说说你是怎么可以开的，git pul的时候，都rm了 luci-app-ssr-plus
我感觉现在lean大真的是顶着很大的压力。
其实为什么要删除ssr-plus分析一下就知道了。

azazjjyjjy, 2020-04-17:
> cd /home/你的id名字/lede/package/lean
> rm -rf luci-app-ssr-plus
> git clone https://github.com/fw876/helloworld.git
> 然后就都有了。
>
> 感觉在linux这种开放平台下，政府是封杀不干净的，公道自在人心，谁说翻墙出去的就一定是反动的。

谢谢！

closed 2020-04-17 by azazjjyjjy

Xixi20160512, 2020-04-17:
> cd /home/你的id名字/lede/package/lean
> rm -rf luci-app-ssr-plus
> git clone https://github.com/fw876/helloworld.git
> 然后就都有了。
>
> 感觉在linux这种开放平台下，政府是封杀不干净的，公道自在人心，谁说翻墙出去的就一定是反动的。

当你的周遭被黑暗和寒冷包围的时候，总会有人站出来成为传薪者。谢谢。

fuokgfw, 2020-04-18:
不管怎么说，感谢Lean大，谢谢

DrakeShi, 2020-04-19:
> cd /home/你的id名字/lede/package/lean
> rm -rf luci-app-ssr-plus
> git clone https://github.com/fw876/helloworld.git
> 然后就都有了。
>
> 感觉在linux这种开放平台下，政府是封杀不干净的，公道自在人心，谁说翻墙出去的就一定是反动的。

求大侠再详细指点下，从哪里插入你上述的指令，本人小白啊

azazjjyjjy, 2020-04-23:
> 求大侠再详细指点下，从哪里插入你上述的指令，本人小白啊

在Linux的命令行下面执行这些。
cd这行是说前进打开这个目录
rm 这行是清除以前的残留
git clone会自动下载你要的这个链接到这个目录里面