coolsnowwolf/lede #4357: > 重新编译了新版本后你会发现,不只是不能分流那么简单了~

h605290713, 2020-04-17:
> 重新编译了新版本后你会发现,不只是不能分流那么简单了~

_Originally posted by @h605290713 in https://github.com/coolsnowwolf/lede/issues/4355#issuecomment-615202134_

ecco2005, 2020-04-17:
酸酸乳没了……这是为什么？

happyboy19, 2020-04-17:
为什么不用比它体验更好的呢？

lcdwork, 2020-04-17:
酸酸乳。。懵了

ecco2005, 2020-04-17:
幸好我存了一份几天前的源码，大家可以去我仓库下载，然后用winscp传到对应目录里再make menuconfig就能看到了

yangabc123, 2020-04-17:
不知道什么原因撤下了

jjit, 2020-04-17:
> 酸酸乳没了……这是为什么？
单独移出来了
https://github.com/coolsnowwolf/lede/blob/master/feeds.conf.default
移到这里了
https://github.com/fw876/helloworld

dolphinpaopao, 2020-04-17:
去看看现在还open的一些PR，个人认为目前没有SSR是更好的，应该可以减少垃圾PR数量。有需要的功能自己多动手搜索添加。

Aes64X, 2020-04-17:
是该增加一些门槛了。去看看单独的55R页面，还有跑去专门发issues支持表存在的。本身宅出去就是有各种顾虑，还有转慢跑过去拉存在的。脑子是被自己吃了？？？还是本身就没脑子？ 建议增加编译门槛。

ecco2005, 2020-04-17:
> > 酸酸乳没了……这是为什么？
> 
> 单独移出来了
> https://github.com/coolsnowwolf/lede/blob/master/feeds.conf.default
> 移到这里了
> https://github.com/fw876/helloworld
> 修改 feeds.conf.default 文件，去掉前面的 #
> 然后运行 ./scripts/feeds 就可以了

明白了。可是还是不懂L大为什么要这么操作

mjyhj, 2020-04-17:
自己去源码更新了什么，不要动不动就发个Issues

dolphinpaopao, 2020-04-17:
> 自己去源码更新了什么，不要动不动就发个Issues

发issue的很多人认为这就是随心所欲的论坛搞版聊，三言两语什么都无就要帮助。看现在作者都已经不鸟了。

iKenzo, 2020-04-17:
炫耀道德和技术优越感无意义，请多一些分享和贡献精神

dolphinpaopao, 2020-04-18:
> 炫耀道德和技术优越感无意义，请多一些分享和贡献精神

人家只是陈述一个现在存在的事实，作者多次强调issue不要拿来随意问问题，不要作为论坛来版聊，你们去看看现在open和closed的，多少是尊重了作者的意愿？让别人奉献享受别人的成果之前是不是应该先端正一下自己的态度？最起码的尊重都做不到就要求别人是不是也是自私自利的一种表现？多少人连搜索都不愿意搜一下，更何谈在线编译上线后不胜其烦的垃圾PR。

closed 2020-04-18 by h605290713